abizovnuralem/go2_ros2_sdk #21: secondary development for non-edu versions

JKK-dr, 2024-04-20:
We're interested in the fantastic open-source work you've done, but we're considering purchasing Go2 Air. However, we're unsure if there's the capability for secondary development on Air. The official customer service mentioned that non-education versions don't provide development interfaces. Have you tried developing on Air? If we use your repository, can it solve this problem?

abizovnuralem, 2024-04-20:
Hi, it depends what you want to develop? Can you provide more details on your product?--Отправлено из мобильной Яндекс Почты

JKK-dr, 2024-04-21:
I want to develop AI-related applications based on the official APIs provided.

abizovnuralem, 2024-04-21:
from the hardware side, there is not so big difference. EDU has foot sensors, an audio module (bi-directional), and a Nvidia jetson. Pro doesn't have a jetson and foot force. Air does not have a jetson, foot force, and an audio module. Other things are the same.  From the software side. the difference between official and non-official ros2 SDK is webrtc protocol, in our case we use it, but it has some limitations with speed (twice lower than dds on the original SDK) and it does not have low_level control (because it was not implemented on webrtc).

JKK-dr, 2024-04-22:
Thank you very much for your response. Our specific AI project involves indoor mobile navigation. Based on high-level motion control, can the AI navigate stairs at different heights?

JKK-dr, 2024-04-22:
Thank you very much for your reply. We're excited about the SDK features of the go2's Air, as we don't intend to delve into low-level motion development (down to the angle of leg motors). We're focusing on large-scale models and reinforcement learning in embodied intelligence (indoor Vision Language Navigation). We'd like to ask you again: Can the air navigate stairs at different heights based on high-level motion control? We appreciate your help tremendously.

JKK-dr, 2024-04-22:
Thank you very much for your reply. We're excited about the SDK features of the go2's Air, as we don't intend to delve into low-level motion development (down to the angle of leg motors). We're focusing on LLMs and reinforcement learning in embodied intelligence (indoor Vision Language Navigation). We'd like to ask you again: Can the air navigate stairs at different heights based on high-level motion control? We appreciate your help tremendously.

abizovnuralem, 2024-04-22:
ofc it can, Air is a great product, also I have developed a sim env https://github.com/abizovnuralem/go2_omniverse, where you can test your algorithms. It will be an easy start for you with these two repos.

abizovnuralem, 2024-04-22:
> Very impressive sim env! does this works with latest unitree firmware? the one with the "extra agile" mode?

yes

closed 2024-05-10 by abizovnuralem