abizovnuralem/go2_ros2_sdk #22: rviz only shows the body of the go2, but not the four legs and camera.

BFZD233, 2024-04-26:
Thanks for sharing the code. I'm just getting started with robotics and I'm having problems with the configuration process, can you provide me with a solution?

After I run "ros2 launch go2_robot_sdk robot.launch.py", rviz only shows the body of the go2, but not the four legs and camera.

The status bar of rviz shows:
FL_calf No transform from [FL_calf] to [odom]
FL_calflower No transform from [FL_calflower] to [odom]
...
base_link Transform OK
front_camera Transform OK

I was prompted by terminal

Translated with DeepL.com (free version)

1714101769.8241620 [INFO] [launch]: All log files can be found below /home/smbu/.ros/log/2024-04-26-11-22-49-823525-smbu-H3CDesk-X500t-4993
1714101769.8242383 [INFO] [launch]: Default logging verbosity is set to INFO
1714101770.2460189 [INFO] [foxglove_bridge-1]: process started with pid [4994]
1714101770.2461760 [INFO] [async_slam_toolbox_node-2]: process started with pid [4996]
1714101770.2462521 [INFO] [controller_server-3]: process started with pid [4998]
1714101770.2463179 [INFO] [smoother_server-4]: process started with pid [5000]
1714101770.2463772 [INFO] [planner_server-5]: process started with pid [5002]
1714101770.2464349 [INFO] [behavior_server-6]: process started with pid [5004]
1714101770.2464912 [INFO] [bt_navigator-7]: process started with pid [5006]
1714101770.2465467 [INFO] [waypoint_follower-8]: process started with pid [5008]
1714101770.2466023 [INFO] [velocity_smoother-9]: process started with pid [5010]
1714101770.2466564 [INFO] [lifecycle_manager-10]: process started with pid [5012]
1714101770.2467093 [INFO] [pointcloud_to_laserscan_node-11]: process started with pid [5014]
1714101770.2467778 [INFO] [robot_state_publisher-12]: process started with pid [5016]
1714101770.2468510 [INFO] [joy_node-13]: process started with pid [5018]
1714101770.2469246 [INFO] [teleop_node-14]: process started with pid [5020]
1714101770.2469964 [INFO] [twist_mux-15]: process started with pid [5022]
1714101770.2470691 [INFO] [go2_driver_node-16]: process started with pid [5024]
1714101770.2471423 [INFO] [go2_proc_text-17]: process started with pid [5026]
1714101770.2472136 [INFO] [ros2_go2_video-18]: process started with pid [5028]
1714101770.2472858 [INFO] [rviz2-19]: process started with pid [5030]
1714101770.3323753 [teleop_node-14] [INFO] [1714101770.331942645] [TeleopTwistJoy]: Teleop enable button 6.
1714101770.3324599 [teleop_node-14] [INFO] [1714101770.331979655] [TeleopTwistJoy]: Turbo on button 7.
1714101770.3325109 [teleop_node-14] [INFO] [1714101770.331984239] [TeleopTwistJoy]: Linear axis x on 1 at scale 0.400000.
1714101770.3325570 [teleop_node-14] [INFO] [1714101770.331991132] [TeleopTwistJoy]: Turbo for linear axis x is scale 1.000000.
1714101770.3326035 [teleop_node-14] [INFO] [1714101770.331994646] [TeleopTwistJoy]: Linear axis y on 0 at scale 0.400000.
1714101770.3326457 [teleop_node-14] [INFO] [1714101770.331998100] [TeleopTwistJoy]: Turbo for linear axis y is scale 1.000000.
1714101770.3326888 [teleop_node-14] [INFO] [1714101770.332001713] [TeleopTwistJoy]: Angular axis yaw on 2 at scale 1.500000.
1714101770.3327289 [teleop_node-14] [INFO] [1714101770.332005273] [TeleopTwistJoy]: Turbo for angular axis yaw is scale 1.500000.
1714101770.3434558 [foxglove_bridge-1] [INFO] [1714101770.343245068] [foxglove_bridge_component_manager]: Load Library: /opt/ros/humble/lib/libfoxglove_bridge_component.so
1714101770.3536663 [foxglove_bridge-1] [INFO] [1714101770.353457369] [foxglove_bridge_component_manager]: Found class: rclcpp_components::NodeFactoryTemplate<foxglove_bridge::FoxgloveBridge>
1714101770.3537610 [foxglove_bridge-1] [INFO] [1714101770.353488868] [foxglove_bridge_component_manager]: Instantiate class: rclcpp_components::NodeFactoryTemplate<foxglove_bridge::FoxgloveBridge>
1714101770.3553498 [foxglove_bridge-1] [INFO] [1714101770.355222117] [foxglove_bridge]: Starting foxglove_bridge (humble, 0.7.4@) with WebSocket++/0.8.2
1714101770.3559506 [foxglove_bridge-1] [INFO] [1714101770.355829268] [foxglove_bridge]: [WS] Server running without TLS
1714101770.3579469 [foxglove_bridge-1] [INFO] [1714101770.357767000] [foxglove_bridge]: [WS] WebSocket server listening at ws://0.0.0.0:8765
1714101770.3581877 [foxglove_bridge-1] [INFO] [1714101770.357769389] [foxglove_bridge]: [WS] WebSocket server run loop started
1714101770.4229238 [go2_driver_node-16] Traceback (most recent call last):
1714101770.4230151 [go2_driver_node-16]   File "/home/smbu/lzd/ROS2/ros2_ws/install/go2_robot_sdk/lib/go2_robot_sdk/go2_driver_node", line 33, in <module>
1714101770.4232149 [go2_driver_node-16]     sys.exit(load_entry_point('go2-robot-sdk==0.0.0', 'console_scripts', 'go2_driver_node')())
1714101770.4232574 [go2_driver_node-16]   File "/home/smbu/lzd/ROS2/ros2_ws/install/go2_robot_sdk/lib/go2_robot_sdk/go2_driver_node", line 25, in importlib_load_entry_point
1714101770.4232841 [go2_driver_node-16]     return next(matches).load()
1714101770.4233065 [go2_driver_node-16]   File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 171, in load
1714101770.4234450 [go2_driver_node-16]     module = import_module(match.group('module'))
1714101770.4234788 [go2_driver_node-16]   File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
1714101770.4235024 [go2_driver_node-16]     return _bootstrap._gcd_import(name[level:], package, level)
1714101770.4235241 [go2_driver_node-16]   File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
1714101770.4236481 [go2_driver_node-16]   File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
1714101770.4236774 [go2_driver_node-16]   File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
1714101770.4237008 [go2_driver_node-16]   File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
1714101770.4237216 [go2_driver_node-16]   File "<frozen importlib._bootstrap_external>", line 883, in exec_module
1714101770.4237406 [go2_driver_node-16]   File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
1714101770.4237599 [go2_driver_node-16]   File "/home/smbu/lzd/ROS2/ros2_ws/install/go2_robot_sdk/lib/python3.10/site-packages/go2_robot_sdk/go2_driver_node.py", line 33, in <module>
1714101770.4238594 [go2_driver_node-16]     from scripts.go2_lidar_decoder import update_meshes_for_cloud2
1714101770.4238887 [go2_driver_node-16]   File "/home/smbu/lzd/ROS2/ros2_ws/install/go2_robot_sdk/lib/python3.10/site-packages/scripts/go2_lidar_decoder.py", line 29, in <module>
1714101770.4239168 [go2_driver_node-16]     from wasmtime import Config, Engine, Store, Module, Instance, Func, FuncType
[... 125 lines cut ...]
1714102490.3981986 [foxglove_bridge-1] [ERROR] [1714102490.385081529] [foxglove_bridge]: [WS] handle_accept error: Operation aborted.
1714102490.3982170 [foxglove_bridge-1] [INFO] [1714102490.385088826] [foxglove_bridge]: [WS] Stopping acceptance of new connections because the underlying transport is no longer listening.
1714102490.3982363 [foxglove_bridge-1] [INFO] [1714102490.385118300] [foxglove_bridge]: [WS] WebSocket server run loop stopped
1714102490.3982544 [foxglove_bridge-1] [INFO] [1714102490.385301723] [foxglove_bridge]: [WS] WebSocket server run loop terminated
1714102490.3982732 [foxglove_bridge-1] [INFO] [1714102490.385329830] [foxglove_bridge]: Shutdown complete
1714102490.4203668 [ERROR] [go2_proc_text-17]: process has died [pid 5026, exit code -2, cmd '/home/smbu/lzd/ROS2/ros2_ws/install/go2_robot_sdk/lib/go2_robot_sdk/go2_proc_text --ros-args'].
1714102490.4976754 [INFO] [teleop_node-14]: process has finished cleanly [pid 5020]
1714102490.4980397 [INFO] [twist_mux-15]: process has finished cleanly [pid 5022]
1714102490.5040469 [INFO] [foxglove_bridge-1]: process has finished cleanly [pid 4994]
1714102490.5055547 [INFO] [robot_state_publisher-12]: process has finished cleanly [pid 5016]
1714102490.5066586 [INFO] [pointcloud_to_laserscan_node-11]: process has finished cleanly [pid 5014]
1714102490.5319357 [INFO] [joy_node-13]: process has finished cleanly [pid 5018]
1714102490.5343614 [INFO] [velocity_smoother-9]: process has finished cleanly [pid 5010]
1714102490.5463502 [INFO] [async_slam_toolbox_node-2]: process has finished cleanly [pid 4996]
1714102490.5533264 [INFO] [smoother_server-4]: process has finished cleanly [pid 5000]
1714102490.5594397 [INFO] [waypoint_follower-8]: process has finished cleanly [pid 5008]
[... cut ...] _vel:=cmd_vel_nav'].
1714102490.8776829 [WARNING] [launch]: user interrupted with ctrl-c (SIGINT) again, ignoring...
1714102491.0863845 [WARNING] [launch]: user interrupted with ctrl-c (SIGINT) again, ignoring...
1714102491.0897021 [ERROR] [planner_server-5]: process has died [pid 5002, exit code -2, cmd '/opt/ros/humble/lib/nav2_planner/planner_server --ros-args --log-level info --ros-args -r __node:=planner_server --params-file /tmp/launch_params_mazeqihn -r /tf:=tf -r /tf_static:=tf_static'].

abizovnuralem, 2024-04-26:
Hello, it seems to me you don’t use right ip address --Отправлено из мобильной Яндекс Почты

1714101770.4235241 [go2_driver_node-16]   File "", line 1050, in _gcd_import
1714101770.4236481 [go2_driver_node-16]   File "", line 1027, in _find_and_load
1714101770.4236774 [go2_driver_node-16]   File "", line 1006, in _find_and_load_unlocked
1714101770.4237008 [go2_driver_node-16]   File "", line 688, in _load_unlocked
1714101770.4237216 [go2_driver_node-16]   File "", line 883, in exec_module
1714101770.4237406 [go2_driver_node-16]   File "", line 241, in _call_with_frames_removed

zmling22, 2024-04-26:
Can only run in WiFi mode? Is it possible to connect using an Ethernet cable？

abizovnuralem, 2024-04-26:
in future updates I will add ethernet dds support for it

heiheiHelix, 2024-04-26:
How can I set the WiFi mode, is there any documentation?
I tried to connect to the go2 AP-mode, but I failed to open the webpage 192.168.12.1. And I don't know the username and password for SSH.

abizovnuralem, 2024-04-26:
you need to use an official unitree mobile app to setup a wifi connection between a dog and a router.

heiheiHelix, 2024-04-27:
Thank for your reply. 
We change to wifi mode and use right ip address.
Now we can 
observe the image capture by Go2 camera, but  the four legs of Go2 is still invisible.

here are screen shot and log of "ros2 launch go2_robot_sdk robot.launch.py",
[image]
[error.txt](https://github.com/abizovnuralem/go2_ros2_sdk/files/15136911/error.txt)

BFZD233, 2024-04-27:
Thank you for sharing, we have solved the problem. 
The problem is mainly arose from conflict between  our conda python and /usr/bin/python3.

closed 2024-04-27 by BFZD233

mvrius, 2024-06-02:
@BFZD233 I'm running into the same problem. What was the solution for you?

Houruizhi, 2025-05-18:
> Thank you for sharing, we have solved the problem. The problem is mainly arose from conflict between our conda python and /usr/bin/python3.

what conflict? Please give some details please