abizovnuralem/go2_ros2_sdk #23: No Odom and Lidar(PointCLoud) Topic

David9696, 2024-04-26:
Hi,

Thanks for sharing the package.

I can use this package to see the camera view from my go2 pro robot. I can see the IMU and go2_states as well.

But I cant see the point cloud(Lidar) from the robot. The topic output is empty. Also. there is no odom topic shown.

May I know what is the problem?

Thanks.

Regards,

David

abizovnuralem, 2024-04-26:
Hello, Can you please provide a running rviz2 screenshot here? also, what wifi version are you using? it should be wifi 6

David9696, 2024-04-26:
Hi,

I am connecting the robot via Ethernet LAN cable. I can see the camera view and robot state. Is it a must to connect via wifi to see the lidar points?

Thanks.

abizovnuralem, 2024-04-26:
yes, currently ethernet is not available. I am working on it.

David9696, 2024-04-26:
Thanks for your reply.

FYI, I connected the robot via LAN. For camera and joint state, I can see and subscribe the ros2 topic. This is a big breakthrough.

Thanks for your great works. Looking forward to have the lidar topic support via LAN. I would remain this issue to be remained open. Thanks.

heiheiHelix, 2024-04-26:
> Hi,
> 
> I am connecting the robot via Ethernet LAN cable. I can see the camera view and robot state. Is it a must to connect via wifi to see the lidar points?
> 
> Thanks.

Hi， 

Hello, I'm also connected via Ethernet, but I can't see the camera or joints. Did you do any additional operations? I would be grateful if you could reply.

closed 2024-05-10 by abizovnuralem

David9696, 2024-05-17:
Hi @abizovnuralem 

Just to update. Apparently is the lidar on the robot is not working. Unitree has swapped the lidar. Now I can see the pointcloud from the robot.

Thanks.

David9696, 2024-05-17:
> Hi，
>
> Hello, I'm also connected via Ethernet, but I can't see the camera or joints. Did you do any additional operations? I would be grateful if you could reply.

U may follow the manual from Unitree for connecting via LAN.