abizovnuralem/go2_ros2_sdk #27: Errors loading geometries - URDF

MayooranT, 2024-05-06:
The Robot model is not visible because of this error for the URDF only. Seems like the .dae meshes are not loaded to display the model properly. At first that could be the path that causing errors but the path is correct. Are mesh files corrupted? 
[image]

abizovnuralem, 2024-05-06:
Hi! What versions of ros and ubuntu are you using?

MayooranT, 2024-05-06:
Ubuntu 20.04 and ROS Humble

abizovnuralem, 2024-05-10:
Files are not corrupted, but it seems to me that for some reason your files could be corrupted on your local copy, can you try to redownload it one more time, please? I just checked it and on my local PC, I don't have this error. The second option is to change package:// to the real full path on your PC and check again (for debug purpose).

closed 2024-05-20 by abizovnuralem