abizovnuralem/go2_ros2_sdk #28: Odom/Odom TF Jumping

David9696, 2024-05-17:
Hi,

When I started the driver launch file(ros2 launch go2_robot_sdk robot.launch.py), I can see the full robot model and pointcloud.

However, the robot model is jumping or shifting in rviz. When I check the tf, it seems that the odom tf is jumping. 

[
[image]
[image]
](url)

As shown in the pictures above, robot pose is changing in alternate in these two positions.

Do you know what is the issue?

Thanks.

David

abizovnuralem, 2024-05-17:
Hello, it seems to me that you have two sdk running on the same time? Have you installed official SDK? you need to spilt them on the network.

David9696, 2024-05-17:
Hi,

Thanks for your reply.

How to check whether 2 sdk are running? I didn't install any official sdk. 

How to split them on the network?

Thanks.

abizovnuralem, 2024-05-17:
1. you can check ros2 topic list, before executing it. it should be empty.
2. Are you using wifi for it? Ethernet is not supported (it will be in near future, or you can make a pr)
3. Wifi should have a strong signal and should be a 6 generation.

David9696, 2024-05-17:
Hi Kindly refer to my inline reply:

1. Can you can check ros2 topic list, before executing it. it should be empty.

This is the output:

/parameter_events
/rosout

But I think is not relevant.

2. Are you using wifi for it? Ethernet is not supported (it will be in near future, or you can make a pr)

Yes. I am using wifi not ethernet.

3. Wifi should have a strong signal and should be a 6 generation.

I think mine is 5 gen.

So the problem is network latency?

The odom is publishing at about 7 hz.
The point cloud is publishing at about 1 hz.

Also, I have this error:

[pointcloud_to_laserscan_node-8] [INFO] [1715915621.065207353] [pointcloud_to_laserscan]: Message Filter dropping message: frame 'odom' at time 1715915620.897 for reason 'Unknown'
[pointcloud_to_laserscan_node-8] terminate called after throwing an instance of 'tf2::ExtrapolationException'
[pointcloud_to_laserscan_node-8]   what():  Lookup would require extrapolation into the past.  Requested time 1715915636.889700 but the earliest data is at time 1715915637.066664, when looking up transform from frame [odom] to frame [radar]
[ERROR] [pointcloud_to_laserscan_node-8]: process has died [pid 31767, exit code -6, cmd '/opt/ros/humble/lib/pointcloud_to_laserscan/pointcloud_to_laserscan_node --ros-args -r __node:=pointcloud_to_laserscan --params-file /tmp/launch_params_uc6lkbvt -r cloud_in:=/point_cloud2'].

Is it mean that my wifi connection is too slow?

abizovnuralem, 2024-05-17:
It can cause this issue, I don't see other problems

David9696, 2024-05-19:
Hi,

I didn't see any issues from ros echo command.

Another thing that I found is that the pointcloud from the lidar is not updating when the robot is moving. Seem like is freezing. Although the pointcloud is publishing at the rate of about 2 Hz but the data seem not updating.

 Secondly, when i control the robot with joystick I would get the following errors:

go2_driver_node-6] INFO:scripts.webrtc_driver:Connection state is closed
[go2_driver_node-6] [INFO] [1716101867.772843663] [go2_driver_node]: Attack!
[go2_driver_node-6] ERROR:asyncio:Task exception was never retrieved
[go2_driver_node-6] future: <Task finished name='Task-4' coro=<RobotBaseNode.run() done, defined at /home/test/ros2_ws/install/go2_robot_sdk/lib/python3.8/site-packages/go2_robot_sdk/go2_driver_node.py:308> exception=InvalidStateError()>
[go2_driver_node-6] Traceback (most recent call last):
[go2_driver_node-6]   File "/home/test/ros2_ws/install/go2_robot_sdk/lib/python3.8/site-packages/go2_robot_sdk/go2_driver_node.py", line 314, in run
[go2_driver_node-6]     self.joy_cmd()
[go2_driver_node-6]   File "/home/test/ros2_ws/install/go2_robot_sdk/lib/python3.8/site-packages/go2_robot_sdk/go2_driver_node.py", line 131, in joy_cmd
[go2_driver_node-6]     self.conn.data_channel.send(self.robot_cmd_vel)
[go2_driver_node-6]   File "/home/test/.local/lib/python3.8/site-packages/aiortc/rtcdatachannel.py", line 182, in send
[go2_driver_node-6]     raise InvalidStateError
[go2_driver_node-6] aiortc.exceptions.InvalidStateError
[go2_driver_node-6] Exception in thread Thread-1:
[go2_driver_node-6] Traceback (most recent call last):
[go2_driver_node-6]   File "/opt/ros/humble/lib/python3.8/site-packages/rclpy/__init__.py", line 170, in spin_once
[go2_driver_node-6]     executor.add_node(node)
[go2_driver_node-6]   File "/opt/ros/humble/lib/python3.8/site-packages/rclpy/executors.py", line 254, in add_node
[go2_driver_node-6]     self._guard.trigger()
[go2_driver_node-6] AttributeError: 'NoneType' object has no attribute 'trigger'
[go2_driver_node-6] 
[go2_driver_node-6] During handling of the above exception, another exception occurred:
[go2_driver_node-6] 
[go2_driver_node-6] Traceback (most recent call last):
[go2_driver_node-6]   File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
[go2_driver_node-6]     self.run()
[go2_driver_node-6]   File "/usr/lib/python3.8/threading.py", line 870, in run
[go2_driver_node-6]     self._target(*self._args, **self._kwargs)
[go2_driver_node-6]   File "/home/test/ros2_ws/install/go2_robot_sdk/lib/python3.8/site-packages/go2_robot_sdk/go2_driver_node.py", line 324, in _spin
[go2_driver_node-6]     rclpy.spin_once(node)
[go2_driver_node-6]   File "/opt/ros/humble/lib/python3.8/site-packages/rclpy/__init__.py", line 173, in spin_once
[go2_driver_node-6]     executor.remove_node(node)
[go2_driver_node-6]   File "/opt/ros/humble/lib/python3.8/site-packages/rclpy/executors.py", line 271, in remove_node
[go2_driver_node-6]     self._guard.trigger()
[go2_driver_node-6] AttributeError: 'NoneType' object has no attribute 'trigger'
[go2_driver_node-6] ERROR:asyncio:Task was destroyed but it is pending!
[go2_driver_node-6] task: <Task pending name='Task-3' coro=<spin() running at /home/test/ros2_ws/install/go2_robot_sdk/lib/python3.8/site-packages/go2_robot_sdk/go2_driver_node.py:332> wait_for=<Future pending cb=[<TaskWakeupMethWrapper object at 0x7e2a643c0340>()]>>
[INFO] [go2_driver_node-6]: process has finished cleanly [pid 94764]

Do you know what is the issue?

David9696, 2024-05-20:
Hi,

Just to update. Everything is working fine after i reinstall ubuntu and ros2. Would close  this issue.

Thanks.

closed 2024-05-20 by David9696