abizovnuralem/go2_ros2_sdk #38: Not responding: custom made keyboard teleop node

MayooranT, 2024-06-18:
Great work. This project has been really helpful to me lately.
I wanted to move the robot using WASD keys as follows.

```
import rclpy
from geometry_msgs.msg import Twist
from pynput import keyboard
import time

class KeyboardTeleopNode:
    def __init__(self):
        self.node = rclpy.create_node('keyboard_teleop')
        self.publisher = self.node.create_publisher(Twist, '/cmd_vel_joy', 10)
        
        self.cmd_vel_msg = Twist()
        print('initiated')

    def start(self):
        with keyboard.Listener(on_press=self.on_key_press, on_release=self.on_key_release) as listener:
            print("listener started")
            listener.join()

    def on_key_press(self, key):
        if hasattr(key, 'char'):
            if key.char == 'w':
                print("   w pressed")
                self.cmd_vel_msg.linear.x = 0.5
            elif key.char == 's':
                print("   s pressed")
                self.cmd_vel_msg.linear.x = -0.5
            elif key.char == 'a':
                print("   a pressed")
                self.cmd_vel_msg.angular.z = 0.5
            elif key.char == 'd':
                print("   d pressed")
                self.cmd_vel_msg.angular.z = -0.5

            self.publish_cmd_vel()

    def on_key_release(self, key):
        if hasattr(key, 'char'):
            if key.char in ['w', 's']:
                # time.sleep(1000)
                print('reset')
                self.cmd_vel_msg.linear.x = 0.0
            elif key.char in ['a', 'd']:
                # time.sleep(1000)
                print('reset')
                self.cmd_vel_msg.angular.z = 0.0

            self.publish_cmd_vel()

    def publish_cmd_vel(self):
        self.publisher.publish(self.cmd_vel_msg)
        print('published')

def main(args=None):
    rclpy.init(args=args)
    keyboard_teleop_node = KeyboardTeleopNode()
    keyboard_teleop_node.start()
    rclpy.spin(keyboard_teleop_node.node)
    keyboard_teleop_node.node.destroy_node()
    rclpy.shutdown()

if __name__ == '__main__':
    main()

```
It is publishing each time to the topic (checked from the printing statements). But, The robot moves only for the first command only and only to the move forward command "w". The following commands get stalled and Robot not responding. Do we have to try any async functions to make it work?
Thanks in advance.

abizovnuralem, 2024-06-18:
H!! Should be async + make sure that move cmd doesn't overwrite with other commands. I will debug it more and see what causes the issue.

MayooranT, 2024-06-23:
Thanks for the reply. I just connected to AP mode and used the SnowyAP for running the same code and it worked! The robot responds for w and a but not for s and d. Seems like it is not processing negative velocities.