abizovnuralem/go2_ros2_sdk #48: Could not Connecting to the GO2 Robot Via cyclonedds

David9696, 2024-07-04:
Hi,

Thanks for updating the package.

I have tried to connect my Go2 robot via cyclonedds but with no success. I can only see the camera view being published with cyclonedds. No other topics. There is no error when running:

ros2 launch go2_robot_sdk robot.launch.py

The output is listed as follows:

david@david-go2:~$ ros2 launch go2_robot_sdk robot.launch.py
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [5834]
[INFO] [ros2_go2_video-2]: process started with pid [5836]
[INFO] [pointcloud_to_laserscan_node-3]: process started with pid [5838]
[INFO] [go2_driver_node-4]: process started with pid [5840]
[INFO] [lidar_to_pointcloud-5]: process started with pid [5860]
[INFO] [rviz2-6]: process started with pid [5865]
[INFO] [joy_node-7]: process started with pid [5867]
[INFO] [teleop_node-8]: process started with pid [5870]
[INFO] [twist_mux-9]: process started with pid [5879]
[robot_state_publisher-1] [INFO] [1720068339.296331065] [robot0.robot_state_publisher]: got segment map
[robot_state_publisher-1] [INFO] [1720068339.296445962] [robot0.robot_state_publisher]: got segment odom
[robot_state_publisher-1] [INFO] [1720068339.296524574] [robot0.robot_state_publisher]: got segment robot0/FL_calf
[robot_state_publisher-1] [INFO] [1720068339.296538993] [robot0.robot_state_publisher]: got segment robot0/FL_calflower
[robot_state_publisher-1] [INFO] [1720068339.296549142] [robot0.robot_state_publisher]: got segment robot0/FL_calflower1
[robot_state_publisher-1] [INFO] [1720068339.296558762] [robot0.robot_state_publisher]: got segment robot0/FL_foot
[robot_state_publisher-1] [INFO] [1720068339.296567882] [robot0.robot_state_publisher]: got segment robot0/FL_hip
[robot_state_publisher-1] [INFO] [1720068339.296576725] [robot0.robot_state_publisher]: got segment robot0/FL_thigh
[robot_state_publisher-1] [INFO] [1720068339.296585755] [robot0.robot_state_publisher]: got segment robot0/FR_calf
[robot_state_publisher-1] [INFO] [1720068339.296594143] [robot0.robot_state_publisher]: got segment robot0/FR_calflower
[robot_state_publisher-1] [INFO] [1720068339.296602433] [robot0.robot_state_publisher]: got segment robot0/FR_calflower1
[robot_state_publisher-1] [INFO] [1720068339.296623263] [robot0.robot_state_publisher]: got segment robot0/FR_foot
[robot_state_publisher-1] [INFO] [1720068339.296631400] [robot0.robot_state_publisher]: got segment robot0/FR_hip
[robot_state_publisher-1] [INFO] [1720068339.296639421] [robot0.robot_state_publisher]: got segment robot0/FR_thigh
[robot_state_publisher-1] [INFO] [1720068339.296647326] [robot0.robot_state_publisher]: got segment robot0/Head_lower
[robot_state_publisher-1] [INFO] [1720068339.296655415] [robot0.robot_state_publisher]: got segment robot0/Head_upper
[robot_state_publisher-1] [INFO] [1720068339.296663931] [robot0.robot_state_publisher]: got segment robot0/RL_calf
[robot_state_publisher-1] [INFO] [1720068339.296671725] [robot0.robot_state_publisher]: got segment robot0/RL_calflower
[robot_state_publisher-1] [INFO] [1720068339.296679947] [robot0.robot_state_publisher]: got segment robot0/RL_calflower1
[robot_state_publisher-1] [INFO] [1720068339.296687808] [robot0.robot_state_publisher]: got segment robot0/RL_foot
[robot_state_publisher-1] [INFO] [1720068339.296695256] [robot0.robot_state_publisher]: got segment robot0/RL_hip
[robot_state_publisher-1] [INFO] [1720068339.296703062] [robot0.robot_state_publisher]: got segment robot0/RL_thigh
[robot_state_publisher-1] [INFO] [1720068339.296711120] [robot0.robot_state_publisher]: got segment robot0/RR_calf
[robot_state_publisher-1] [INFO] [1720068339.296718766] [robot0.robot_state_publisher]: got segment robot0/RR_calflower
[robot_state_publisher-1] [INFO] [1720068339.296727322] [robot0.robot_state_publisher]: got segment robot0/RR_calflower1
[robot_state_publisher-1] [INFO] [1720068339.296735141] [robot0.robot_state_publisher]: got segment robot0/RR_foot
[robot_state_publisher-1] [INFO] [1720068339.296743170] [robot0.robot_state_publisher]: got segment robot0/RR_hip
[robot_state_publisher-1] [INFO] [1720068339.296761856] [robot0.robot_state_publisher]: got segment robot0/RR_thigh
[robot_state_publisher-1] [INFO] [1720068339.296771826] [robot0.robot_state_publisher]: got segment robot0/base_footprint
[robot_state_publisher-1] [INFO] [1720068339.296779882] [robot0.robot_state_publisher]: got segment robot0/base_link
[robot_state_publisher-1] [INFO] [1720068339.296787792] [robot0.robot_state_publisher]: got segment robot0/front_camera
[robot_state_publisher-1] [INFO] [1720068339.296795497] [robot0.robot_state_publisher]: got segment robot0/imu
[robot_state_publisher-1] [INFO] [1720068339.296803871] [robot0.robot_state_publisher]: got segment robot0/radar
[ros2_go2_video-2] [INFO] [1720068339.374542638] [go2_video]: Starting go2_video
[teleop_node-8] [INFO] [1720068339.377911188] [TeleopTwistJoy]: Linear axis x on 1 at scale 0.400000.
[teleop_node-8] [INFO] [1720068339.377983334] [TeleopTwistJoy]: Linear axis y on 0 at scale 0.400000.
[teleop_node-8] [INFO] [1720068339.377991050] [TeleopTwistJoy]: Linear axis z on 3 at scale 1.000000.
[teleop_node-8] [INFO] [1720068339.377995642] [TeleopTwistJoy]: Angular axis yaw on 2 at scale 0.500000.
[twist_mux-9] [INFO] [1720068339.381227858] [twist_mux]: Topic handler 'topics.joy' subscribed to topic 'cmd_vel_joy': timeout = 0.500000s , priority = 10.
[twist_mux-9] [INFO] [1720068339.381884912] [twist_mux]: Topic handler 'topics.navigation' subscribed to topic 'cmd_vel': timeout = 0.500000s , priority = 15.
[ros2_go2_video-2] [INFO] [1720068339.382297223] [go2_video]: Connecting to the robot's video stream
[ros2_go2_video-2] Logger is already initialized
[rviz2-6] [INFO] [1720068339.846642925] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-6] [INFO] [1720068339.846850329] [rviz2]: OpenGl version: 4.5 (GLSL 4.5)
[rviz2-6] [INFO] [1720068339.947892180] [rviz2]: Stereo is NOT SUPPORTED
[go2_driver_node-4] [INFO] [1720068340.569747019] [go2_driver_node]: Received ip list: ['192.168.123.161']
[go2_driver_node-4] [INFO] [1720068340.570313149] [go2_driver_node]: Connection type is cyclonedds
[rviz2-6] [ERROR] [1720068340.695827079] [rviz2]: The link is static or has unrealistic inertia, so the equivalent inertia box will not be shown.
[rviz2-6] 
[rviz2-6] [ERROR] [1720068340.695973718] [rviz2]: The link is static or has unrealistic inertia, so the equivalent inertia box will not be shown.
[rviz2-6] 
[rviz2-6] [INFO] [1720068341.074495690] [rviz2]: Stereo is NOT SUPPORTED
[lidar_to_pointcloud-5] /opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/node.py:441: UserWarning: when declaring parameter named 'robot_ip_lst', declaring a parameter only providing its name is deprecated. You have to either:
[lidar_to_pointcloud-5] 	- Pass a name and a default value different to "PARAMETER NOT SET" (and optionally a descriptor).
[lidar_to_pointcloud-5] 	- Pass a name and a parameter type.
[lidar_to_pointcloud-5] 	- Pass a name and a descriptor with `dynamic_typing=True
[lidar_to_pointcloud-5]   warnings.warn(
[lidar_to_pointcloud-5] /opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/node.py:441: UserWarning: when declaring parameter named 'map_name', declaring a parameter only providing its name is deprecated. You have to either:
[lidar_to_pointcloud-5] 	- Pass a name and a default value different to "PARAMETER NOT SET" (and optionally a descriptor).
[lidar_to_pointcloud-5] 	- Pass a name and a parameter type.
[lidar_to_pointcloud-5] 	- Pass a name and a descriptor with `dynamic_typing=True
[lidar_to_pointcloud-5]   warnings.warn(
[lidar_to_pointcloud-5] /opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/node.py:441: UserWarning: when declaring parameter named 'map_save', declaring a parameter only providing its name is deprecated. You have to either:
[lidar_to_pointcloud-5] 	- Pass a name and a default value different to "PARAMETER NOT SET" (and optionally a descriptor).
[lidar_to_pointcloud-5] 	- Pass a name and a parameter type.
[lidar_to_pointcloud-5] 	- Pass a name and a descriptor with `dynamic_typing=True
[lidar_to_pointcloud-5]   warnings.warn(
[ros2_go2_video-2] [INFO] [1720068345.126848027] [go2_video]: Input stream size: (1280, 720)
[ros2_go2_video-2] [INFO] [1720068345.126939570] [go2_video]: Output stream size (1280, 720)

Rviz shown incomplete robot model:

[image]

I have set the following command in bashrc too: 

export CONN_TYPE="cyclonedds"

Did I missing any setting? 

Btw, the webrtc mode is working as per normal. I can access all the topics that published from the robot.

Hope you could enlighten me.

Thanks.

David

abizovnuralem, 2024-07-04:
Hello, the ip for the cyclonesDDS and webrtc are different, In your case I think you are using the wrong IP. Another possible issue is the cycloneDDS,  please check ros2 topic list if you don't have it reboot the pc.

David9696, 2024-07-05:
Hi,

Thanks for your reply. Yes, for cyclonesDDS and webrtc I used different IP as one of them is connected via Ethernet and the other one is connected via wifi.

For cyclonesDDS, I used the following IP:

192.168.123.161

which is the default go2 pro IP. Is it correct?

With cyclonesDDS, I can see all the ros2 topic. But only camera topic is publishing data. The rest are empty.

Is cyclonesDDS only supported for edu version?

Thanks.

abizovnuralem, 2024-07-05:
cyclonesDDS is only supported for edu version

David9696, 2024-07-06:
Well noted. Thanks.

closed 2024-07-06 by David9696