abizovnuralem/go2_ros2_sdk #52: Cannot control go2 when the second time.

Lin-jun-xiang, 2024-07-10:
I extracted the `WebRTC` part for controlling `Go2` (**without ROS**) but encountered the following issues:

### webrtc_driver.py

```python

import base64
import hashlib
import json
import logging
import time

import aiohttp
from aiortc import RTCPeerConnection, RTCSessionDescription
from aiortc.contrib.media import MediaBlackhole

logging.basicConfig(level=logging.INFO)
logger = logging.getLogger(__name__)

class Go2Connection():
    def __init__(self, robot_ip, robot_num, token="", on_validated=None, on_message=None, on_open=None):
        self.pc = RTCPeerConnection()
        self.robot_ip = robot_ip
        self.robot_num = str(robot_num)
        self.token = token
        self.robot_validation = "PENDING"
        self.on_validated = on_validated
        self.on_message = on_message
        self.on_open = on_open
        
        self.audio_track = MediaBlackhole()
        self.video_track = MediaBlackhole()
        
        self.data_channel = self.pc.createDataChannel("data", id=0)
        self.data_channel.on("open", self.on_data_channel_open)
        self.data_channel.on("message", self.on_data_channel_message)
        
        self.pc.on("track", self.on_track)
        self.pc.on("connectionstatechange", self.on_connection_state_change)

    def on_connection_state_change(self):
        logger.info(f"Connection state is {self.pc.connectionState}")

    def on_track(self, track):
        logger.info(f"Receiving {track.kind}")

    async def generate_offer(self):
        await self.audio_track.start()
        await self.video_track.start()
        offer = await self.pc.createOffer()
        await self.pc.setLocalDescription(offer)
        return offer.sdp
    
    async def set_answer(self, sdp):
        answer = RTCSessionDescription(sdp, type="answer")
        await self.pc.setRemoteDescription(answer)
    
    def on_data_channel_open(self):
        logger.info("Data channel is open")
        if self.on_open:
            self.on_open()
    
    def on_data_channel_message(self, msg):
        logger.info(f"Received message: {msg}")

        if self.data_channel.readyState != "open":
            self.data_channel._setReadyState("open")

        try:
            if isinstance(msg, str):
[... 310 lines cut ...]
    if conn.robot_validation != "SUCCESS":
        logger.error("Failed to validate robot connection")
        return

    logger.info("Robot connection validated successfully")

    cmd = gen_command(1029)
    conn.data_channel.send(cmd)

    await asyncio.sleep(10)

    cmd = gen_command(1029)
    conn.data_channel.send(cmd)

    await asyncio.sleep(5)

    await conn.pc.close()
    logger.info("Disconnected from robot to stop movement")

if __name__ == "__main__":
    asyncio.run(main())

```

What might be causing this issue? How can I ensure that different functions are called correctly without placing all commands in the same coroutine?

Lin-jun-xiang, 2024-07-11:
Maybe the data send not correctly complete event loop in first time call fun?

abizovnuralem, 2024-07-11:
Hi, there is no problem controlling the dog in real-time with the joystick from PC, something is wrong with the data that you are sending to it.

abizovnuralem, 2024-07-11:
Also if you don't provide a conn token to fun1 and fun2 , you only will have 1 connection running at the same time.

Lin-jun-xiang, 2024-07-11:
@abizovnuralem 

Thanks your reply.

Is the `conn token` can be any like the following?

```python
conn = Go2Connection(
    robot_ip=ROBOT_IP,
    robot_num="0",
    token="Go2",
    on_open=lambda: logger.info("Data channel opened"),
    on_validated=lambda _: logger.info("Robot validated"),
    on_message=lambda msg, _, __: logger.info(f"Message received: {msg}")
)

async def go2_scrape():
    conn.data_channel.send(
        gen_command(ROBOT_CMD['Scrape'])
    )
```

abizovnuralem, 2024-07-11:
Obtaining security token
Connecting to your device without a security token is possible and might allow a connection to be established. However, this method limits you to a single active connection at any time. To simultaneously use multiple clients, such as a WebRTC-based application and a phone app, a valid security token is necessary. This ensures secure, multi-client access to your device.

One way is to sniff the traffic between the dog and your phone. Assuming that you have Linux or Mac:

Run tinyproxy or any other HTTP proxy on your computer
Set your computer's IP and port as HTTP proxy on your phone
Run wireshark or ngrep on your box sniffing port 8081 like ngrep port 8081.
Look for the token in the TCP stream after you connect your phone to the dog via the app
The token looks like this in the request payload:

{
    "token": "eyJ0eXAiOizI1NiJtlbiI[..]CI6MTcwODAxMzUwOX0.hiWOd9tNCIPzOOLNA",
    "sdp": "v=0\r\no=- ",
    "id": "STA_localNetwork",
    "type": "offer"
}
Another option is to obtain token via the /login/email endpoint.

curl -vX POST https://global-robot-api.unitree.com/login/email -d "email=<EMAIL>&password=<MD5 hash of your password>"

closed 2024-07-12 by Lin-jun-xiang