abizovnuralem/go2_ros2_sdk #53: how to imcrease the frame of the pointclouds

Zedzou, 2024-07-11:
Can I increase the frame rate by reducing the number of point clouds.

abizovnuralem, 2024-07-11:
Are you using webrtc?

Zedzou, 2024-07-11:
Yes, I use your feat_add_camino of the go2_ros_sdk project, and the frame rate of the output pointcloud is just about 2.

abizovnuralem, 2024-07-12:
webrtc provide the fix about of points (also filtered), we cannot change the value

Zedzou, 2024-07-13:
Thank you for your answer. Because the frame rate of the pointclouds from the lidar is too low, the go2 robot always run into an obstacle.  For another question. why the go2 can’t go to the target point that behind it, when I give an target point that behind the go2, It seems go2 can’t planner the path by the local_planner, but after I turn around the go2 to towards to the target point, It also can plan the path. Thank you .

---- Replied Message ----
| From | ***@***.***> |
| Date | 07/12/2024 13:31 |
| To | abizovnuralem/go2_ros2_sdk ***@***.***> |
| Cc | Zedzou ***@***.***>,
Author ***@***.***> |
| Subject | Re: [abizovnuralem/go2_ros2_sdk] how to imcrease the frame of the pointclouds (Issue #53) |

webrtc provide the fix about of points (also filtered), we cannot change the value

abizovnuralem, 2024-07-27:
working on fixing the nav2 config, will update the repo soon

closed 2024-08-03 by abizovnuralem