abizovnuralem/go2_ros2_sdk #55: How to move by python client?

Lin-jun-xiang, 2024-07-12:
No matter how I set the speed, the distance moved is same and a little bit.

```python
conn = Go2Connection(
    robot_ip=ROBOT_IP,
    robot_num="0",
    token="token",
    on_open=lambda: logger.info("Data channel opened"),
    on_validated=lambda _: logger.info("Robot validated"),
    on_message=lambda msg, _, __: logger.info(f"Message received: {msg}")
)

# after connected...
cmd = gen_command(cmd=1008, x=1.0, y=0., z=0.)
conn.data_channel.send(cmd)
time.sleep(5)

cmd = gen_command(cmd=1008, x=3.0, y=0., z=0.)
conn.data_channel.send(cmd)
time.sleep(5)
```

abizovnuralem, 2024-07-12:
gen_command has some filter. You need to do some change on it/

Lin-jun-xiang, 2024-07-15:
@abizovnuralem 

This is my define `gen_command`, which will send like `gen_mov_command` in this project.

```python
def gen_command(
    cmd: int,
    topic: str = "rt/api/sport/request",
    **kwarg
):
    parameters = cmd if not kwarg else kwarg

    command = {
                "type": "msg", 
                "topic": topic, 
                "data": {
                    "header":
                        {
                            "identity":
                                {
                                    "id": generate_id(), 
                                    "api_id": cmd
                                }
                        },
                    "parameter": json.dumps(parameters)          
                    }
                }
    command = json.dumps(command)
    return command

def gen_mov_command(x: float, y: float, z: float):
    command = {
        "type": "msg", 
        "topic": "rt/api/sport/request", 
        "data": {
            "header":
                {
                    "identity":
                        {
                            "id": generate_id(), 
                            "api_id": 1008
                        }
                },
            "parameter": json.dumps(
                {
                    "x": x, 
                    "y": y, 
                    "z": z
                })
                
            }
        }
    command = json.dumps(command)
    return command
```

Lin-jun-xiang, 2024-07-18:
@abizovnuralem 

I have seen the [example](https://support.unitree.com/home/zh/developer/Creating_customer_applications) from official document:

```c++
#include <unitree/robot/client/client.hpp>
#include <unitree/robot/go2/sport/sport_client.hpp>

int main(int argc, char **argv)
{
  if (argc < 2)
  {
    std::cout << "Usage: " << argv[0] << " networkInterface" << std::endl;
    exit(-1);
  }
  //初始化接口
  unitree::robot::ChannelFactory::Instance()->Init(0, argv[1]);
  
  //实例化sport_client并初始化
  unitree::robot::go2::SportClient sport_client;
  sport_client.SetTimeout(10.0f);//超时时间
  sport_client.Init();
  sport_client.WaitLeaseApplied();
  int moveTime = 3; // move 3 second
  auto startTime = std::chrono::steady_clock::now();
  while(true)
  { 
    
    sport_client.Move(0.2,0,0); //设置X方向上期望速度，控制Go2机器人向前走一段距离
    sport_client.BodyHeight(-0.07); //降低身体高度
    
    // 获取当前时间点
    auto currentTime = std::chrono::steady_clock::now();
    // 计算经过的时间
    auto elapsedTime = std::chrono::duration_cast<std::chrono::seconds>(currentTime - startTime);
    // 如果经过的时间大于等于两秒，退出循环
    if (elapsedTime.count() >= moveTime) 
    {
        break;
    }
  }
  
    sleep(2); 
    sport_client.BodyHeight(0); // 恢复默认高度
    return 0;
}
```

However, if I use a `while` loop to control go2, the `data_channel` will get a lot of message.

After the while loop completing, the go2 will **still move until kill the running thread. (In my case, the thread should be keep.)**

```python
def gen_command(
    cmd: int,
    topic: str = "rt/api/sport/request",
    **kwarg
):
    parameters = cmd if not kwarg else kwarg

    command = {
                "type": "msg", 
                "topic": topic, 
                "data": {
                    "header":
                        {
                            "identity":
                                {
                                    "id": generate_id(), 
                                    "api_id": cmd
                                }
                        },
                    "parameter": json.dumps(parameters)          
                    }
                }
    command = json.dumps(command)
    return command

start_time = time.time()
move_time = 2

while True:
    conn.data_channel.send(gen_command(1008, x=-0.2, y=0, z=0))

    current_time = time.time()

    if (current_time - start_time) >= move_time:
        conn.data_channel.send(gen_command(1008, x=0., y=0, z=0))
        conn.data_channel.send(gen_command(1003)) # StopMove
        print('Stop Move')
        break

```