abizovnuralem/go2_ros2_sdk #57: Install experience on 22.04 Numpy version issues and MissingCodecParameters

jliphard, 2024-07-22:
This is clean install, going through the `README.md` line by line.

* `colcon build` gives: `1 package had stderr output: ros2_go2_video`. Using `VERBOSE=1 colcon build --event-handlers console_direct+` does not give any more useful information. 
* The `lidar_to_pointcloud`  dies dues to Numpy issue, specifically, `_ARRAY_API` not found, which is due to Numpy being at `2.0.1`. **FIX**: `pip install numpy==1.26.4`
* The `ros2_go2_video` does not work and complains about `src/main.rs:107:62 - failed to create decoder: MissingCodecParameters.` 

After the `pip install numpy==1.26.4`, the lidar point cloud works (beautifully), but the video remains awol.

jliphard, 2024-07-22:
Well, I terminated everything, brought it back up and the video just started, and the `MissingCodecParameters` is now gone.

abizovnuralem, 2024-07-22:
Thanks @jliphard for your hard work on this project!

closed 2024-07-27 by abizovnuralem