abizovnuralem/go2_ros2_sdk #58: Cannot use the current package with a GO2 Air

roboticswithjulia, 2024-07-22:
Hello,
I would like to use the current package with the GO2 Air, but as far as I can see, when I am trying to retrieve the topics I get nothing.
I tried Wi-fi and also Ethernet. 
I get this topics:
```
/clicked_point
/cmd_vel_joy
/go2_camera/color/image
/goal_pose
/initialpose
/joy
/parameter_events
/robot0/point_cloud2
/robot0/robot_description
/rosout
/scan
/tf
/tf_static
```
But ros2 topic echo does not retrieve any data of them.

Is there anything else I could try just to make this package work?

Thank you big.

Júlia

jliphard, 2024-07-22:
This is working perfectly on my Go2 Air. The `roboverse` discord should be able to help you.

abizovnuralem, 2024-07-23:
Hello Júlia, can you give me more details on your error? Is it via webrtc? what steps have you done?

BenCaunt, 2024-07-24:
I had similar behavior to this with my Go2 pro.  The issue I was having was that my Go2 app was still connected to the robot.  With the way the WebRTC authentication is implemented only one client can be connected as once, thus you need to disconnect the app before connecting with ROS.  

My revised workflow is as follows:

1. Use the Unitree Go2 app diagnostics section to get the IP of the robot. 
2. Disconnect the app from the Go2. On IOS this was as simple as swiping up to close the app from running in the background.  Just turning off the screen was not sufficient. 
3. Run the launch file.  
4. Check if you are properly connected by a) looking at the launch file log or b) `ros2 topic list` and seeing if `/robot0/odom` is in the list, if so you are good to go.

closed 2024-07-27 by abizovnuralem

roboticswithjulia, 2024-07-27:
Hello all, thank you big for your help! I think that the issue here is that first I have to fulfill the steps of roboverse.  I did not have done any change in my robot yet, feel sorry about it I am quite new on this. I will try to fulfill the roboverse steps and if I have any doubt I will let you know for sure. Thank you!