abizovnuralem/go2_ros2_sdk #67: how to connect by cyclonedds

umsh315, 2024-08-01:
Hello,

First, I would like to apologize for my limited knowledge in English, robotics, and communication.

I attempted to use the information from GitHub to output point cloud data, monitor joints in real-time, and get camera information from the Unitree Go2. I couldn't figure out how to set up with WebRTC, so I tried setting CONN_TYPE to cyclonedds. The output was as follows.

Previously, when I failed, the parts that were not connected to the robot's body were displayed at the origin. However, this time, nothing was displayed, so I thought I might be getting closer to a solution and decided to ask for help.

Thank you for your assistance.

/////////////////////////////////////////////////////////////////////////////////
ros2 launch go2_robot_sdk robot.launch.py
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2024-08-01-00-19-27-200812-4776c52403b8-23907
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [23912]
[INFO] [ros2_go2_video-2]: process started with pid [23914]
[INFO] [pointcloud_to_laserscan_node-3]: process started with pid [23916]
[INFO] [go2_driver_node-4]: process started with pid [23918]
[INFO] [lidar_to_pointcloud-5]: process started with pid [23920]
[INFO] [rviz2-6]: process started with pid [23922]
[INFO] [joy_node-7]: process started with pid [23924]
[INFO] [teleop_node-8]: process started with pid [23926]
[INFO] [twist_mux-9]: process started with pid [23928]
[INFO] [foxglove_bridge-10]: process started with pid [23930]
[INFO] [async_slam_toolbox_node-11]: process started with pid [23932]
[INFO] [controller_server-12]: process started with pid [23934]
[INFO] [smoother_server-13]: process started with pid [23936]
[INFO] [planner_server-14]: process started with pid [23938]
[INFO] [behavior_server-15]: process started with pid [23940]
[INFO] [bt_navigator-16]: process started with pid [23942]
[INFO] [waypoint_follower-17]: process started with pid [23944]
[INFO] [velocity_smoother-18]: process started with pid [23946]
[INFO] [lifecycle_manager-19]: process started with pid [23948]
[twist_mux-9] [INFO] [1722471567.889998956] [twist_mux]: Topic handler 'topics.joy' subscribed to topic 'cmd_vel_joy': timeout = 0.500000s , priority = 10.
[twist_mux-9] [INFO] [1722471567.899245857] [twist_mux]: Topic handler 'topics.navigation' subscribed to topic 'cmd_vel': timeout = 0.500000s , priority = 5.
[async_slam_toolbox_node-11] [INFO] [1722471567.906760698] [slam_toolbox]: Node using stack size 40000000
[lifecycle_manager-19] [INFO] [1722471567.909006597] [lifecycle_manager_navigation]: Creating
[smoother_server-13] [INFO] [1722471567.916207431] [smoother_server]:
[behavior_server-15] [INFO] [1722471567.918183681] [behavior_server]:
[behavior_server-15]    behavior_server lifecycle node launched.
[behavior_server-15]    Waiting on external lifecycle transitions to activate
[behavior_server-15]    See https://design.ros2.org/articles/node_lifecycle.html for more information.
[planner_server-14] [INFO] [1722471567.920238783] [planner_server]:
[planner_server-14]     planner_server lifecycle node launched.
[planner_server-14]     Waiting on external lifecycle transitions to activate
[planner_server-14]     See https://design.ros2.org/articles/node_lifecycle.html for more information.
[smoother_server-13]    smoother_server lifecycle node launched.
[smoother_server-13]    Waiting on external lifecycle transitions to activate
[teleop_node-8] [INFO] [1722471567.924247185] [TeleopTwistJoy]: Linear axis x on 1 at scale 0.400000.
[smoother_server-13]    See https://design.ros2.org/articles/node_lifecycle.html for more information.
[teleop_node-8] [INFO] [1722471567.924383739] [TeleopTwistJoy]: Linear axis y on 0 at scale 0.400000.
[teleop_node-8] [INFO] [1722471567.927615105] [TeleopTwistJoy]: Angular axis yaw on 3 at scale 1.500000.
[velocity_smoother-18] [INFO] [1722471567.929006745] [velocity_smoother]:
[velocity_smoother-18]  velocity_smoother lifecycle node launched.
[velocity_smoother-18]  Waiting on external lifecycle transitions to activate
[velocity_smoother-18]  See https://design.ros2.org/articles/node_lifecycle.html for more information.
[planner_server-14] [INFO] [1722471567.931138992] [planner_server]: Creating
[smoother_server-13] [INFO] [1722471567.934417261] [smoother_server]: Creating smoother server
[bt_navigator-16] [INFO] [1722471567.944186828] [bt_navigator]:
[bt_navigator-16]       bt_navigator lifecycle node launched.
[bt_navigator-16]       Waiting on external lifecycle transitions to activate
[bt_navigator-16]       See https://design.ros2.org/articles/node_lifecycle.html for more information.
[bt_navigator-16] [INFO] [1722471567.944762127] [bt_navigator]: Creating
[lifecycle_manager-19] [INFO] [1722471567.949884958] [lifecycle_manager_navigation]: Creating and initializing lifecycle service clients
[controller_server-12] [INFO] [1722471567.955249945] [controller_server]:
[controller_server-12]  controller_server lifecycle node launched.
[controller_server-12]  Waiting on external lifecycle transitions to activate
[controller_server-12]  See https://design.ros2.org/articles/node_lifecycle.html for more information.
[waypoint_follower-17] [INFO] [1722471567.960888492] [waypoint_follower]:
[waypoint_follower-17]  waypoint_follower lifecycle node launched.
[... 429 lines cut ...]
[controller_server-12] [INFO] [1722471694.167822128] [local_costmap.local_costmap]: Destroying
[controller_server-12] [ERROR] [1722471694.167945925] [local_costmap.local_costmap]: Unable to start transition 5 from current state shuttingdown: Could not publish transition: publisher's context is invalid, at ./src/rcl/publisher.c:389, at ./src/rcl_lifecycle.c:368
[controller_server-12] [WARN] [1722471694.167975954] [rclcpp_lifecycle]: Shutdown error in destruction of LifecycleNode: final state(shuttingdown)
[controller_server-12] [INFO] [1722471694.172811750] [controller_server]: Destroying
[controller_server-12] [ERROR] [1722471694.172852381] [controller_server]: Unable to start transition 5 from current state shuttingdown: Could not publish transition: publisher's context is invalid, at ./src/rcl/publisher.c:389, at ./src/rcl_lifecycle.c:368
[controller_server-12] [WARN] [1722471694.172863453] [rclcpp_lifecycle]: Shutdown error in destruction of LifecycleNode: final state(shuttingdown)
[INFO] [controller_server-12]: process has finished cleanly [pid 23934]
[planner_server-14] [INFO] [1722471694.375506102] [global_costmap.global_costmap]: Cleaning up
[planner_server-14] [INFO] [1722471694.400415911] [global_costmap.global_costmap]: Destroying bond (global_costmap) to lifecycle manager.
[planner_server-14] [INFO] [1722471694.400484889] [planner_server]: Running Nav2 LifecycleNode rcl preshutdown (planner_server)
[planner_server-14] [INFO] [1722471694.400500239] [planner_server]: Deactivating
[planner_server-14] [INFO] [1722471694.400509699] [planner_server]: Deactivating plugin GridBased of type NavfnPlanner
[planner_server-14] [INFO] [1722471694.400514780] [planner_server]: Destroying bond (planner_server) to lifecycle manager.
[planner_server-14] [INFO] [1722471694.400545886] [planner_server]: Cleaning up
[planner_server-14] [INFO] [1722471694.402020097] [planner_server]: Cleaning up plugin GridBased of type NavfnPlanner
[planner_server-14] [INFO] [1722471694.407005933] [planner_server]: Destroying plugin GridBased of type NavfnPlanner
[planner_server-14] [INFO] [1722471694.435315943] [planner_server]: Destroying bond (planner_server) to lifecycle manager.
[planner_server-14] [INFO] [1722471694.436613822] [global_costmap.global_costmap]: Destroying
[planner_server-14] [ERROR] [1722471694.436657664] [global_costmap.global_costmap]: Unable to start transition 5 from current state shuttingdown: Could not publish transition: publisher's context is invalid, at ./src/rcl/publisher.c:389, at ./src/rcl_lifecycle.c:368
[planner_server-14] [WARN] [1722471694.436667917] [rclcpp_lifecycle]: Shutdown error in destruction of LifecycleNode: final state(shuttingdown)
[planner_server-14] [INFO] [1722471694.438078192] [planner_server]: Destroying
[planner_server-14] [ERROR] [1722471694.438089335] [planner_server]: Unable to start transition 5 from current state shuttingdown: Could not publish transition: publisher's context is invalid, at ./src/rcl/publisher.c:389, at ./src/rcl_lifecycle.c:368
[planner_server-14] [WARN] [1722471694.438093633] [rclcpp_lifecycle]: Shutdown error in destruction of LifecycleNode: final state(shuttingdown)
[INFO] [planner_server-14]: process has finished cleanly [pid 23938]
[image]

abizovnuralem, 2024-08-03:
chatGPT:
The logs show that the problem is related to incompatibility between the numpy and scipy library versions. This particular error occurs because numpy 2.0.1 is used, which is not compatible with the installed scipy version. In particular, it states that the scipy version requires numpy >= 1.17.3 and < 1.25.0.

umsh315, 2024-08-06:
Thank you. It seems that the connection is still not working. I apologize for the basic question, but do I need to attach a wireless LAN adapter to the Unitree Go2 for the WebRTC connection?

abizovnuralem, 2024-08-07:
No you do not need to do that, it should work out of the box

umsh315, 2024-08-07:
Thank you. So, if wlan0 is connected in the Data section of the Go2 app under Automatic Machine Inspection, does that mean we can connect by simply exporting the IP address of wlan0 and "webrtc" without any additional special operations?

umsh315, 2024-08-08:
No matter how many times I try, I keep seeing an image of a robot with disconnected legs in the initial position and "NO IMAGE" displayed. Could the Python version 3.10.13 be affecting this in some way?

umsh315, 2024-08-08:
[rviz2-6] [INFO] [1723082619.571413866] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-6] [INFO] [1723082619.571557597] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-6] [INFO] [1723082619.589620414] [rviz2]: Stereo is NOT SUPPORTED

Could this part be causing the issue? I apologize for asking multiple times.

abizovnuralem, 2024-08-11:
Can you send a full text after the launch, please?

umsh315, 2024-08-15:
ubuntu@B249112:~/ros2_ws$ colcon build
Starting >>> go2_interfaces
Starting >>> coco_detector
Starting >>> ros2_go2_video
Starting >>> unitree_go
Finished <<< coco_detector [0.93s]
--- stderr: ros2_go2_video
    Finished `colcon` profile [optimized] target(s) in 0.57s
---
Finished <<< ros2_go2_video [1.12s]
Finished <<< go2_interfaces [1.29s]
Starting >>> go2_robot_sdk
Finished <<< go2_robot_sdk [0.81s]
Finished <<< unitree_go [5.30s]

Summary: 5 packages finished [5.55s]
  1 package had stderr output: ros2_go2_video
ubuntu@B249112:~/ros2_ws$ source install/setup.bash
ubuntu@B249112:~/ros2_ws$ export ROBOT_IP="192.168.0.176"
ubuntu@B249112:~/ros2_ws$ export CONN_TYPE="webrtc"
ubuntu@B249112:~/ros2_ws$ ros2 launch go2_robot_sdk robot.launch.py
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2024-08-15-00-55-59-445061-B249112-8078
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [8082]
[INFO] [ros2_go2_video-2]: process started with pid [8084]
[INFO] [pointcloud_to_laserscan_node-3]: process started with pid [8086]
[INFO] [go2_driver_node-4]: process started with pid [8088]
[INFO] [lidar_to_pointcloud-5]: process started with pid [8090]
[INFO] [rviz2-6]: process started with pid [8092]
[INFO] [joy_node-7]: process started with pid [8094]
[INFO] [teleop_node-8]: process started with pid [8096]
[INFO] [twist_mux-9]: process started with pid [8098]
[INFO] [foxglove_bridge-10]: process started with pid [8100]
[INFO] [async_slam_toolbox_node-11]: process started with pid [8102]
[INFO] [controller_server-12]: process started with pid [8104]
[INFO] [smoother_server-13]: process started with pid [8106]
[INFO] [planner_server-14]: process started with pid [8108]
[INFO] [behavior_server-15]: process started with pid [8110]
[INFO] [bt_navigator-16]: process started with pid [8112]
[INFO] [waypoint_follower-17]: process started with pid [8114]
[INFO] [velocity_smoother-18]: process started with pid [8116]
[INFO] [lifecycle_manager-19]: process started with pid [8118]
[teleop_node-8] [INFO] [1723683360.087667837] [TeleopTwistJoy]: Linear axis x on 1 at scale 0.400000.
[velocity_smoother-18] [INFO] [1723683360.088649681] [velocity_smoother]:
[velocity_smoother-18]  velocity_smoother lifecycle node launched.
[velocity_smoother-18]  Waiting on external lifecycle transitions to activate
[velocity_smoother-18]  See https://design.ros2.org/articles/node_lifecycle.html for more information.
[smoother_server-13] [INFO] [1723683360.090585214] [smoother_server]:
[smoother_server-13]    smoother_server lifecycle node launched.
[smoother_server-13]    Waiting on external lifecycle transitions to activate
[smoother_server-13]    See https://design.ros2.org/articles/node_lifecycle.html for more information.
[teleop_node-8] [INFO] [1723683360.092452788] [TeleopTwistJoy]: Linear axis y on 0 at scale 0.400000.
[teleop_node-8] [INFO] [1723683360.092485283] [TeleopTwistJoy]: Angular axis yaw on 3 at scale 1.500000.
[robot_state_publisher-1] [INFO] [1723683360.098775070] [robot_state_publisher]: got segment FL_calf
[robot_state_publisher-1] [INFO] [1723683360.099266630] [robot_state_publisher]: got segment FL_calflower
[robot_state_publisher-1] [INFO] [1723683360.099288505] [robot_state_publisher]: got segment FL_calflower1
[robot_state_publisher-1] [INFO] [1723683360.099296885] [robot_state_publisher]: got segment FL_foot
[robot_state_publisher-1] [INFO] [1723683360.099305146] [robot_state_publisher]: got segment FL_hip
[robot_state_publisher-1] [INFO] [1723683360.099312277] [robot_state_publisher]: got segment FL_thigh
[robot_state_publisher-1] [INFO] [1723683360.099322949] [robot_state_publisher]: got segment FR_calf
[robot_state_publisher-1] [INFO] [1723683360.099329703] [robot_state_publisher]: got segment FR_calflower
[robot_state_publisher-1] [INFO] [1723683360.099338305] [robot_state_publisher]: got segment FR_calflower1
[robot_state_publisher-1] [INFO] [1723683360.099347397] [robot_state_publisher]: got segment FR_foot
[robot_state_publisher-1] [INFO] [1723683360.099360622] [robot_state_publisher]: got segment FR_hip
[robot_state_publisher-1] [INFO] [1723683360.099369192] [robot_state_publisher]: got segment FR_thigh
[robot_state_publisher-1] [INFO] [1723683360.099376765] [robot_state_publisher]: got segment Head_lower
[robot_state_publisher-1] [INFO] [1723683360.099382075] [robot_state_publisher]: got segment Head_upper
[robot_state_publisher-1] [INFO] [1723683360.099388856] [robot_state_publisher]: got segment RL_calf
[robot_state_publisher-1] [INFO] [1723683360.099395013] [robot_state_publisher]: got segment RL_calflower
[robot_state_publisher-1] [INFO] [1723683360.099401205] [robot_state_publisher]: got segment RL_calflower1
[... 375 lines cut ...]
[controller_server-12] [ERROR] [1723683404.187833551] [local_costmap.local_costmap]: Unable to start transition 5 from current state shuttingdown: Could not publish transition: publisher's context is invalid, at ./src/rcl/publisher.c:389, at ./src/rcl_lifecycle.c:368
[controller_server-12] [WARN] [1723683404.187855266] [rclcpp_lifecycle]: Shutdown error in destruction of LifecycleNode: final state(shuttingdown)
[controller_server-12] [INFO] [1723683404.191249258] [controller_server]: Destroying
[... cut ...]
[planner_server-14] [INFO] [1723683404.495391736] [planner_server]: Destroying
[planner_server-14] [ERROR] [1723683404.495412894] [planner_server]: Unable to start transition 5 from current state shuttingdown: Could not publish transition: publisher's context is invalid, at ./src/rcl/publisher.c:389, at ./src/rcl_lifecycle.c:368
[planner_server-14] [WARN] [1723683404.495418851] [rclcpp_lifecycle]: Shutdown error in destruction of LifecycleNode: final state(shuttingdown)
[INFO] [planner_server-14]: process has finished cleanly [pid 8108]
[image]

umsh315, 2024-08-15:
[image]

abizovnuralem, 2024-08-15:
It seems to me that node with video doest work, can you comment ros2_go2_video node inside launch and try again please?

umsh315, 2024-08-15:
```
# Copyright (c) 2024, RoboVerse community
#
# Redistribution and use in source and binary forms, with or without
# modification, are permitted provided that the following conditions are met:
#
# 1. Redistributions of source code must retain the above copyright notice, this
#    list of conditions and the following disclaimer.
#
# 2. Redistributions in binary form must reproduce the above copyright notice,
#    this list of conditions and the following disclaimer in the documentation
#    and/or other materials provided with the distribution.
#
# THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS"
# AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE
# IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE
# DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR CONTRIBUTORS BE LIABLE
# FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL
# DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR
# SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER
# CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY,
# OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE
# OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.

import os
from ament_index_python.packages import get_package_share_directory
from launch import LaunchDescription
from launch.conditions import UnlessCondition
from launch.substitutions import LaunchConfiguration
from launch_ros.actions import Node
from launch.actions import IncludeLaunchDescription
from launch.launch_description_sources import FrontendLaunchDescriptionSource, PythonLaunchDescriptionSource

def generate_launch_description():

    use_sim_time = LaunchConfiguration('use_sim_time', default='false')
    no_rviz2 = LaunchConfiguration('no_rviz2', default='false')

    robot_token = os.getenv('ROBOT_TOKEN', '')
    robot_ip = os.getenv('ROBOT_IP', '')
    robot_ip_lst = robot_ip.replace(" ", "").split(",")
    conn_mode = "single" if len(robot_ip_lst) == 1 else "multi"

    map_name = os.getenv('MAP_NAME', '3d_map')
    save_map = os.getenv('MAP_SAVE', "true")
    conn_type = os.getenv('CONN_TYPE', 'webrtc')

    if conn_mode == 'single':
        rviz_config = "single_robot_conf.rviz"
    else:
        rviz_config = "multi_robot_conf.rviz"

    if conn_type == 'cyclonedds':
        rviz_config = "cyclonedds_config.rviz"

    urdf_file_name = 'multi_go2.urdf'
    urdf = os.path.join(
        get_package_share_directory('go2_robot_sdk'),
        "urdf",
        urdf_file_name)
    with open(urdf, 'r') as infp:
        robot_desc = infp.read()

    robot_desc_modified_lst = []

    for i in range(len(robot_ip_lst)):
        robot_desc_modified_lst.append(
            robot_desc.format(robot_num=f"robot{i}"))

    urdf_launch_nodes = []
[... 169 lines cut ...]
            PythonLaunchDescriptionSource([
                os.path.join(get_package_share_directory(
                    'slam_toolbox'), 'launch', 'online_async_launch.py')
            ]),
            launch_arguments={
                'params_file': slam_toolbox_config,
                'use_sim_time': use_sim_time,
            }.items(),
        ),

        IncludeLaunchDescription(
            PythonLaunchDescriptionSource([
                os.path.join(get_package_share_directory(
                    'nav2_bringup'), 'launch', 'navigation_launch.py')
            ]),
            launch_arguments={
                'params_file': nav2_config,
                'use_sim_time': use_sim_time,
            }.items(),
        ),
    ])

```

Is this the correct way to comment out the node?

umsh315, 2024-08-15:
```
ubuntu@B249112:~/ros2_ws$ ros2 launch go2_robot_sdk robot.launch.py
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2024-08-15-05-57-45-728270-B249112-5715
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [5719]
[INFO] [pointcloud_to_laserscan_node-2]: process started with pid [5721]
[INFO] [go2_driver_node-3]: process started with pid [5723]
[INFO] [lidar_to_pointcloud-4]: process started with pid [5725]
[INFO] [rviz2-5]: process started with pid [5727]
[INFO] [joy_node-6]: process started with pid [5729]
[INFO] [teleop_node-7]: process started with pid [5731]
[INFO] [twist_mux-8]: process started with pid [5733]
[INFO] [foxglove_bridge-9]: process started with pid [5735]
[INFO] [async_slam_toolbox_node-10]: process started with pid [5737]
[INFO] [controller_server-11]: process started with pid [5739]
[INFO] [smoother_server-12]: process started with pid [5741]
[INFO] [planner_server-13]: process started with pid [5743]
[INFO] [behavior_server-14]: process started with pid [5745]
[INFO] [bt_navigator-15]: process started with pid [5747]
[INFO] [waypoint_follower-16]: process started with pid [5750]
[INFO] [velocity_smoother-17]: process started with pid [5752]
[INFO] [lifecycle_manager-18]: process started with pid [5754]
[twist_mux-8] [INFO] [1723701466.352287136] [twist_mux]: Topic handler 'topics.joy' subscribed to topic 'cmd_vel_joy': timeout = 0.500000s , priority = 10.
[lifecycle_manager-18] [INFO] [1723701466.357233682] [lifecycle_manager_navigation]: Creating
[twist_mux-8] [INFO] [1723701466.362529725] [twist_mux]: Topic handler 'topics.navigation' subscribed to topic 'cmd_vel': timeout = 0.500000s , priority = 5.
[lifecycle_manager-18] [INFO] [1723701466.367769063] [lifecycle_manager_navigation]: Creating and initializing lifecycle service clients
[planner_server-13] [INFO] [1723701466.368592153] [planner_server]:
[planner_server-13]     planner_server lifecycle node launched.
[planner_server-13]     Waiting on external lifecycle transitions to activate
[planner_server-13]     See https://design.ros2.org/articles/node_lifecycle.html for more information.
[waypoint_follower-16] [INFO] [1723701466.375231012] [waypoint_follower]:
[waypoint_follower-16]  waypoint_follower lifecycle node launched.
[waypoint_follower-16]  Waiting on external lifecycle transitions to activate
[waypoint_follower-16]  See https://design.ros2.org/articles/node_lifecycle.html for more information.
[waypoint_follower-16] [INFO] [1723701466.377255059] [waypoint_follower]: Creating
[smoother_server-12] [INFO] [1723701466.379160263] [smoother_server]:
[smoother_server-12]    smoother_server lifecycle node launched.
[smoother_server-12]    Waiting on external lifecycle transitions to activate
[smoother_server-12]    See https://design.ros2.org/articles/node_lifecycle.html for more information.
[planner_server-13] [INFO] [1723701466.379964817] [planner_server]: Creating
[smoother_server-12] [INFO] [1723701466.381181325] [smoother_server]: Creating smoother server
[async_slam_toolbox_node-10] [INFO] [1723701466.382994377] [slam_toolbox]: Node using stack size 40000000
[teleop_node-7] [INFO] [1723701466.384250354] [TeleopTwistJoy]: Linear axis x on 1 at scale 0.400000.
[teleop_node-7] [INFO] [1723701466.384345088] [TeleopTwistJoy]: Linear axis y on 0 at scale 0.400000.
[teleop_node-7] [INFO] [1723701466.384355494] [TeleopTwistJoy]: Angular axis yaw on 3 at scale 1.500000.
[controller_server-11] [INFO] [1723701466.386305407] [controller_server]:
[controller_server-11]  controller_server lifecycle node launched.
[controller_server-11]  Waiting on external lifecycle transitions to activate
[controller_server-11]  See https://design.ros2.org/articles/node_lifecycle.html for more information.
[robot_state_publisher-1] [INFO] [1723701466.389418134] [robot_state_publisher]: got segment FL_calf
[robot_state_publisher-1] [INFO] [1723701466.389637429] [robot_state_publisher]: got segment FL_calflower
[robot_state_publisher-1] [INFO] [1723701466.389651387] [robot_state_publisher]: got segment FL_calflower1
[robot_state_publisher-1] [INFO] [1723701466.389658495] [robot_state_publisher]: got segment FL_foot
[robot_state_publisher-1] [INFO] [1723701466.389664499] [robot_state_publisher]: got segment FL_hip
[robot_state_publisher-1] [INFO] [1723701466.389670268] [robot_state_publisher]: got segment FL_thigh
[robot_state_publisher-1] [INFO] [1723701466.389676941] [robot_state_publisher]: got segment FR_calf
[robot_state_publisher-1] [INFO] [1723701466.389683163] [robot_state_publisher]: got segment FR_calflower
[robot_state_publisher-1] [INFO] [1723701466.389688429] [robot_state_publisher]: got segment FR_calflower1
[robot_state_publisher-1] [INFO] [1723701466.389693424] [robot_state_publisher]: got segment FR_foot
[robot_state_publisher-1] [INFO] [1723701466.389698691] [robot_state_publisher]: got segment FR_hip
[robot_state_publisher-1] [INFO] [1723701466.389704046] [robot_state_publisher]: got segment FR_thigh
[robot_state_publisher-1] [INFO] [1723701466.389709284] [robot_state_publisher]: got segment Head_lower
[robot_state_publisher-1] [INFO] [1723701466.389714756] [robot_state_publisher]: got segment Head_upper
[robot_state_publisher-1] [INFO] [1723701466.389721255] [robot_state_publisher]: got segment RL_calf
[robot_state_publisher-1] [INFO] [1723701466.389725846] [robot_state_publisher]: got segment RL_calflower
[... cut ...]
[controller_server-11] [ERROR] [1723701483.447633738] [controller_server]: Unable to start transition 5 from current state shuttingdown: Could not publish transition: publisher's context is invalid, at ./src/rcl/publisher.c:389, at ./src/rcl_lifecycle.c:368
[controller_server-11] [WARN] [1723701483.447651665] [rclcpp_lifecycle]: Shutdown error in destruction of LifecycleNode: final state(shuttingdown)
[INFO] [controller_server-11]: process has finished cleanly [pid 5739]

```
I haven't checked everything, but the output screen in RViz is in the same condition as before.

abizovnuralem, 2024-08-15:
What ubuntu version are you using?

umsh315, 2024-08-15:
i use ubuntu 22.04 ros2 humble

closed 2024-09-05 by umsh315

S4b3, 2024-09-30:
Hi OP, since you marked the issue as completed, could you describe what the solution was? Thank you :)

umsh315, 2024-09-30:
@S4b3 
> Hi OP, since you marked the issue as completed, could you describe what the solution was? Thank you :)

Hello. I think the issue was that the IP address of wlan0 inside the Go2 was not specified, and at that time, the firmware version of the Go2 was the latest one, which the package did not support. Currently, it has already been addressed, and I am now able to use it myself.

S4b3, 2024-10-01:
> @S4b3
>
>
> Hello. I think the issue was that the IP address of wlan0 inside the Go2 was not specified, and at that time, the firmware version of the Go2 was the latest one, which the package did not [...]

Thank you for the reply! I assume then since I'm experiencing pretty much the same problem (everything works under webrtc but not under cyclonedds) the issue could be similar. We updated Go2's firmware pretty recently so that might be the reason. Thanks again, might not seem much but that helped me a lot :)