abizovnuralem/go2_ros2_sdk #73: MissingCodecParameters

chuanmew, 2024-09-04:
I did a clean install on Ubuntu 22.04 with humble and I got the same error mentioned in #57.
I followed the method there and installed numpy 1.26.4.  Everything works except this one:
The ros2_go2_video does not work and complains about src/main.rs:107:62 - failed to create decoder: MissingCodecParameters.
Reboot several times but video never starts.
I am using conda virtual environment. Not sure if that is the problem.

chuanmew, 2024-09-05:
I could see everything. And rust is 1.79
I think it wasn't a problem with the code but connection to the dog. I somehow made it work by manually restarting the service related to webrtc using Unitree app. Simply reboot the dog didn't work. Also after that it keeps working even after another reboot. 
I don't know if the app is using the same webrtc thing as this repo but the video in app worked even before I restart the service.

abizovnuralem, 2024-09-12:
@tfoldi Hey Tamas, it looks like something has changed in https://github.com/tfoldi/ros2_go2_video/tree/794d70b97b447f951b8c3cde8a5a76996c98a39e and we need to update your repo to support new webrtc method

tfoldi, 2024-09-21:
I took a different approach and used Legion's new method to take the video directly from Python. Please check  if #84  solves your problem

closed 2024-09-21 by tfoldi