abizovnuralem/go2_ros2_sdk #74: no more connection by webrtc

Peenut173, 2024-09-05:
Hello,

I have been using your package for my internship project to perform experiments which needed wireless connection. The thing is since the new firmware update, I am not able to use webrtc protocol anymore, is it normal? If you have a solution, I take it as I am quite stuck right now to finish my experiments.

Thanks in advance.

Regards,

Peenut173.

abizovnuralem, 2024-09-05:
Hello! It looks like the new version has a different way of communicating with the mobile app. Thanks for this issue, my advice is to roll back to 1.0.25 and dont use 1.1.1 firmware. In that case, your mobile app will not work, but the dog will.

peterhuisman, 2024-09-05:
Same here, just received my Go2 EDU with the latest firmware. Am trying this repo and cannot connect to the dog over webrtc.

abizovnuralem, 2024-09-05:
> Same here, just received my Go2 EDU with the latest firmware. Am trying this repo and cannot connect to the dog over webrtc.

You can use a wired connection with cycloneDDS over ethernet. It should work with new firmware.

Peenut173, 2024-09-05:
Thanks for your quick answers, do you have any idea on how to get back to previous firmware version?

abizovnuralem, 2024-09-05:
> Thanks for your quick answers, do you have any idea on how to get back to previous firmware version?

I haven't done it before, but I think this option should be inside a mobile app.

Peenut173, 2024-09-05:
Unfortunately no, Unitree said it was officially not doable 😭. Anyway I will try to find a way to fix this problem. Thank you! 😉

abizovnuralem, 2024-09-09:
Hello, @Peenut173 ! Can you help me check a new method for webrtc that works with 1.1.1? 

https://github.com/abizovnuralem/go2_ros2_sdk/tree/firmware_1_1_1_update

I would appreciate your help. I can't test it and debut it by myself because I am using old firmware. Thanks!

abizovnuralem, 2024-09-09:
@peterhuisman, I would also appreciate your help integrating the new webrtc connection to this repo.

Peenut173, 2024-09-09:
Hello @abizovnuralem ! Yes of course, I'll get back to you as soon as I have tested it!

peterhuisman, 2024-09-09:
> @peterhuisman, I would also appreciate your help integrating the new webrtc connection to this repo.

Tried the new src, still receive the same error...

`[controller_server-12] [INFO] [1725881490.048107599] [local_costmap.local_costmap]: Activating
[controller_server-12] [INFO] [1725881490.048124212] [local_costmap.local_costmap]: Checking transform
[controller_server-12] [INFO] [1725881490.048241444] [local_costmap.local_costmap]: start
[rviz2-6] [ERROR] [1725881490.118126521] [rviz2]: The link is static or has unrealistic inertia, so the equivalent inertia box will not be shown.
[rviz2-6] 
[rviz2-6] [ERROR] [1725881490.118251698] [rviz2]: The link is static or has unrealistic inertia, so the equivalent inertia box will not be shown.
[rviz2-6] 
[rviz2-6] [INFO] [1725881490.161899564] [rviz2]: Stereo is NOT SUPPORTED
[lidar_to_pointcloud-5] /usr/lib/python3/dist-packages/scipy/__init__.py:146: UserWarning: A NumPy version >=1.17.3 and <1.25.0 is required for this version of SciPy (detected version 1.26.4
[lidar_to_pointcloud-5]   warnings.warn(f"A NumPy version >={np_minversion} and <{np_maxversion}"
[go2_driver_node-4] [INFO] [1725881490.614899118] [go2_driver_node]: Received ip list: ['172.28.164.106']
[go2_driver_node-4] [INFO] [1725881490.615240558] [go2_driver_node]: Connection type is webrtc
[go2_driver_node-4] [INFO] [1725881490.615528621] [go2_driver_node]: Connection mode is single
[go2_driver_node-4] ERROR:asyncio:Task exception was never retrieved
[go2_driver_node-4] future: <Task finished name='Task-4' coro=<RobotBaseNode.run() done, defined at /home/klminnovation/Desktop/ros2_ws_111/install/go2_robot_sdk/lib/python3.10/site-packages/go2_robot_sdk/go2_driver_node.py:479> exception=ClientConnectorError(ConnectionKey(host='172.28.164.106', port=8081, is_ssl=False, ssl=True, proxy=None, proxy_auth=None, proxy_headers_hash=None), ConnectionRefusedError(111, "Connect call failed ('172.28.164.106', 8081)"))>
[go2_driver_node-4] Traceback (most recent call last):
[go2_driver_node-4]   File "/home/klminnovation/.local/lib/python3.10/site-packages/aiohttp/connector.py", line 1073, in _wrap_create_connection
[go2_driver_node-4]     sock = await aiohappyeyeballs.start_connection(
[go2_driver_node-4]   File "/home/klminnovation/.local/lib/python3.10/site-packages/aiohappyeyeballs/impl.py", line 104, in start_connection
[go2_driver_node-4]     raise first_exception
[go2_driver_node-4]   File "/home/klminnovation/.local/lib/python3.10/site-packages/aiohappyeyeballs/impl.py", line 81, in start_connection
[go2_driver_node-4]     sock = await _connect_sock(
[go2_driver_node-4]   File "/home/klminnovation/.local/lib/python3.10/site-packages/aiohappyeyeballs/impl.py", line 166, in _connect_sock
[go2_driver_node-4]     await loop.sock_connect(sock, address)
[go2_driver_node-4]   File "/usr/lib/python3.10/asyncio/selector_events.py", line 501, in sock_connect
[go2_driver_node-4]     return await fut
[go2_driver_node-4]   File "/usr/lib/python3.10/asyncio/selector_events.py", line 541, in _sock_connect_cb
[go2_driver_node-4]     raise OSError(err, f'Connect call failed {address}')
[go2_driver_node-4] ConnectionRefusedError: [Errno 111] Connect call failed ('172.28.164.106', 8081)
[go2_driver_node-4] 
[go2_driver_node-4] The above exception was the direct cause of the following exception:
[go2_driver_node-4] 
[go2_driver_node-4] Traceback (most recent call last):
[go2_driver_node-4]   File "/home/klminnovation/Desktop/ros2_ws_111/install/go2_robot_sdk/lib/python3.10/site-packages/go2_robot_sdk/go2_driver_node.py", line 483, in run
[go2_driver_node-4]     await self.conn[robot_num].connect()
[go2_driver_node-4]   File "/home/klminnovation/Desktop/ros2_ws_111/install/go2_robot_sdk/lib/python3.10/site-packages/scripts/webrtc_driver.py", line 140, in connect
[go2_driver_node-4]     async with session.post(url, json=data, headers=headers) as resp:
[go2_driver_node-4]   File "/home/klminnovation/.local/lib/python3.10/site-packages/aiohttp/client.py", line 1353, in __aenter__
[go2_driver_node-4]     self._resp = await self._coro
[go2_driver_node-4]   File "/home/klminnovation/.local/lib/python3.10/site-packages/aiohttp/client.py", line 657, in _request
[go2_driver_node-4]     conn = await self._connector.connect(
[go2_driver_node-4]   File "/home/klminnovation/.local/lib/python3.10/site-packages/aiohttp/connector.py", line 564, in connect
[go2_driver_node-4]     proto = await self._create_connection(req, traces, timeout)
[go2_driver_node-4]   File "/home/klminnovation/.local/lib/python3.10/site-packages/aiohttp/connector.py", line 975, in _create_connection
[go2_driver_node-4]     _, proto = await self._create_direct_connection(req, traces, timeout)
[go2_driver_node-4]   File "/home/klminnovation/.local/lib/python3.10/site-packages/aiohttp/connector.py", line 1350, in _create_direct_connection
[go2_driver_node-4]     raise last_exc
[go2_driver_node-4]   File "/home/klminnovation/.local/lib/python3.10/site-packages/aiohttp/connector.py", line 1319, in _create_direct_connection
[go2_driver_node-4]     transp, proto = await self._wrap_create_connection(
[go2_driver_node-4]   File "/home/klminnovation/.local/lib/python3.10/site-packages/aiohttp/connector.py", line 1088, in _wrap_create_connection
[go2_driver_node-4]     raise client_error(req.connection_key, exc) from exc
[... cut ...]
[go2_driver_node-4] AttributeError: 'SingleThreadedExecutor' object has no attribute '_sigint_gc'
[go2_driver_node-4] ERROR:asyncio:Task was destroyed but it is pending!
[go2_driver_node-4] task: <Task pending name='Task-2' coro=<spin() running at /home/klminnovation/Desktop/ros2_ws_111/install/go2_robot_sdk/lib/python3.10/site-packages/go2_robot_sdk/go2_driver_node.py:506> wait_for=<Future pending cb=[Task.task_wakeup()]>>
[INFO] [go2_driver_node-4]: process has finished cleanly [pid 19566]

`

abizovnuralem, 2024-09-09:
@peterhuisman thanks, have you seen in our logs Failed to get answer from server: Reason:?

umsh315, 2024-09-11:
I also own a Go2EDU with version 1.1.1, so I tried running it. I hope this can be of help to you. Additionally, I was able to connect using https://github.com/legion1581/go2_webrtc_connect .
[errorlog.txt](https://github.com/user-attachments/files/16958242/errorlog.txt)

abizovnuralem, 2024-09-11:
@umsh315 Thanks! I took all the new handshake methods from that repo, can you try one more time? I just fix your error

umsh315, 2024-09-11:
I encountered this error. There may be something wrong with my environment, but it seems that the content of the error has indeed changed.
[errorlog2.txt](https://github.com/user-attachments/files/16958832/errorlog2.txt)

abizovnuralem, 2024-09-11:
@umsh315 and you rm -rf log install build folders and redo colcon build again please?

Peenut173, 2024-09-11:
okay I have tested your new version and got this error:
INFO] [launch]: All log files can be found below /home/peenut/.ros/log/2024-09-11-11-13-27-056445-peenut-Nitro-AN517-54-16697
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [16710]
[INFO] [ros2_go2_video-2]: process started with pid [16712]
[INFO] [pointcloud_to_laserscan_node-3]: process started with pid [16714]
[INFO] [go2_driver_node-4]: process started with pid [16716]
[INFO] [lidar_to_pointcloud-5]: process started with pid [16718]
[INFO] [joy_node-6]: process started with pid [16720]
[INFO] [teleop_node-7]: process started with pid [16722]
[INFO] [twist_mux-8]: process started with pid [16724]
[INFO] [foxglove_bridge-9]: process started with pid [16726]
[INFO] [relay-10]: process started with pid [16728]
[twist_mux-8] [INFO] [1726046007.358990783] [twist_mux]: Topic handler 'topics.joy' subscribed to topic 'cmd_vel_joy': timeout = 0.500000s , priority = 10.
[teleop_node-7] [INFO] [1726046007.359647508] [TeleopTwistJoy]: Linear axis x on 1 at scale 0.400000.
[teleop_node-7] [INFO] [1726046007.359691523] [TeleopTwistJoy]: Linear axis y on 0 at scale 0.400000.
[teleop_node-7] [INFO] [1726046007.359696687] [TeleopTwistJoy]: Linear axis z on 3 at scale 1.000000.
[teleop_node-7] [INFO] [1726046007.359700843] [TeleopTwistJoy]: Angular axis yaw on 2 at scale 0.500000.
[twist_mux-8] [INFO] [1726046007.359983745] [twist_mux]: Topic handler 'topics.navigation' subscribed to topic 'cmd_vel': timeout = 0.500000s , priority = 15.
[ros2_go2_video-2] [INFO] [1726046007.384092007] [go2_video]: Starting go2_video
[robot_state_publisher-1] [INFO] [1726046007.384863283] [robot0.robot_state_publisher]: got segment map
[robot_state_publisher-1] [INFO] [1726046007.384924313] [robot0.robot_state_publisher]: got segment odom
[robot_state_publisher-1] [INFO] [1726046007.384933252] [robot0.robot_state_publisher]: got segment robot0/FL_calf
[robot_state_publisher-1] [INFO] [1726046007.384939385] [robot0.robot_state_publisher]: got segment robot0/FL_calflower
[robot_state_publisher-1] [INFO] [1726046007.384945401] [robot0.robot_state_publisher]: got segment robot0/FL_calflower1
[robot_state_publisher-1] [INFO] [1726046007.384951314] [robot0.robot_state_publisher]: got segment robot0/FL_foot
[robot_state_publisher-1] [INFO] [1726046007.384956988] [robot0.robot_state_publisher]: got segment robot0/FL_hip
[robot_state_publisher-1] [INFO] [1726046007.384962774] [robot0.robot_state_publisher]: got segment robot0/FL_thigh
[robot_state_publisher-1] [INFO] [1726046007.384968622] [robot0.robot_state_publisher]: got segment robot0/FR_calf
[robot_state_publisher-1] [INFO] [1726046007.384974422] [robot0.robot_state_publisher]: got segment robot0/FR_calflower
[robot_state_publisher-1] [INFO] [1726046007.384980259] [robot0.robot_state_publisher]: got segment robot0/FR_calflower1
[robot_state_publisher-1] [INFO] [1726046007.384986099] [robot0.robot_state_publisher]: got segment robot0/FR_foot
[robot_state_publisher-1] [INFO] [1726046007.384992078] [robot0.robot_state_publisher]: got segment robot0/FR_hip
[robot_state_publisher-1] [INFO] [1726046007.384997858] [robot0.robot_state_publisher]: got segment robot0/FR_thigh
[robot_state_publisher-1] [INFO] [1726046007.385003727] [robot0.robot_state_publisher]: got segment robot0/Head_lower
[robot_state_publisher-1] [INFO] [1726046007.385009332] [robot0.robot_state_publisher]: got segment robot0/Head_upper
[robot_state_publisher-1] [INFO] [1726046007.385015033] [robot0.robot_state_publisher]: got segment robot0/RL_calf
[robot_state_publisher-1] [INFO] [1726046007.385020673] [robot0.robot_state_publisher]: got segment robot0/RL_calflower
[robot_state_publisher-1] [INFO] [1726046007.385026384] [robot0.robot_state_publisher]: got segment robot0/RL_calflower1
[robot_state_publisher-1] [INFO] [1726046007.385032042] [robot0.robot_state_publisher]: got segment robot0/RL_foot
[robot_state_publisher-1] [INFO] [1726046007.385037685] [robot0.robot_state_publisher]: got segment robot0/RL_hip
[robot_state_publisher-1] [INFO] [1726046007.385043478] [robot0.robot_state_publisher]: got segment robot0/RL_thigh
[robot_state_publisher-1] [INFO] [1726046007.385049097] [robot0.robot_state_publisher]: got segment robot0/RR_calf
[robot_state_publisher-1] [INFO] [1726046007.385057327] [robot0.robot_state_publisher]: got segment robot0/RR_calflower
[robot_state_publisher-1] [INFO] [1726046007.385067190] [robot0.robot_state_publisher]: got segment robot0/RR_calflower1
[robot_state_publisher-1] [INFO] [1726046007.385077342] [robot0.robot_state_publisher]: got segment robot0/RR_foot
[robot_state_publisher-1] [INFO] [1726046007.385087282] [robot0.robot_state_publisher]: got segment robot0/RR_hip
[robot_state_publisher-1] [INFO] [1726046007.385095745] [robot0.robot_state_publisher]: got segment robot0/RR_thigh
[robot_state_publisher-1] [INFO] [1726046007.385104125] [robot0.robot_state_publisher]: got segment robot0/base_footprint
[robot_state_publisher-1] [INFO] [1726046007.385114158] [robot0.robot_state_publisher]: got segment robot0/base_link
[robot_state_publisher-1] [INFO] [1726046007.385124013] [robot0.robot_state_publisher]: got segment robot0/front_camera
[robot_state_publisher-1] [INFO] [1726046007.385133938] [robot0.robot_state_publisher]: got segment robot0/imu
[robot_state_publisher-1] [INFO] [1726046007.385144026] [robot0.robot_state_publisher]: got segment robot0/radar
[... cut ...]
[ros2_go2_video-2] failed to create decoder: BackendError(ffmpeg::Error(5: Input/output error))
[ros2_go2_video-2] note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
[ERROR] [ros2_go2_video-2]: process has died [pid 16712, exit code 101, cmd '/media/peenut/programmes/dossier_LORIA/robot_go2/code_go2/go2_ws/install/ros2_go2_video/lib/ros2_go2_video/ros2_go2_video --ros-args --params-file /tmp/launch_params_8nhtvl2p'].

it says there is no module "sdp". Is the webrtc connection supposed to use the AP of the GO2 or not?

abizovnuralem, 2024-09-11:
you should pull again, @Peenut173

roboticsN, 2024-09-11:
Ok, just updated the master branch, now it should work! But we still need to rewrite the video stream node to support the new method. I would be happy if someone would do that.

abizovnuralem, 2024-09-11:
Ok, just updated the master branch, now it should work! But we still need to rewrite the video stream node to support the new method. I would be happy if someone would do that.

abizovnuralem, 2024-09-11:
@Peenut173 Can you check it on your side and if webrtc is working you can close this issue

umsh315, 2024-09-12:
i can use this thanks!
[image]

peterhuisman, 2024-09-12:
> @peterhuisman thanks, have you seen in our logs Failed to get answer from server: Reason:? Also I have added additional logs, can you try again please?

With this latest update, I am able to connect and have the lidar image.
Thanks for the help!!

Peenut173, 2024-09-12:
Personally, I have this problem:
[go2_driver_node-3] [INFO] [1726127831.299035764] [go2_driver_node]: Received ip list: ['192.168.12.1']
[go2_driver_node-3] [INFO] [1726127831.299348527] [go2_driver_node]: Connection type is webrtc
[go2_driver_node-3] [INFO] [1726127831.299570769] [go2_driver_node]: Connection mode is single
[go2_driver_node-3] INFO:scripts.webrtc_driver:Connection state is connecting
[go2_driver_node-3] INFO:scripts.webrtc_driver:Connection state is failed

it tried to connect for a fairly long time before failing

Peenut173, 2024-09-12:
> Personally, I have this problem: [go2_driver_node-3] [INFO] [1726127831.299035764] [go2_driver_node]: Received ip list: ['192.168.12.1'] [go2_driver_node-3] [INFO] [1726127831.299348527] [...]

Nevermind, it now works, but i don't see the robot anymore on rviz (nav2 and the other stuffs work), how can i put it again?

abizovnuralem, 2024-09-12:
You dont see the urdf of the dog? Can you make a screenshot here please?

Peenut173, 2024-09-12:
[image]

abizovnuralem, 2024-09-12:
You can add this using robot model inside rviz

Peenut173, 2024-09-12:
okay now it works, thanks a lot for your help and great job for your reactivity for this firmware update!

closed 2024-09-12 by Peenut173

chrisuzuki4134, 2024-09-25:
Hi I am having issues where I believe port 9991 for webrtc is closed and will not me connect to the unitree go2 robot using webrtc.
The following is my connection error:
[go2_driver_node-3] ERROR:root:An error occurred: HTTPConnectionPool(host='192.168.0.166', port=9991): Max retries exceeded with url: /con_notify (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fdf36e374c0>: Failed to establish a new connection: [Errno 111] Connection refused'))
[go2_driver_node-3] ERROR:asyncio:Task exception was never retrieved
[go2_driver_node-3] future: <Task finished name='Task-4' coro=<RobotBaseNode.run() done, defined at /home/xtechlabrobot/ROS2/ros2_ws/install/go2_robot_sdk/lib/python3.10/site-packages/go2_robot_sdk/go2_driver_node.py:526> exception=ValueError('Failed to receive initial public key response with new method.')>
[go2_driver_node-3] Traceback (most recent call last):
[go2_driver_node-3]   File "/home/xtechlabrobot/ROS2/ros2_ws/install/go2_robot_sdk/lib/python3.10/site-packages/go2_robot_sdk/go2_driver_node.py", line 530, in run
[go2_driver_node-3]     await self.conn[robot_num].connect()
[go2_driver_node-3]   File "/home/xtechlabrobot/ROS2/ros2_ws/install/go2_robot_sdk/lib/python3.10/site-packages/scripts/webrtc_driver.py", line 334, in connect
[go2_driver_node-3]     raise ValueError(
[go2_driver_node-3] ValueError: Failed to receive initial public key response with new method.
[go2_driver_node-3] Exception in thread Thread-1 (_spin):
[go2_driver_node-3] Traceback (most recent call last):
[go2_driver_node-3]   File "/usr/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
[go2_driver_node-3]     self.run()
[go2_driver_node-3]   File "/usr/lib/python3.10/threading.py", line 953, in run
[go2_driver_node-3]     self._target(*self._args, **self._kwargs)
[go2_driver_node-3]   File "/home/xtechlabrobot/ROS2/ros2_ws/install/go2_robot_sdk/lib/python3.10/site-packages/go2_robot_sdk/go2_driver_node.py", line 545, in _spin
[go2_driver_node-3]     rclpy.spin_once(node)
[go2_driver_node-3]   File "/home/xtechlabrobot/ROS2/ros2_humble/install/rclpy/local/lib/python3.10/dist-packages/rclpy/__init__.py", line 203, in spin_once
[go2_driver_node-3]     executor = get_global_executor() if executor is None else executor
[go2_driver_node-3]   File "/home/xtechlabrobot/ROS2/ros2_humble/install/rclpy/local/lib/python3.10/dist-packages/rclpy/__init__.py", line 106, in get_global_executor
[go2_driver_node-3]     __executor = SingleThreadedExecutor()
[go2_driver_node-3]   File "/home/xtechlabrobot/ROS2/ros2_humble/install/rclpy/local/lib/python3.10/dist-packages/rclpy/executors.py", line 721, in __init__
[go2_driver_node-3]     super().__init__(context=context)
[go2_driver_node-3]   File "/home/xtechlabrobot/ROS2/ros2_humble/install/rclpy/local/lib/python3.10/dist-packages/rclpy/executors.py", line 172, in __init__
[go2_driver_node-3]     self._guard = GuardCondition(
[go2_driver_node-3]   File "/home/xtechlabrobot/ROS2/ros2_humble/install/rclpy/local/lib/python3.10/dist-packages/rclpy/guard_condition.py", line 23, in __init__
[go2_driver_node-3]     with self._context.handle:
[go2_driver_node-3] AttributeError: __enter__
[go2_driver_node-3] Exception ignored in: <function Executor.__del__ at 0x7fe1b2ab7250>
[go2_driver_node-3] Traceback (most recent call last):
[go2_driver_node-3]   File "/home/xtechlabrobot/ROS2/ros2_humble/install/rclpy/local/lib/python3.10/dist-packages/rclpy/executors.py", line 243, in __del__
[go2_driver_node-3]     if self._sigint_gc is not None:
[go2_driver_node-3] AttributeError: 'SingleThreadedExecutor' object has no attribute '_sigint_gc'
[go2_driver_node-3] ERROR:asyncio:Task was destroyed but it is pending!
[go2_driver_node-3] task: <Task pending name='Task-2' coro=<spin() running at /home/xtechlabrobot/ROS2/ros2_ws/install/go2_robot_sdk/lib/python3.10/site-packages/go2_robot_sdk/go2_driver_node.py:554> wait_for=<Future pending cb=[Task.task_wakeup()]>>
[INFO] [go2_driver_node-3]: process has finished cleanly [pid 37748]