abizovnuralem/go2_ros2_sdk #81: Raw Lidar Data

rihat99, 2024-09-13:
Is it possible to get raw Lidar data using Webrtc? The same as in "/utlidar/cloud" in CycloneDDS connection. By default /point_cloud2 has processed data and points are structured. My goal is to get access to unprocessed raw point cloud.

rihat99, 2024-09-13:
I am not sure if it's technically possible, but it would be nice for Edu version with Jetson to somehow configure CycloneDDS to forward all ROS2 topics from LAN connection (main GO2 board) to Wi-Fi connection (via external module). I am new to CycloneDDS, but I suppose such bridge is possible to configure. It would allow us to have access to all ROS2 topic on workstation connected to same Wi-Fi. If you have any ideas how to do it or have hits, I will try it.

abizovnuralem, 2024-09-17:
It is possible to do this if you have access to the internal Ubuntu system. Currently, Unitree prevents logging in to the Linux system.

closed 2024-09-17 by abizovnuralem