abizovnuralem/go2_ros2_sdk #82: Connection error with GO2

CarlosJuarezTec, 2024-09-13:
Hi, I've been trying to use the package, almost everything works, except that the communication cannot be established with the robot despite my best efforts. I'm posting the error message, hope someone can help me. 

[go2_driver_node-3] ERROR:root:An error occurred: HTTPConnectionPool(host='192.168.12.53', port=9991): Max retries exceeded with url: /con_notify (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7ba396bd4430>: Failed to establish a new connection: [Errno 111] Connection refused'))
[go2_driver_node-3] ERROR:asyncio:Task exception was never retrieved
[go2_driver_node-3] future: <Task finished name='Task-4' coro=<RobotBaseNode.run() done, defined at /home/ubuntudev22/ros2_ws/install/go2_robot_sdk/lib/python3.10/site-packages/go2_robot_sdk/go2_driver_node.py:479> exception=ValueError('Failed to receive initial public key response with new method.')>
[go2_driver_node-3] Traceback (most recent call last):
[go2_driver_node-3]   File "/home/ubuntudev22/ros2_ws/install/go2_robot_sdk/lib/python3.10/site-packages/go2_robot_sdk/go2_driver_node.py", line 483, in run
[go2_driver_node-3]     await self.conn[robot_num].connect()
[go2_driver_node-3]   File "/home/ubuntudev22/ros2_ws/install/go2_robot_sdk/lib/python3.10/site-packages/scripts/webrtc_driver.py", line 334, in connect
[go2_driver_node-3]     raise ValueError(
[go2_driver_node-3] ValueError: Failed to receive initial public key response with new method.
[go2_driver_node-3] Exception in thread Thread-1 (_spin):
[go2_driver_node-3] Traceback (most recent call last):
[go2_driver_node-3]   File "/usr/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
[go2_driver_node-3]     self.run()
[go2_driver_node-3]   File "/usr/lib/python3.10/threading.py", line 953, in run
[go2_driver_node-3]     self._target(*self._args, **self._kwargs)
[go2_driver_node-3]   File "/home/ubuntudev22/ros2_ws/install/go2_robot_sdk/lib/python3.10/site-packages/go2_robot_sdk/go2_driver_node.py", line 498, in _spin
[go2_driver_node-3]     rclpy.spin_once(node)
[go2_driver_node-3]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/__init__.py", line 199, in spin_once
[go2_driver_node-3]     executor = get_global_executor() if executor is None else executor
[go2_driver_node-3]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/__init__.py", line 102, in get_global_executor
[go2_driver_node-3]     __executor = SingleThreadedExecutor()
[go2_driver_node-3]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 721, in __init__
[go2_driver_node-3]     super().__init__(context=context)
[go2_driver_node-3]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 172, in __init__
[go2_driver_node-3]     self._guard = GuardCondition(
[go2_driver_node-3]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/guard_condition.py", line 23, in __init__
[go2_driver_node-3]     with self._context.handle:
[go2_driver_node-3] AttributeError: __enter__
[go2_driver_node-3] Exception ignored in: <function Executor.__del__ at 0x7ba3989ce0e0>
[go2_driver_node-3] Traceback (most recent call last):
[go2_driver_node-3]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 243, in __del__
[go2_driver_node-3]     if self._sigint_gc is not None:
[go2_driver_node-3] AttributeError: 'SingleThreadedExecutor' object has no attribute '_sigint_gc'
[go2_driver_node-3] ERROR:asyncio:Task was destroyed but it is pending!
[go2_driver_node-3] task: <Task pending name='Task-2' coro=<spin() running at /home/ubuntudev22/ros2_ws/install/go2_robot_sdk/lib/python3.10/site-packages/go2_robot_sdk/go2_driver_node.py:507> wait_for=<Future pending cb=[Task.task_wakeup()]>>

abizovnuralem, 2024-09-13:
Hello! Are you connected to the mobile app at the same time?

CarlosJuarezTec, 2024-09-13:
I was, but I've tried closing the app and even rebooting the phone but error still stands.

abizovnuralem, 2024-09-13:
If you try to connect to ros2 don't use app at the same time

CarlosJuarezTec, 2024-09-18:
Hi, I've been trying to solve the issue but even without using the app at the same time i Cannot connect

CarlosJuarezTec, 2024-09-19:
I've confirmed that going back to the commit mentioned made the communication work. Also I tried using it with a computer with an older clone of the git and worked too. 

Most likely something is wrong with the newer commit.

abizovnuralem, 2024-09-20:
The latest commit has firmware 1.1.1 support, so if you are still on the previous version for some reason, you can use old commits or  https://github.com/abizovnuralem/go2_ros2_sdk/releases/tag/v0.2.0. That means they don't have back support. Thanks for this issue!

closed 2024-09-20 by abizovnuralem