abizovnuralem/go2_ros2_sdk #83: Error installing ROS2

luispla22, 2024-09-16:
When I try to install the ros-image-tools I obtain the "unable to locate package" message.
Do I have to add any repository or install any dependency before? Thanks.
$sudo apt install ros-humble-image-tools
E: Unable to locate package ros-humble-image-tools

abizovnuralem, 2024-09-17:
Hello! Can you try
sudo apt update
sudo apt install ros-humble-desktop

luispla22, 2024-09-17:
I obtain the same error: "unable to locate package".

abizovnuralem, 2024-09-17:
Have you tried:
source /opt/ros/humble/setup.bash

or 
sudo apt update && sudo apt install curl gnupg2 lsb-release
curl -s https://raw.githubusercontent.com/ros/rosdistro/master/ros.asc | sudo apt-key add -
sudo sh -c 'echo "deb http://packages.ros.org/ros2/ubuntu $(lsb_release -cs) main" > /etc/apt/sources.list.d/ros2-latest.list'
sudo apt update

abizovnuralem, 2024-09-20:
Have you been able to fix that issue? If not, please reopen the new thread and I will try to help you, thanks!

closed 2024-09-20 by abizovnuralem