abizovnuralem/go2_ros2_sdk #87: Can't update static costmap layer, no map received. Connecting to the GO2 Robot Via cyclonedds

Erenrobot, 2024-09-25:
**Hello, I am trying to launch "ros2 launch go2_robot_sdk robot.launch.py" file but It is not showing anything on RVIZ(I am using ROS2 humble on Ubuntu 22.04). There is no connection error and I have already tried the IP address export ROBOT_IP="192.168.123.222" is working on the official python and C++ SDK of Unitree Robotics. **

The ros2 node list is:
/behavior_server
/bt_navigator
/bt_navigator_navigate_through_poses_rclcpp_node
/bt_navigator_navigate_to_pose_rclcpp_node
/controller_server
/foxglove_bridge
/foxglove_bridge_component_manager
/global_costmap/global_costmap
/go2_driver_node
/joy_node
/lidar_to_pointcloud
/lifecycle_manager_navigation
/local_costmap/local_costmap
/planner_server
/pointcloud_to_laserscan
/robot_state_publisher
/rviz2
/slam_toolbox
/smoother_server
/teleop_node
/transform_listener_impl_556ce3c99620
/transform_listener_impl_55b04bf80370
/transform_listener_impl_5d78826f8b20
/transform_listener_impl_755eec01beb0
/transform_listener_impl_79d6d0002450
/transform_listener_impl_7c25dc008020
/transform_listener_impl_7fb0b8008a50
/twist_mux
/velocity_smoother
/waypoint_follower

[image]

It only allows me to change the topic to /robot_description which is only the robot body without legs and other components.

**The output of launch is:**

eren@eren:~/go2Sim_ws$ ros2 launch go2_robot_sdk robot.launch.py
[INFO] [launch]: All log files can be found below /home/eren/.ros/log/2024-09-24-19-44-33-729942-eren-74228
[INFO] [launch]: Default logging verbosity is set to INFO
IP list: ['192.168.123.222']
[INFO] [robot_state_publisher-1]: process started with pid [74230]
[INFO] [pointcloud_to_laserscan_node-2]: process started with pid [74232]
[INFO] [go2_driver_node-3]: process started with pid [74234]
[INFO] [lidar_to_pointcloud-4]: process started with pid [74236]
[INFO] [rviz2-5]: process started with pid [74238]
[INFO] [joy_node-6]: process started with pid [74240]
[INFO] [teleop_node-7]: process started with pid [74242]
[INFO] [twist_mux-8]: process started with pid [74244]
[INFO] [foxglove_bridge-9]: process started with pid [74246]
[INFO] [async_slam_toolbox_node-10]: process started with pid [74248]
[INFO] [controller_server-11]: process started with pid [74250]
[INFO] [smoother_server-12]: process started with pid [74252]
[INFO] [planner_server-13]: process started with pid [74255]
[INFO] [behavior_server-14]: process started with pid [74259]
[INFO] [bt_navigator-15]: process started with pid [74261]
[INFO] [waypoint_follower-16]: process started with pid [74263]
[INFO] [velocity_smoother-17]: process started with pid [74266]
[INFO] [lifecycle_manager-18]: process started with pid [74269]
[rviz2-5] Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[twist_mux-8] [INFO] [1727232274.114590363] [twist_mux]: Topic handler 'topics.joy' subscribed to topic 'cmd_vel_joy': timeout = 0.500000s , priority = 10.
[twist_mux-8] [INFO] [1727232274.114879369] [twist_mux]: Topic handler 'topics.navigation' subscribed to topic 'cmd_vel': timeout = 0.500000s , priority = 5.
[robot_state_publisher-1] [INFO] [1727232274.117613452] [robot_state_publisher]: got segment FL_calf
[robot_state_publisher-1] [INFO] [1727232274.117659034] [robot_state_publisher]: got segment FL_calflower
[robot_state_publisher-1] [INFO] [1727232274.117666826] [robot_state_publisher]: got segment FL_calflower1
[robot_state_publisher-1] [INFO] [1727232274.117672110] [robot_state_publisher]: got segment FL_foot
[robot_state_publisher-1] [INFO] [1727232274.117676750] [robot_state_publisher]: got segment FL_hip
[... 207 lines cut ...]
[behavior_server-14] [INFO] [1727232277.379923852] [behavior_server]: Activating drive_on_heading
[behavior_server-14] [INFO] [1727232277.379932984] [behavior_server]: Activating assisted_teleop
[behavior_server-14] [INFO] [1727232277.379942069] [behavior_server]: Activating wait
[behavior_server-14] [INFO] [1727232277.379953928] [behavior_server]: Creating bond (behavior_server) to lifecycle manager.
[lifecycle_manager-18] [INFO] [1727232277.482664441] [lifecycle_manager_navigation]: Server behavior_server connected with bond.
[lifecycle_manager-18] [INFO] [1727232277.482852288] [lifecycle_manager_navigation]: Activating bt_navigator
[bt_navigator-15] [INFO] [1727232277.483406172] [bt_navigator]: Activating
[bt_navigator-15] [INFO] [1727232277.532326076] [bt_navigator]: Creating bond (bt_navigator) to lifecycle manager.
[lifecycle_manager-18] [INFO] [1727232277.635591408] [lifecycle_manager_navigation]: Server bt_navigator connected with bond.
[lifecycle_manager-18] [INFO] [1727232277.635735787] [lifecycle_manager_navigation]: Activating waypoint_follower
[waypoint_follower-16] [INFO] [1727232277.636284281] [waypoint_follower]: Activating
[waypoint_follower-16] [INFO] [1727232277.636421515] [waypoint_follower]: Creating bond (waypoint_follower) to lifecycle manager.
[lifecycle_manager-18] [INFO] [1727232277.742070398] [lifecycle_manager_navigation]: Server waypoint_follower connected with bond.
[lifecycle_manager-18] [INFO] [1727232277.742112421] [lifecycle_manager_navigation]: Activating velocity_smoother
[velocity_smoother-17] [INFO] [1727232277.742497316] [velocity_smoother]: Activating
[velocity_smoother-17] [INFO] [1727232277.742570085] [velocity_smoother]: Creating bond (velocity_smoother) to lifecycle manager.
[lifecycle_manager-18] [INFO] [1727232277.844513365] [lifecycle_manager_navigation]: Server velocity_smoother connected with bond.
[lifecycle_manager-18] [INFO] [1727232277.844611177] [lifecycle_manager_navigation]: Managed nodes are active
[lifecycle_manager-18] [INFO] [1727232277.844642936] [lifecycle_manager_navigation]: Creating bond timer...
[controller_server-11] [WARN] [1727232284.663560079] [local_costmap.local_costmap]: Can't update static costmap layer, no map received
[planner_server-13] [WARN] [1727232286.161617844] [global_costmap.global_costmap]: Can't update static costmap layer, no map received
[controller_server-11] [WARN] [1727232294.663520942] [local_costmap.local_costmap]: Can't update static costmap layer, no map received
[planner_server-13] [WARN] [1727232296.240300947] [global_costmap.global_costmap]: Can't update static costmap layer, no map received
[controller_server-11] [WARN] [1727232304.663535256] [local_costmap.local_costmap]: Can't update static costmap layer, no map received
[planner_server-13] [WARN] [1727232306.215343854] [global_costmap.global_costmap]: Can't update static costmap layer, no map received

abizovnuralem, 2024-09-26:
Hello, what kind of dog firmware are you using?

KingJacM, 2024-09-27:
> Hello, what kind of dog firmware are you using?

FYI, my robot is edu version, I am not sure of the firmware you are referring to. I connected it by ethernet and later realized that I might need to connect by WIFI, but when I set the connection type to cyclonedds, still nothing appears.

abizovnuralem, 2024-09-29:
You can try a webrtc method first, for the cyclonesdds can you please share your output here with this cmd:
ros2 topic list
Thanks! @KingJacM

minolta001, 2024-10-03:
Hello there, I actually have the same issue. Also, did you try to check the message from publisher? I can see publisher topics but most of them publish nothing.

abizovnuralem, 2024-10-07:
> > Hello, what kind of dog firmware are you using?
> 
> We are using the latest firmware. Also, we cannot connect the robot to the Wifi. We can only connect from Ethernet. We have the EDU version too.

Can you switch to cyclonedds method please and try? You can learn how to do that from the readme instructions.

KingJacM, 2024-10-16:
> You can try a webrtc method first, for the cyclonesdds can you please share your output here with this cmd: ros2 topic list Thanks! @KingJacM

@abizovnuralem The topic list shows:
```
humble@humble-virtual-machine:~/Documents/ros2_ws$ ros2 topic list
/4g_traffic_report
/all_edge
/all_node
/api/audio_hub/request
/api/audio_hub/response
/api/bashrunner/request
/api/bashrunner/response
/api/gpt/request
/api/gpt/response
/api/obstacles_avoid/request
/api/obstacles_avoid/response
/api/robot_state/request
/api/robot_state/response
/api/sport/request
/api/sport/response
/api/sport_lease/request
/api/sport_lease/response
/api/uwbswitch/request
/api/uwbswitch/response
/api/videohub/request
/api/videohub/response
/api/vui/request
/api/vui/response
/audioreceiver
/audiosender
/frontvideostream
/gpt_cmd
/gptflowfeedback
/lf/lowstate
/lf/sportmodestate
/lio_sam_ros2/mapping/odometry
/lowcmd
/lowstate
/mf/sportmodestate
/multiplestate
/parameter_events
/pctoimage_local
/public_network_status
/qt_add_edge
/qt_add_node
/qt_command
/qt_modify_edge
/qt_modify_node
/qt_notice
/query_result_edge
/query_result_node
/rosout
/rtc_status
/selftest
/servicestate
/servicestateactivate
/sportmodestate
/utlidar/cloud
/utlidar/cloud_deskewed
/utlidar/foot_position
/utlidar/grid_map
/utlidar/height_map
/utlidar/height_map_array
/utlidar/imu
/utlidar/lidar_state
/utlidar/map_state
/utlidar/mapping_cmd
/utlidar/range_info
/utlidar/range_map
/utlidar/robot_odom
/utlidar/robot_pose
/utlidar/switch
/utlidar/voxel_map
/utlidar/voxel_map_compressed
/uwbstate
/uwbswitch
/videohub/inner
/webrtcreq
/webrtcres
/wirelesscontroller
```

The error message from running `ros2 launch go2_robot_sdk robot.launch.py
` is:
```foxglove_bridge-9] [WARN] [1729061046.863346920] [foxglove_bridge]: Failed to add channel for topic "/all_edge" (graph_msg/msg/Edge): package 'graph_msg' not found, searching: [/home/humble/Documents/ros2_ws/install/go2_robot_sdk, /home/humble/Documents/ros2_ws/install/unitree_go, /home/humble/Documents/ros2_ws/install/go2_interfaces, /home/humble/Documents/ros2_ws/install/coco_detector, /opt/ros/humble]
[controller_server-11] [WARN] [1729061050.614756948] [local_costmap.local_costmap]: Can't update static costmap layer, no map received
[planner_server-13] [WARN] [1729061057.390710984] [global_costmap.global_costmap]: Can't update static costmap layer, no map received
[controller_server-11] [WARN] [1729061060.614934259] [local_costmap.local_costmap]: Can't update static costmap layer, no map received
```

ImperiumKontratakuje, 2024-10-16:
First of all, I would like to congratulate you on this fantastic project and thank you for the time, effort, and dedication you’ve put into it. **Thank you very much, and congratulations on your expertise!** :)

In my case, I’m also experiencing issues connecting via CONN_TYPE="cyclonedds".

Below are some initial/entry parameters:

### **1. Robot**

**A. Model**
    
Unitree Go2 Pro, purchased in October 2024.
Hardware version: V2.0
Software version: V1.1.1

**B. Network**

On the built-in network port (RJ45 socket accessible under the flap on the back), for the internal computer called "GO2 MCU," a static IP address 192.168.123.161 is set.
When connected to a DHCP server, the network interface also obtains an additional dynamic IP address from the DHCP pool. Scanning the address with nmap shows that the SSH service is running and open on port 22.

> Nmap scan report for 192.168.123.161
> Host is up (0.00026s latency).
> Not shown: 999 closed ports
> PORT   STATE SERVICE
> 22/tcp open  ssh

Unfortunately, I haven't found any information about the login credentials (user/password) for the internal computer "GO2 MCU" (what a shame).

The available information regarding login credentials for the Unitree Go2 EDU version only refers to the additional external computer on the NVIDIA Jetson Orin platform, called "GO2 External," which has the IP address 192.168.123.18 and credentials unitree/123.

**C. Strange behavior / issues / no Wi-Fi network**

With my Go2 unit, I have a similar problem as @Erenrobot and cannot connect to the Wi-Fi network from the "GO2 MCU" – only the 4G network and LAN port are functional.
After disabling the 4G option in the mobile app, I lost all connection with the robot, but I managed to re-enable the 4G option by directly connecting the GO2 to the Wi-Fi router via LAN cable. This allowed the mobile app to reconnect with the robot. I haven't checked if this TCP/IP communication is happening locally or if the traffic goes outside and returns.
I’m also unsure how these 4G connections work in my "PRO" version. From what I’ve read/seen, there is a 4G modem inside with an eSIM card, but I’m still not sure who is paying for it :)

### **2. Testing go2_ros2_sdk**

The project compiles successfully on Ubuntu 24.04 under ROS 2 Humble.
The ROS 2 nodes for the parameters and "webrtc" mode work correctly (SLAM, LIDAR, CAMERA, ODOMETRY).

```
export ROBOT_IP="192.168.123.161" 
export CONN_TYPE="webrtc"
ros2 launch go2_robot_sdk robot.launch.py 
```

When switching to CONN_TYPE="cyclonedds", I couldn’t establish a connection with the robot.

As far as I know, the default DDS implementation in ROS 2 Humble is Fast DDS, not CycloneDDS.

Therefore, I changed the default DDS for ROS 2 Humble to rmw_cyclonedds_cpp, 

```
sudo apt install ros-humble-rmw-cyclonedds-cpp

source /opt/ros/humble/setup.bash
source ~/ros2_ws/install/setup.bash

export RMW_IMPLEMENTATION="rmw_cyclonedds_cpp"
export ROBOT_IP="192.168.123.161"
export CONN_TYPE="cyclonedds"
```

**but this did not resolve the issue, and I still cannot connect to the robot :(**

[image]

abizovnuralem, 2024-10-17:
@ImperiumKontratakuje , Thank you, for your warm words, I wish you all the same x 10! 
For your problem with wifi:
if you want to use this repo, you don't need to do such modifications, you need to use the mobile app and search for the robot IP address, then, paste it and run this SDK, that's all. Also, I recommend to use webrtc method with this repo, as it has camera stream integration.

ImperiumKontratakuje, 2024-10-17:
Thank you for your response. I have successfully tested go2_ros2_sdk in WebRTC mode, but based on the diagram, it seems that DDS is the native communication method for ROS2/SDK. This type of communication has many advantages: it eliminates additional translation overhead, ensures lower latency, and provides better reliability in data transmission. Furthermore, DDS is better suited for distributed systems, allowing for more efficient data exchange between robot components without the need for conversion.
[image]

I'm not an expert, but all the vehicles I've built so far have natively used DDS, and it would be great if go2_ros2_sdk could be run in CycloneDDS mode for my Go2 "Pro" model, as this would be far more optimal than translating through WebRTC.

ImperiumKontratakuje, 2024-11-11:
I'm glad that my information was helpful. Without the ability to access the Unitree Go via SSH and unlock DDS, this robot really becomes just an expensive toy. Unlocking the robot greatly expands its capabilities—now, our only limits are our knowledge and skills.

I haven’t dedicated much time to testing yet, and so far, I haven't managed to get go2_ros2_sdk working with DDS support. However, I did succeed with the [go2_robot](https://github.com/IntelligentRoboticsLabs/go2_robot) project, which currently doesn’t support the camera or SLAM, but this functionality should be added soon.

As for DDS, after unlocking, it’s available only through the LAN port, not via Wi-Fi. I attempted to modify the DDS rules by editing the configuration file /unitree/etc/cyclonedds.xml, changing it from:

```
<?xml version="1.0" encoding="UTF-8"?>
<CycloneDDS>
  <Domain Id="any">
    <General>
      <Interfaces>
        <NetworkInterface name="eth0" priority="default" multicast="default" />
      </Interfaces>
    </General>
    <SharedMemory>
      <Enable>true</Enable>
    </SharedMemory>
    <Discovery>
      <EnableTopicDiscoveryEndpoints>true</EnableTopicDiscoveryEndpoints>
    </Discovery>
  </Domain>
</CycloneDDS>
```

to:

```
<Interfaces>
  <NetworkInterface name="eth0" priority="default" multicast="default" />
  <NetworkInterface name="wlan0" priority="default" multicast="default" />
</Interfaces>

```

This change should allow CycloneDDS to use both eth0 and wlan0. However, after a system reboot, the file reverts to its previous state, suggesting it is dynamically generated. I haven't yet figured out how to modify it permanently, so for now, DDS is accessible only through the eth0 interface.

I'm very interested in hearing about your tests with DDS in this project. If you make any progress, please share your insights!

For now, I’ll focus on learning Isaac Sim, as many projects in the field of reinforcement learning simulations use Omniverse software: https://github.com/abizovnuralem/go2_omniverse.

tackgeun, 2024-11-14:
@ImperiumKontratakuje Thank you for sharing your current status.

Although your goal is to connect cycloneDDS via both wired and wireless connections, I found that using WebRTC over a wired connection works to some extent. (As I reviewed above, I think you didn't try this one.) I’m focusing on wired communication because I need to control a robot dog outdoors with my external computing resource attached on go2.

However, there's an unusual setup required. Changing the IP to go2's static IP (192.168.123.161) alone doesn’t work. I also need to run source unitree_ros2/install/setup.bash before launching the go2 unofficial SDK with WebRTC and the static IP.

I’m currently investigating why this is necessary.

Thank you for your suggestions and comments.

abizovnuralem, 2024-12-09:
@Erenrobot . If you have additional questions, please reopen the issue, thanks!

closed 2024-12-09 by abizovnuralem