abizovnuralem/go2_ros2_sdk #88: [go2_driver_node-4] INFO:scripts.webrtc_driver:Connection state is closed

kkjing0414, 2024-09-29:
Hello dear author, I would like to ask what is the cause of [go2_driver_node-4] INFO:scripts.webrtc_driver:Connection state is closed? Hope to get your reply, thank you
[image]

tfoldi, 2024-09-29:
what is your robot's firmware version? can you share the full debug log of `go2_driver_node` ? you can run that independently with `ros2 run go2_robot_sdk go2_driver_node`

kkjing0414, 2024-10-07:
> ros2 run go2_robot_sdk go2_driver_node

kkjing0414, 2024-10-07:
[image]
 this log？

kkjing0414, 2024-10-07:
I'm sorry, I don't know what to think about my version, I just re-installed it, there is a worse problem, at first it was the remote control dog walk, but the point cloud did not change, and there are these errors: [RVIZ2-5] [ERROR] [1728307688.446746262] [rviz2]: The link is static or has unrealistic inertia, so the equivalent inertia box will not be shown.
[rviz2-5]
[RVIZ2-5] [ERROR] [1728307688.447129319] [rviz2]: The link is static or has unrealistic inertia, so the equivalent inertia box will not be shown.
[rviz2-5]
[RVIZ2-5] [INFO] [1728307689.306326667] [rviz2]: Stereo is NOT SUPPORTED
[controller_server-11] [WARN] [1728307694.725134500] [local_Costmap. local_costmap]: Can't update static costmap layer, no map received
[planner_server-13] [WARN] [1728307696.176749087] [global_costmap.global_costmap]: Can't update static costmap layer, no map received.
Then came the launch, which could only load half the dog's body, which was even worse.
[image]

abizovnuralem, 2024-10-17:
@kkjing0414  Hello, what connection method are you using?

kkjing0414, 2024-10-18:
@abizovnuralem webrtc

closed 2024-12-03 by kkjing4444