abizovnuralem/go2_ros2_sdk #91: I don´t think coco detection is working for me.

luiplapin1, 2024-10-01:
When I launch 
<code>source install/setup.bash
ros2 run coco_detector coco_detector_node
</code>

and on another terminal

<code>source install/setup.bash
ros2 topic echo /detected_objects</code>

I get nothing on the terminal. Do I have to pre-install any requirement or to active the GPU on the Go2?

abizovnuralem, 2024-10-07:
Hello, do you see any error in logs?

luiplapin1, 2024-10-07:
Not really, I get it blank

abizovnuralem, 2024-10-07:
It would help if you ran a sdk first and then see if you have errors

luispla22, 2024-10-10:
[image]
I get some error logs when I run the sdk.

abizovnuralem, 2024-10-17:
Hello @luispla22 , have you sourced your code in the terminal?

luispla22, 2024-11-07:
Yes! I've sourced the code, but still not working. I'm now using a "temporal fix" what is using yolo in the jetson, but I would like to have the integration with ROS2. 😪

abizovnuralem, 2024-12-09:
@luispla22 , If you have additional questions, please reopen the issue, thanks!

closed 2024-12-09 by abizovnuralem