abizovnuralem/go2_ros2_sdk #92: Is it possible to run the sdk with the python3.10 on conda?

minolta001, 2024-10-03:
Hi, I try run `ros2 launch go2_robot_sdk robot.launch.py` in the conda environment with python 3.10.

However, I got `from aiortc import MediaStreamTrack`, which is actually available in the conda python 3.10. It seems that ROS2 uses the  python in `/use/lib/python3.10'. 

May I know when you compile the files, are you using any virtual env or you are using the python on the bare machine?

abizovnuralem, 2024-10-03:
Hello, it looks like an environmental problem. BTW when you open an Issue do you see a template?

minolta001, 2024-10-03:
@abizovnuralem Hello. Thanks for the reply! Yes, I saw a template, but that is for bug report. So I assume I should not use it. Please correct me if I am wrong.

May I know what virtual env you are using when running your go2 SDK? I am also checking your another project `go2_omniverse`. As you suggest to use miniconda, the official document of ROS 2 suggest to avoid using conda due to some path related issues. I am bit lost here, may I know you are using the same virtual env when run both go2 SDK and go2_omniverse, or you are running them seperatly?

abizovnuralem, 2024-10-07:
Hello @minolta001 , for the go2_ros2_sdk we don't need to use virtual env

minolta001, 2024-10-07:
Hello @abizovnuralem  
I am currently using the latest go2_sdk, which support 1.1.1, but the firmware on my Go2 Pro is 1.0.25, will this be a potential reason for the problem?

abizovnuralem, 2024-10-12:
you should use previous commits with 1.0.25, because the latest is only support 1.1.1 firmware

abizovnuralem, 2024-10-17:
Due to inactivity, I am closing this thread. If you have additional questions, please open a new issue, thanks!

closed 2024-10-17 by abizovnuralem