abizovnuralem/go2_ros2_sdk #98: Why ROS2 can not find the python module?

is3rlab, 2024-10-22:
Hello，

I installed the package in the Pycharm Conda environment. However, when I launch the go2_robot_sdk robot.launch.py, I can not the python modules such as "aiortc", "open3d", correctly. I try several methods to set the path, it did not work. Please check more details as below:

IP list: ['']
[INFO] [robot_state_publisher-1]: process started with pid [54100]
[INFO] [pointcloud_to_laserscan_node-2]: process started with pid [54102]
[INFO] [go2_driver_node-3]: process started with pid [54104]
[INFO] [lidar_to_pointcloud-4]: process started with pid [54106]
[INFO] [rviz2-5]: process started with pid [54108]
[INFO] [joy_node-6]: process started with pid [54110]
[INFO] [teleop_node-7]: process started with pid [54112]
[INFO] [twist_mux-8]: process started with pid [54114]
[INFO] [foxglove_bridge-9]: process started with pid [54116]
[INFO] [async_slam_toolbox_node-10]: process started with pid [54118]
[INFO] [controller_server-11]: process started with pid [54120]
[INFO] [smoother_server-12]: process started with pid [54122]
[INFO] [planner_server-13]: process started with pid [54124]
[INFO] [behavior_server-14]: process started with pid [54126]
[INFO] [bt_navigator-15]: process started with pid [54128]
[INFO] [waypoint_follower-16]: process started with pid [54130]
[INFO] [velocity_smoother-17]: process started with pid [54132]
[INFO] [lifecycle_manager-18]: process started with pid [54134]
[teleop_node-7] [INFO] [1729619790.437410666] [TeleopTwistJoy]: Linear axis x on 1 at scale 0.200000.
[teleop_node-7] [INFO] [1729619790.437467305] [TeleopTwistJoy]: Linear axis y on 3 at scale 0.200000.
[teleop_node-7] [INFO] [1729619790.437472416] [TeleopTwistJoy]: Angular axis yaw on 6 at scale 0.200000.
[twist_mux-8] [INFO] [1729619790.438412433] [twist_mux]: Topic handler 'topics.joy' subscribed to topic 'cmd_vel_joy': timeout = 0.500000s , priority = 10.
[twist_mux-8] [INFO] [1729619790.440085606] [twist_mux]: Topic handler 'topics.navigation' subscribed to topic 'cmd_vel': timeout = 0.500000s , priority = 5.
[velocity_smoother-17] [INFO] [1729619790.444324242] [velocity_smoother]: 
[velocity_smoother-17]  velocity_smoother lifecycle node launched. 
[velocity_smoother-17]  Waiting on external lifecycle transitions to activate
[velocity_smoother-17]  See https://design.ros2.org/articles/node_lifecycle.html for more information.
[controller_server-11] [INFO] [1729619790.444678831] [controller_server]: 
[controller_server-11]  controller_server lifecycle node launched. 
[controller_server-11]  Waiting on external lifecycle transitions to activate
[controller_server-11]  See https://design.ros2.org/articles/node_lifecycle.html for more information.
[controller_server-11] [INFO] [1729619790.448720210] [controller_server]: Creating controller server
[robot_state_publisher-1] [INFO] [1729619790.449064340] [robot_state_publisher]: got segment FL_calf
[robot_state_publisher-1] [INFO] [1729619790.449157803] [robot_state_publisher]: got segment FL_calflower
[robot_state_publisher-1] [INFO] [1729619790.449170851] [robot_state_publisher]: got segment FL_calflower1
[robot_state_publisher-1] [INFO] [1729619790.449178619] [robot_state_publisher]: got segment FL_foot
[robot_state_publisher-1] [INFO] [1729619790.449185533] [robot_state_publisher]: got segment FL_hip
[robot_state_publisher-1] [INFO] [1729619790.449192125] [robot_state_publisher]: got segment FL_thigh
[robot_state_publisher-1] [INFO] [1729619790.449198504] [robot_state_publisher]: got segment FR_calf
[robot_state_publisher-1] [INFO] [1729619790.449204931] [robot_state_publisher]: got segment FR_calflower
[robot_state_publisher-1] [INFO] [1729619790.449211702] [robot_state_publisher]: got segment FR_calflower1
[robot_state_publisher-1] [INFO] [1729619790.449218105] [robot_state_publisher]: got segment FR_foot
[robot_state_publisher-1] [INFO] [1729619790.449224603] [robot_state_publisher]: got segment FR_hip
[robot_state_publisher-1] [INFO] [1729619790.449230942] [robot_state_publisher]: got segment FR_thigh
[robot_state_publisher-1] [INFO] [1729619790.449237343] [robot_state_publisher]: got segment Head_lower
[robot_state_publisher-1] [INFO] [1729619790.449243654] [robot_state_publisher]: got segment Head_upper
[robot_state_publisher-1] [INFO] [1729619790.449249801] [robot_state_publisher]: got segment RL_calf
[robot_state_publisher-1] [INFO] [1729619790.449256068] [robot_state_publisher]: got segment RL_calflower
[robot_state_publisher-1] [INFO] [1729619790.449262603] [robot_state_publisher]: got segment RL_calflower1
[robot_state_publisher-1] [INFO] [1729619790.449269096] [robot_state_publisher]: got segment RL_foot
[robot_state_publisher-1] [INFO] [1729619790.449275451] [robot_state_publisher]: got segment RL_hip
[robot_state_publisher-1] [INFO] [1729619790.449282287] [robot_state_publisher]: got segment RL_thigh
[robot_state_publisher-1] [INFO] [1729619790.449288903] [robot_state_publisher]: got segment RR_calf
[robot_state_publisher-1] [INFO] [1729619790.449295350] [robot_state_publisher]: got segment RR_calflower
[robot_state_publisher-1] [INFO] [1729619790.449301918] [robot_state_publisher]: got segment RR_calflower1
[robot_state_publisher-1] [INFO] [1729619790.449308386] [robot_state_publisher]: got segment RR_foot
[robot_state_publisher-1] [INFO] [1729619790.449314797] [robot_state_publisher]: got segment RR_hip
[robot_state_publisher-1] [INFO] [1729619790.449321495] [robot_state_publisher]: got segment RR_thigh
[robot_state_publisher-1] [INFO] [1729619790.449327996] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-1] [INFO] [1729619790.449334612] [robot_state_publisher]: got segment base_link
[robot_state_publisher-1] [INFO] [1729619790.449341157] [robot_state_publisher]: got segment front_camera
[robot_state_publisher-1] [INFO] [1729619790.449347678] [robot_state_publisher]: got segment imu
[robot_state_publisher-1] [INFO] [1729619790.449354427] [robot_state_publisher]: got segment map
[robot_state_publisher-1] [INFO] [1729619790.449361185] [robot_state_publisher]: got segment odom
[robot_state_publisher-1] [INFO] [1729619790.449367759] [robot_state_publisher]: got segment radar
[... 192 lines cut ...]
[behavior_server-14] [INFO] [1729619793.655156968] [behavior_server]: Activating drive_on_heading
[behavior_server-14] [INFO] [1729619793.655161905] [behavior_server]: Activating assisted_teleop
[behavior_server-14] [INFO] [1729619793.655166606] [behavior_server]: Activating wait
[behavior_server-14] [INFO] [1729619793.655175478] [behavior_server]: Creating bond (behavior_server) to lifecycle manager.
[lifecycle_manager-18] [INFO] [1729619793.757910077] [lifecycle_manager_navigation]: Server behavior_server connected with bond.
[lifecycle_manager-18] [INFO] [1729619793.757969643] [lifecycle_manager_navigation]: Activating bt_navigator
[bt_navigator-15] [INFO] [1729619793.758462935] [bt_navigator]: Activating
[bt_navigator-15] [INFO] [1729619793.792520815] [bt_navigator]: Creating bond (bt_navigator) to lifecycle manager.
[lifecycle_manager-18] [INFO] [1729619793.895822025] [lifecycle_manager_navigation]: Server bt_navigator connected with bond.
[lifecycle_manager-18] [INFO] [1729619793.895901293] [lifecycle_manager_navigation]: Activating waypoint_follower
[waypoint_follower-16] [INFO] [1729619793.896287416] [waypoint_follower]: Activating
[waypoint_follower-16] [INFO] [1729619793.896334483] [waypoint_follower]: Creating bond (waypoint_follower) to lifecycle manager.
[lifecycle_manager-18] [INFO] [1729619794.001139954] [lifecycle_manager_navigation]: Server waypoint_follower connected with bond.
[lifecycle_manager-18] [INFO] [1729619794.001204813] [lifecycle_manager_navigation]: Activating velocity_smoother
[velocity_smoother-17] [INFO] [1729619794.001466715] [velocity_smoother]: Activating
[velocity_smoother-17] [INFO] [1729619794.001517667] [velocity_smoother]: Creating bond (velocity_smoother) to lifecycle manager.
[lifecycle_manager-18] [INFO] [1729619794.107740008] [lifecycle_manager_navigation]: Server velocity_smoother connected with bond.
[lifecycle_manager-18] [INFO] [1729619794.107904238] [lifecycle_manager_navigation]: Managed nodes are active
[lifecycle_manager-18] [INFO] [1729619794.107973019] [lifecycle_manager_navigation]: Creating bond timer...
[controller_server-11] [WARN] [1729619800.885751996] [local_costmap.local_costmap]: Can't update static costmap layer, no map received
[... cut ...]
[controller_server-11] [WARN] [1729619810.885635841] [local_costmap.local_costmap]: Can't update static costmap layer, no map received
[planner_server-13] [WARN] [1729619812.318996828] [global_costmap.global_costmap]: Can't update static costmap layer, no map received
[controller_server-11] [WARN] [1729619820.885694708] [local_costmap.local_costmap]: Can't update static costmap layer, no map received
[planner_server-13] [WARN] [1729619822.324199985] [global_costmap.global_costmap]: Can't update static costmap layer, no map received

tfoldi, 2024-10-26:
did you update the submodules recursively? `aiortc` installed/included from source

tfoldi, 2024-10-26:
also, the `requirements.txt` in the top-level folder lists all these modules. did you install them in your conda env?

is3rlab, 2024-11-05:
> also, the `requirements.txt` in the top-level folder lists all these modules. did you install them in your conda env?

Yes, I installed all the requirements in the conda env. However, when I run the "ros2 launch go2_robot_sdk robot.launch.py", it can not find the module.

edugm94, 2025-02-13:
> > also, the `requirements.txt` in the top-level folder lists all these modules. did you install them in your conda env?
> 
> Yes, I installed all the requirements in the conda env. However, when I run the "ros2 launch go2_robot_sdk robot.launch.py", it can not find the module.

Hi, 

could you solve this issue?

tfoldi, 2025-02-14:
can you show the output of the `pip list | grep aiortc` before running `ros2 launch go2_robot_sdk robot.launch.py`

edugm94, 2025-02-14:
Hi @tfoldi ,

In the installation, it is mentioned: " For example, open3d does not yet support python3.12 and therefore you will need to set up a 3.11 venv first etc.". Can I run this project in a Conda environment? And if so, what are the requirements it should fulfill? 

Many thanks for considering my request.

tfoldi, 2025-02-14:
what OS version and ROS version are you using? probably it might be easier to run this as a docker container

closed 2025-05-08 by abizovnuralem

Leonana69, 2026-01-02:
I got the same problem when I tried to run the sdk with conda environment. It's the problem of which python is used by ROS2. If you look at the `ros2_ws/install/go2_robot_sdk/lib/go2_robot_sdk/go2_driver_node`, you will see `#!/usr/bin/python3` which means ROS2 will use your system python by default. So you need to change it to your conda python like `/home/username/miniconda3/envs/go2/bin/python3`. I don't know if there is other solution but I just use this Makefile script to replace the python string.
```
.PHONY: build

build:
	colcon build
	@{ \
		PYTHON=$$(which python3 2>/dev/null || which python 2>/dev/null || echo "/usr/bin/python3"); \
		echo "Detected Python: $$PYTHON"; \
		find ./install -type f -executable -exec grep -l "^#!/usr/bin/python3" {} \; 2>/dev/null | while read -r file; do \
			if [ -f "$$file" ]; then \
				echo "Updating shebang in: $$file"; \
				sed -i '1s|^#!/usr/bin/python3|#!'"$$PYTHON"'|' "$$file"; \
				dos2unix "$$file" 2>/dev/null || sed -i 's/\r$$//' "$$file"; \
				chmod +x "$$file"; \
			fi; \
		done; \
	}
```