abizovnuralem/go2_ros2_sdk #101: Does the unitree_api package exist?

kkjing0414, 2024-10-30:
Dear author, I'm sorry to bother you again, and thank you very much for your help before. When I ran it, it showed that I could not find the unitree_api package, I would like to ask the author if you have this package on GitHub?

abizovnuralem, 2024-10-30:
Hello! Can you please send me a full log of the error?

abizovnuralem, 2024-10-30:
How did you clone the repo? Have you done git clone --recurse-submodules https://github.com/abizovnuralem/go2_ros2_sdk.git src ?

kkjing0414, 2024-10-30:
[INFO] [launch]: All log files can be found below /home/unitree/.ros/log/2024-10-30-19-39-29-512138-ubuntu-4921
[INFO] [launch]: Default logging verbosity is set to INFO
IP list: ['192.168.55.112']
[INFO] [robot_state_publisher-1]: process started with pid [4922]
[INFO] [pointcloud_to_laserscan_node-2]: process started with pid [4924]
[INFO] [go2_driver_node-3]: process started with pid [4926]
[INFO] [lidar_to_pointcloud-4]: process started with pid [4928]
[INFO] [rviz2-5]: process started with pid [4930]
[INFO] [joy_node-6]: process started with pid [4932]
[INFO] [teleop_node-7]: process started with pid [4934]
[INFO] [twist_mux-8]: process started with pid [4938]
[INFO] [foxglove_bridge-9]: process started with pid [4972]
[INFO] [async_slam_toolbox_node-10]: process started with pid [4976]
[INFO] [controller_server-11]: process started with pid [4987]
[INFO] [smoother_server-12]: process started with pid [4997]
[INFO] [planner_server-13]: process started with pid [5002]
[INFO] [behavior_server-14]: process started with pid [5016]
[INFO] [bt_navigator-15]: process started with pid [5031]
[INFO] [waypoint_follower-16]: process started with pid [5033]
[INFO] [velocity_smoother-17]: process started with pid [5043]
[INFO] [lifecycle_manager-18]: process started with pid [5051]
[robot_state_publisher-1] [INFO] [1730288371.214761065] [robot_state_publisher]: got segment FL_calf
[robot_state_publisher-1] [INFO] [1730288371.215025023] [robot_state_publisher]: got segment FL_calflower
[robot_state_publisher-1] [INFO] [1730288371.215050433] [robot_state_publisher]: got segment FL_calflower1
[robot_state_publisher-1] [INFO] [1730288371.215060802] [robot_state_publisher]: got segment FL_foot
[robot_state_publisher-1] [INFO] [1730288371.215069219] [robot_state_publisher]: got segment FL_hip
[robot_state_publisher-1] [INFO] [1730288371.215076931] [robot_state_publisher]: got segment FL_thigh
[robot_state_publisher-1] [INFO] [1730288371.215085316] [robot_state_publisher]: got segment FR_calf
[robot_state_publisher-1] [INFO] [1730288371.215093093] [robot_state_publisher]: got segment FR_calflower
[robot_state_publisher-1] [INFO] [1730288371.215101061] [robot_state_publisher]: got segment FR_calflower1
[robot_state_publisher-1] [INFO] [1730288371.215109062] [robot_state_publisher]: got segment FR_foot
[robot_state_publisher-1] [INFO] [1730288371.215117255] [robot_state_publisher]: got segment FR_hip
[robot_state_publisher-1] [INFO] [1730288371.215124904] [robot_state_publisher]: got segment FR_thigh
[robot_state_publisher-1] [INFO] [1730288371.215132712] [robot_state_publisher]: got segment Head_lower
[robot_state_publisher-1] [INFO] [1730288371.215140361] [robot_state_publisher]: got segment Head_upper
[robot_state_publisher-1] [INFO] [1730288371.215148361] [robot_state_publisher]: got segment RL_calf
[robot_state_publisher-1] [INFO] [1730288371.215156362] [robot_state_publisher]: got segment RL_calflower
[robot_state_publisher-1] [INFO] [1730288371.215164203] [robot_state_publisher]: got segment RL_calflower1
[robot_state_publisher-1] [INFO] [1730288371.215171883] [robot_state_publisher]: got segment RL_foot
[robot_state_publisher-1] [INFO] [1730288371.215179884] [robot_state_publisher]: got segment RL_hip
[robot_state_publisher-1] [INFO] [1730288371.215187693] [robot_state_publisher]: got segment RL_thigh
[robot_state_publisher-1] [INFO] [1730288371.215195341] [robot_state_publisher]: got segment RR_calf
[robot_state_publisher-1] [INFO] [1730288371.215203246] [robot_state_publisher]: got segment RR_calflower
[robot_state_publisher-1] [INFO] [1730288371.215211311] [robot_state_publisher]: got segment RR_calflower1
[robot_state_publisher-1] [INFO] [1730288371.215219216] [robot_state_publisher]: got segment RR_foot
[robot_state_publisher-1] [INFO] [1730288371.215226832] [robot_state_publisher]: got segment RR_hip
[robot_state_publisher-1] [INFO] [1730288371.215234417] [robot_state_publisher]: got segment RR_thigh
[robot_state_publisher-1] [INFO] [1730288371.215242674] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-1] [INFO] [1730288371.215250866] [robot_state_publisher]: got segment base_link
[robot_state_publisher-1] [INFO] [1730288371.215259027] [robot_state_publisher]: got segment front_camera
[robot_state_publisher-1] [INFO] [1730288371.215266932] [robot_state_publisher]: got segment imu
[robot_state_publisher-1] [INFO] [1730288371.215275412] [robot_state_publisher]: got segment map
[robot_state_publisher-1] [INFO] [1730288371.215283061] [robot_state_publisher]: got segment odom
[robot_state_publisher-1] [INFO] [1730288371.215290966] [robot_state_publisher]: got segment radar
[teleop_node-7] [INFO] [1730288371.184742546] [TeleopTwistJoy]: Linear axis x on 1 at scale 0.200000.
[teleop_node-7] [INFO] [1730288371.184899360] [TeleopTwistJoy]: Linear axis y on 3 at scale 0.200000.
[teleop_node-7] [INFO] [1730288371.184913665] [TeleopTwistJoy]: Angular axis yaw on 6 at scale 0.200000.
[twist_mux-8] [INFO] [1730288371.228989377] [twist_mux]: Topic handler 'topics.joy' subscribed to topic 'cmd_vel_joy': timeout = 0.500000s , priority = 10.
[twist_mux-8] [INFO] [1730288371.234469427] [twist_mux]: Topic handler 'topics.navigation' subscribed to topic 'cmd_vel': timeout = 0.500000s , priority = 5.
[async_slam_toolbox_node-10] [INFO] [1730288371.272586617] [slam_toolbox]: Node using stack size 40000000
[bt_navigator-15] [INFO] [1730288371.400424485] [bt_navigator]:
[... cut ...] go, /home/unitree/ros2_ws/install/go2_interfaces, /home/unitree/ros2_ws/install/coco_detector, /opt/ros/humble]
[foxglove_bridge-9] [WARN] [1730288372.289172675] [foxglove_bridge]: Failed to add channel for topic "/api/sport/request" (unitree_api/msg/Request): package 'unitree_api' not found, searching: [/home/unitree/ros2_ws/install/go2_robot_sdk, /home/unitree/ros2_ws/install/unitree_go, /home/unitree/ros2_ws/install/go2_interfaces, /home/unitree/ros2_ws/install/coco_detector, /opt/ros/humble]

kkjing0414, 2024-10-30:
There is another problem, if I open the map a day ago, my dog always can't find the correct position of his own, the new map is more accurate, I would like to ask if this has something to do with the time, the location of the objects in my field has not changed.

kkjing0414, 2024-10-30:
Thank you very much for your help. Due to the time difference, I can't reply soon. Please forgive me.

kkjing0414, 2024-10-31:
[image]

kkjing0414, 2024-10-31:
My dog has a drifting problem. Have you ever had one？

abizovnuralem, 2024-11-11:
You should configure your nav2 params, I think some of the control issues (delays of the network and etc)

abizovnuralem, 2024-12-09:
@kkjing4444 , If you have additional questions, please reopen the issue, thanks!

closed 2024-12-09 by abizovnuralem