abizovnuralem/go2_ros2_sdk #102: Gazebo Simulation Integration

jasonjabbour, 2024-11-03:
Hi, thanks for the helpful repo! I would like to first run a simulation of the Go2 with this autonomy stack instead of using the physical Go2. Is there an integration with Gazebo that we could potentially use?

I have seen the [Isaac Sim repo](https://github.com/abizovnuralem/go2_omniverse) but I would like to stick with Gazebo due to other constraints of mine. 

Thanks!

tfoldi, 2024-11-05:
have you tried this https://github.com/anujjain-dev/unitree-go2-ros2 ?

jasonjabbour, 2024-11-05:
@tfoldi Thanks! Yea, I ended up settling on that repo. It only has a working teleoperated controller, so I'm currently working on adding autonomous navigation to the stack. Exploring the slam_toolbox and nav2. 

I'm also not sure how easy it will be to transfer that stack to the real Go2 quadruped once it's time, but I guess we will find out soon 🙃