abizovnuralem/go2_ros2_sdk #104: [global_costmap.global_costmap]: Can't update static costmap layer, no map received

Uchiwuwu, 2024-11-08:
Hi,

I try to run the robot.launch.py. The rviz and robot pops up fine but I don't recieve any map from my unitree. Do you know what is wrong with it?

`[INFO] [launch]: All log files can be found below /home/uchiwuwu/.ros/log/2024-11-08-09-41-53-364221-uchiwuwu-Inspiron-5593-16788
[INFO] [launch]: Default logging verbosity is set to INFO
IP list: ['192.168.24.73']
[INFO] [robot_state_publisher-1]: process started with pid [16789]
[INFO] [pointcloud_to_laserscan_node-2]: process started with pid [16791]
[INFO] [go2_driver_node-3]: process started with pid [16793]
[INFO] [lidar_to_pointcloud-4]: process started with pid [16795]
[INFO] [rviz2-5]: process started with pid [16797]
[INFO] [joy_node-6]: process started with pid [16799]
[INFO] [teleop_node-7]: process started with pid [16801]
[INFO] [twist_mux-8]: process started with pid [16803]
[INFO] [joint_state_publisher_gui-9]: process started with pid [16833]
[INFO] [foxglove_bridge-10]: process started with pid [16847]
[INFO] [async_slam_toolbox_node-11]: process started with pid [16849]
[INFO] [controller_server-12]: process started with pid [16859]
[INFO] [smoother_server-13]: process started with pid [16870]
[INFO] [planner_server-14]: process started with pid [16899]
[INFO] [behavior_server-15]: process started with pid [16913]
[INFO] [bt_navigator-16]: process started with pid [16933]
[INFO] [waypoint_follower-17]: process started with pid [16944]
[INFO] [velocity_smoother-18]: process started with pid [16946]
[INFO] [lifecycle_manager-19]: process started with pid [16969]
[robot_state_publisher-1] [INFO] [1731076914.211033696] [robot_state_publisher]: got segment FL_calf
[robot_state_publisher-1] [INFO] [1731076914.211151615] [robot_state_publisher]: got segment FL_calflower
[robot_state_publisher-1] [INFO] [1731076914.211164472] [robot_state_publisher]: got segment FL_calflower1
[robot_state_publisher-1] [INFO] [1731076914.211172185] [robot_state_publisher]: got segment FL_foot
[robot_state_publisher-1] [INFO] [1731076914.211179098] [robot_state_publisher]: got segment FL_hip
[robot_state_publisher-1] [INFO] [1731076914.211185830] [robot_state_publisher]: got segment FL_thigh
[robot_state_publisher-1] [INFO] [1731076914.211192532] [robot_state_publisher]: got segment FR_calf
[robot_state_publisher-1] [INFO] [1731076914.211199412] [robot_state_publisher]: got segment FR_calflower
[robot_state_publisher-1] [INFO] [1731076914.211206664] [robot_state_publisher]: got segment FR_calflower1
[robot_state_publisher-1] [INFO] [1731076914.211213981] [robot_state_publisher]: got segment FR_foot
[robot_state_publisher-1] [INFO] [1731076914.211220617] [robot_state_publisher]: got segment FR_hip
[robot_state_publisher-1] [INFO] [1731076914.211227307] [robot_state_publisher]: got segment FR_thigh
[robot_state_publisher-1] [INFO] [1731076914.211234271] [robot_state_publisher]: got segment Head_lower
[robot_state_publisher-1] [INFO] [1731076914.211241544] [robot_state_publisher]: got segment Head_upper
[robot_state_publisher-1] [INFO] [1731076914.211248059] [robot_state_publisher]: got segment RL_calf
[robot_state_publisher-1] [INFO] [1731076914.211254892] [robot_state_publisher]: got segment RL_calflower
[robot_state_publisher-1] [INFO] [1731076914.211262074] [robot_state_publisher]: got segment RL_calflower1
[robot_state_publisher-1] [INFO] [1731076914.211269456] [robot_state_publisher]: got segment RL_foot
[robot_state_publisher-1] [INFO] [1731076914.211276554] [robot_state_publisher]: got segment RL_hip
[robot_state_publisher-1] [INFO] [1731076914.211283891] [robot_state_publisher]: got segment RL_thigh
[robot_state_publisher-1] [INFO] [1731076914.211290861] [robot_state_publisher]: got segment RR_calf
[robot_state_publisher-1] [INFO] [1731076914.211297587] [robot_state_publisher]: got segment RR_calflower
[robot_state_publisher-1] [INFO] [1731076914.211305318] [robot_state_publisher]: got segment RR_calflower1
[robot_state_publisher-1] [INFO] [1731076914.211312207] [robot_state_publisher]: got segment RR_foot
[robot_state_publisher-1] [INFO] [1731076914.211319173] [robot_state_publisher]: got segment RR_hip
[robot_state_publisher-1] [INFO] [1731076914.211326564] [robot_state_publisher]: got segment RR_thigh
[robot_state_publisher-1] [INFO] [1731076914.211333649] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-1] [INFO] [1731076914.211340604] [robot_state_publisher]: got segment base_link
[robot_state_publisher-1] [INFO] [1731076914.211347381] [robot_state_publisher]: got segment front_camera
[robot_state_publisher-1] [INFO] [1731076914.211354610] [robot_state_publisher]: got segment imu
[robot_state_publisher-1] [INFO] [1731076914.211362539] [robot_state_publisher]: got segment map
[robot_state_publisher-1] [INFO] [1731076914.211369966] [robot_state_publisher]: got segment odom
[robot_state_publisher-1] [INFO] [1731076914.211377553] [robot_state_publisher]: got segment radar
[teleop_node-7] [INFO] [1731076914.204477662] [TeleopTwistJoy]: Linear axis x on 1 at scale 0.200000.
[teleop_node-7] [INFO] [1731076914.204534225] [TeleopTwistJoy]: Linear axis y on 3 at scale 0.200000.
[teleop_node-7] [INFO] [1731076914.204542006] [TeleopTwistJoy]: Angular axis yaw on 6 at scale 0.200000.
[twist_mux-8] [INFO] [1731076914.204574538] [twist_mux]: Topic handler 'topics.joy' subscribed to topic 'cmd_vel_joy': timeout = 0.500000s , priority = 10.
[twist_mux-8] [INFO] [1731076914.205104184] [twist_mux]: Topic handler 'topics.navigation' subscribed to topic 'cmd_vel': timeout = 0.500000s , priority = 5.
[foxglove_bridge-10] [INFO] [1731076914.232253058] [foxglove_bridge_component_manager]: Load Library: /opt/ros/humble/lib/libfoxglove_bridge_component.so
[foxglove_bridge-10] [INFO] [1731076914.240946380] [foxglove_bridge_component_manager]: Found class: rclcpp_components::NodeFactoryTemplate<foxglove_bridge::FoxgloveBridge>
[foxglove_bridge-10] [INFO] [1731076914.241009218] [foxglove_bridge_component_manager]: Instantiate class: rclcpp_components::NodeFactoryTemplate<foxglove_bridge::FoxgloveBridge>
[foxglove_bridge-10] [INFO] [1731076914.243158489] [foxglove_bridge]: Starting foxglove_bridge (humble, 0.8.0@) with WebSocket++/0.8.2
[foxglove_bridge-10] [INFO] [1731076914.243684146] [foxglove_bridge]: [WS] Server running without TLS
[foxglove_bridge-10] [INFO] [1731076914.243940106] [foxglove_bridge]: [WS] WebSocket server listening at ws://0.0.0.0:8765
[... 181 lines cut ...]
[behavior_server-15] [INFO] [1731076918.065474871] [behavior_server]: Activating assisted_teleop
[behavior_server-15] [INFO] [1731076918.065484604] [behavior_server]: Activating wait
[behavior_server-15] [INFO] [1731076918.065496893] [behavior_server]: Creating bond (behavior_server) to lifecycle manager.
[lifecycle_manager-19] [INFO] [1731076918.168148078] [lifecycle_manager_navigation]: Server behavior_server connected with bond.
[lifecycle_manager-19] [INFO] [1731076918.168197230] [lifecycle_manager_navigation]: Activating bt_navigator
[bt_navigator-16] [INFO] [1731076918.168406997] [bt_navigator]: Activating
[bt_navigator-16] [INFO] [1731076918.192791387] [bt_navigator]: Creating bond (bt_navigator) to lifecycle manager.
[lifecycle_manager-19] [INFO] [1731076918.295165847] [lifecycle_manager_navigation]: Server bt_navigator connected with bond.
[lifecycle_manager-19] [INFO] [1731076918.295208922] [lifecycle_manager_navigation]: Activating waypoint_follower
[waypoint_follower-17] [INFO] [1731076918.295445034] [waypoint_follower]: Activating
[waypoint_follower-17] [INFO] [1731076918.295476096] [waypoint_follower]: Creating bond (waypoint_follower) to lifecycle manager.
[lifecycle_manager-19] [INFO] [1731076918.397287181] [lifecycle_manager_navigation]: Server waypoint_follower connected with bond.
[lifecycle_manager-19] [INFO] [1731076918.397335633] [lifecycle_manager_navigation]: Activating velocity_smoother
[velocity_smoother-18] [INFO] [1731076918.397644359] [velocity_smoother]: Activating
[velocity_smoother-18] [INFO] [1731076918.397714559] [velocity_smoother]: Creating bond (velocity_smoother) to lifecycle manager.
[lifecycle_manager-19] [INFO] [1731076918.499855386] [lifecycle_manager_navigation]: Server velocity_smoother connected with bond.
[lifecycle_manager-19] [INFO] [1731076918.499901802] [lifecycle_manager_navigation]: Managed nodes are active
[lifecycle_manager-19] [INFO] [1731076918.499916378] [lifecycle_manager_navigation]: Creating bond timer...
[... cut ...]
[controller_server-12] [WARN] [1731076935.207575289] [local_costmap.local_costmap]: Can't update static costmap layer, no map received
[planner_server-14] [WARN] [1731076936.786239909] [global_costmap.global_costmap]: Can't update static costmap layer, no map received
[controller_server-12] [WARN] [1731076945.207514536] [local_costmap.local_costmap]: Can't update static costmap layer, no map received
[planner_server-14] [WARN] [1731076946.785546319] [global_costmap.global_costmap]: Can't update static costmap layer, no map received
`

tackgeun, 2024-11-10:
I'm also experiencing this issue time to time. Refer to issue #87 for related questions. It might be helpful to handle the issue.

First, type "ros2 topic list" and validate topics. If the connection is good, topics are similar to issue #87 and image from front camera should appear. If topics and camera image are valid, it is usually caused by inappropriate dependencies. Please do properly install the dependencies.

abizovnuralem, 2025-08-02:
I’ve refactored the project. If you encounter any problems, please feel free to reopen the issue.