abizovnuralem/go2_ros2_sdk #105: The path my dog takes is an S-curve shape, but the planned path is normal.

kkjing0414, 2024-11-10:
Who can help me, the path my dog takes is an S-curve shape, but the planned path is normal.
[image]
The green arrow is my own, from the beginning to the end.

abizovnuralem, 2024-11-11:
Hello, you should setup the correct nav2 params according to your situation, it looks like the problem with nav2 control service configs.

jam131, 2024-11-19:
I have the same problem, and I’ve been changing the settings for a while but I haven’t been able to find a good configuration. Have you found any specific setting that could be significantly affecting this issue? Thank you very much for your attention.

kkjing0414, 2024-11-19:
Thank you for finally having someone with the same problem as me, but unfortunately, I also tried for a long time, and it didn't work.

arvillion, 2024-11-19:
Does this spinning behavior originate from the recovery server when the robot is not able to find a path?

abizovnuralem, 2024-12-09:
Hello, dear @arvillion @jam131 @kkjing4444 , I just replied in this thread https://github.com/abizovnuralem/go2_ros2_sdk/issues/107#issuecomment-2527643103

closed 2025-05-08 by abizovnuralem