abizovnuralem/go2_ros2_sdk #107: Nav2Goal not working

amin994work, 2024-11-28:
Hello, Thank you for this the Go2 robot dog works well with controller and the lidar data output is amazing as well. I am facing some issues with the Nav2Goal, it is not working when I click and drag it to a goal it does not respond and the Go2 does not move. the log what i get is:
[1732788468.337890632] [rviz_navigation_dialog_action_client]: NavigateToPose will be called using the BT Navigator's default behavior tree.
[rviz2-5] [ERROR] [1732788468.340055419] [rviz_navigation_dialog_action_client]: Goal was rejected by server

Just want to know what am I doing wrong. It will be really great if you can assist me. Thank you.

abizovnuralem, 2024-12-09:
Hello, nav2 is an experimental feature and needs more time to adjust some params, sorry for that, @amin994work

amin994work, 2024-12-10:
Thank you for the reply, I understand.

edugm94, 2025-03-03:
Hi @abizovnuralem ,

Is there any update in this regard? Thanks so much for your response

abizovnuralem, 2025-08-02:
I’ve refactored the project. If you encounter any problems, please feel free to reopen the issue.

closed 2025-08-02 by abizovnuralem