abizovnuralem/go2_ros2_sdk #108: Can't update static costmap layer, no map received. Connecting to the GO2 Robot Via webrtc

Francesco-Pagano, 2024-12-06:
I followed the guide, on a Unitree Go2 EDU Plus, after running the commands:

- source install/setup.bash
- export ROBOT_IP="my_robot_ip"
- export CONN_TYPE="webrtc"
- ros2 launch go2_robot_sdk robot.launch.py

I get the error "Can't update static costmap layer, no map received."
Also, on rviz I get the error on RobotModel (as shown in the attached image)
The model in rviz does not move and I do not receive any points with the lidar.
The only thing I can get is the camera streaming.

[image]

abizovnuralem, 2024-12-09:
Hello, can you please give me more details? Are you using the robot's wifi IP address? How did you get it? have you ping it in the terminal?

Francesco-Pagano, 2024-12-09:
Yes, we are using the robot ip address by exporting the ROBOT_IP variable as shown in the readme.
We got the address from the mobile application, which is 192.168.123.221.
We can also see another address on the network which is 192.168.123.18.
We can ping both addresses, but can only connect through ssh to 192.168.123.18.
We think that the connection works because we are able to see the streaming of the camera when using 192.168.123.221

leonardofalanga, 2024-12-28:
Hi,
I have the same problem... Are there any solutions?

abizovnuralem, 2025-01-17:
Do you robot ip is 192.168.123.221, have you tried to play with rviz ? Also, can you print the ros2 topics list ?

jackhou66, 2025-03-27:
> Yes, we are using the robot ip address by exporting the ROBOT_IP variable as shown in the readme. We got the address from the mobile application, which is 192.168.123.221. We can also see another [...]

have you work it out , the same issue

jackhou66, 2025-03-27:
> Your robot ip is 192.168.123.221, have you tried to play with rviz ? Also, can you print the ros2 topics list ?

my situation is  the robot and lap can connect, but no any topic publish ,and rviz2 can be opened when running . just no any published messages  ,,,,via CONN_TYPE="webrtc"

DaojiePENG, 2025-03-27:
I also encontered the same issue details are as follows: 
I use: `export ROBOT_IP="192.168.43.109"
export CONN_TYPE="webrtc"`
I get the ip from the go2's app and I am able to ping the ip: 

[image]

```
(env_unitree_go2) daojie@daojie-ASUS:~/ros2_ws$ ping 192.168.43.109
PING 192.168.43.109 (192.168.43.109) 56(84) bytes of data.
64 bytes from 192.168.43.109: icmp_seq=1 ttl=64 time=11.8 ms
64 bytes from 192.168.43.109: icmp_seq=2 ttl=64 time=11.9 ms
64 bytes from 192.168.43.109: icmp_seq=3 ttl=64 time=9.05 ms
64 bytes from 192.168.43.109: icmp_seq=4 ttl=64 time=57.0 ms
64 bytes from 192.168.43.109: icmp_seq=5 ttl=64 time=10.7 ms
64 bytes from 192.168.43.109: icmp_seq=6 ttl=64 time=10.3 ms
64 bytes from 192.168.43.109: icmp_seq=7 ttl=64 time=14.5 ms
64 bytes from 192.168.43.109: icmp_seq=8 ttl=64 time=10.2 ms
64 bytes from 192.168.43.109: icmp_seq=9 ttl=64 time=7.96 ms
64 bytes from 192.168.43.109: icmp_seq=10 ttl=64 time=14.0 ms
^C
--- 192.168.43.109 ping statistics ---
10 packets transmitted, 10 received, 0% packet loss, time 9010ms
rtt min/avg/max/mdev = 7.959/15.737/57.019/13.891 ms
(env_unitree_go2) daojie@daojie-ASUS:~/ros2_ws$ 

```

```
(env_unitree_go2) daojie@daojie-ASUS:~/ros2_ws$ ros2 launch go2_robot_sdk robot.launch.py
[INFO] [launch]: All log files can be found below /home/daojie/.ros/log/2025-03-27-23-43-39-969861-daojie-ASUS-394234
[INFO] [launch]: Default logging verbosity is set to INFO
IP list: ['192.168.43.109']
[INFO] [robot_state_publisher-1]: process started with pid [394235]
[INFO] [pointcloud_to_laserscan_node-2]: process started with pid [394237]
[INFO] [go2_driver_node-3]: process started with pid [394239]
[INFO] [lidar_to_pointcloud-4]: process started with pid [394241]
[INFO] [rviz2-5]: process started with pid [394243]
[INFO] [joy_node-6]: process started with pid [394245]
[INFO] [teleop_node-7]: process started with pid [394247]
[INFO] [twist_mux-8]: process started with pid [394249]
[INFO] [foxglove_bridge-9]: process started with pid [394251]
[INFO] [async_slam_toolbox_node-10]: process started with pid [394253]
[INFO] [controller_server-11]: process started with pid [394255]
[INFO] [smoother_server-12]: process started with pid [394257]
[INFO] [planner_server-13]: process started with pid [394259]
[INFO] [behavior_server-14]: process started with pid [394261]
[INFO] [bt_navigator-15]: process started with pid [394263]
[INFO] [waypoint_follower-16]: process started with pid [394265]
[INFO] [velocity_smoother-17]: process started with pid [394267]
[INFO] [lifecycle_manager-18]: process started with pid [394269]
[twist_mux-8] [INFO] [1743090220.159596503] [twist_mux]: Topic handler 'topics.joy' subscribed to topic 'cmd_vel_joy': timeout = 0.500000s , priority = 10.
[twist_mux-8] [INFO] [1743090220.161305276] [twist_mux]: Topic handler 'topics.navigation' subscribed to topic 'cmd_vel': timeout = 0.500000s , priority = 5.
[robot_state_publisher-1] [INFO] [1743090220.166168328] [robot_state_publisher]: got segment FL_calf
[robot_state_publisher-1] [INFO] [1743090220.166253840] [robot_state_publisher]: got segment FL_calflower
[robot_state_publisher-1] [INFO] [1743090220.166276202] [robot_state_publisher]: got segment FL_calflower1
[robot_state_publisher-1] [INFO] [1743090220.166281261] [robot_state_publisher]: got segment FL_foot
[robot_state_publisher-1] [INFO] [1743090220.166285930] [robot_state_publisher]: got segment FL_hip
[robot_state_publisher-1] [INFO] [1743090220.166290218] [robot_state_publisher]: got segment FL_thigh
[robot_state_publisher-1] [INFO] [1743090220.166294967] [robot_state_publisher]: got segment FR_calf
[robot_state_publisher-1] [INFO] [1743090220.166303103] [robot_state_publisher]: got segment FR_calflower
[robot_state_publisher-1] [INFO] [1743090220.166308082] [robot_state_publisher]: got segment FR_calflower1
[robot_state_publisher-1] [INFO] [1743090220.166313011] [robot_state_publisher]: got segment FR_foot
[robot_state_publisher-1] [INFO] [1743090220.166318381] [robot_state_publisher]: got segment FR_hip
[robot_state_publisher-1] [INFO] [1743090220.166323782] [robot_state_publisher]: got segment FR_thigh
[robot_state_publisher-1] [INFO] [1743090220.166328891] [robot_state_publisher]: got segment Head_lower
[robot_state_publisher-1] [INFO] [1743090220.166333209] [robot_state_publisher]: got segment Head_upper
[robot_state_publisher-1] [INFO] [1743090220.166337678] [robot_state_publisher]: got segment RL_calf
[robot_state_publisher-1] [INFO] [1743090220.166343078] [robot_state_publisher]: got segment RL_calflower
[robot_state_publisher-1] [INFO] [1743090220.166354510] [robot_state_publisher]: got segment RL_calflower1
[... 201 lines cut ...]
[lifecycle_manager-18] [INFO] [1743090230.659856741] [lifecycle_manager_navigation]: Managed nodes are active
[lifecycle_manager-18] [INFO] [1743090230.659863934] [lifecycle_manager_navigation]: Creating bond timer...
[controller_server-11] [WARN] [1743090237.419934014] [local_costmap.local_costmap]: Can't update static costmap layer, no map received
[planner_server-13] [WARN] [1743090238.861498974] [global_costmap.global_costmap]: Can't update static costmap layer, no map received
[controller_server-11] [WARN] [1743090247.419956298] [local_costmap.local_costmap]: Can't update static costmap layer, no map received
[planner_server-13] [WARN] [1743090248.867332755] [global_costmap.global_costmap]: Can't update static costmap layer, no map received
[... cut ...]
[planner_server-13] [WARN] [1743090288.874540278] [global_costmap.global_costmap]: Can't update static costmap layer, no map received
[controller_server-11] [WARN] [1743090297.419948381] [local_costmap.local_costmap]: Can't update static costmap layer, no map received
[planner_server-13] [WARN] [1743090298.880966666] [global_costmap.global_costmap]: Can't update static costmap layer, no map received
[controller_server-11] [WARN] [1743090307.419942907] [local_costmap.local_costmap]: Can't update static costmap layer, no map received

```

The rviz looks like this:

[image]

Any ideas to solve this issue?

DMeng-max, 2025-05-17:
@DaojiePENG 你们解决这个问题了吗，我们也遇到了这个问题

DMeng-max, 2025-05-17:
@DaojiePENG  We also had this problem. Did you solve it

lambdavi, 2025-05-19:
Following, same problem! @DaojiePENG

Francesco-Pagano, 2025-05-19:
> [@DaojiePENG](https://github.com/DaojiePENG) We also had this problem. Did you solve it

No, we haven't solved it yet

FishDentistry, 2025-05-22:
Same problem here!

Francesco-Pagano, 2025-05-29:
> We meet the same issue with [@Francesco-Pagano](https://github.com/Francesco-Pagano) , any ideas to solve?

Honestly, I have no idea.

> [@DaojiePENG](https://github.com/DaojiePENG) , it is literally in the error logs:
> 
> ```
> [go2_driver_node-3] aioice submodule is not initalized. please init submodules recursively
> ```
> 
> can you double confirm that you initalized the submodules per README?

But here I have a different log and it connects correctly to the aioice

closed 2025-08-02 by abizovnuralem