abizovnuralem/go2_ros2_sdk #111: Foxglove not showing map

hchlhwang, 2024-12-27:
I am getting a channel schema parsing failure message when visualizing via Foxglove. As seen in the attached image, it says "Unsupported message encoding." for /utlidar/map_state, /utlidar/voxel_map_compressed, and /config_change_status. 
I believe this may be related to a custom message setup. How did you manage to visualize the map using Foxglove as shown in the README? I would appreciate any guidance. Thanks!

[image]

### Steps to reproduce
```
source install/setup.bash
export ROBOT_IP=192.168.0.XXX
export CONN_TYPE=webrtc
ros2 launch go2_robot_sdk robot.launch.py
```
### System Info
- Local (Ubuntu 20.04 - amd)
- Remote (Jetson AGX Orin - Ubuntu 22.04 - arm)
- Connection type: webrtc

abizovnuralem, 2025-01-17:
Hello! Compressed map
Cannot be accessed because it is encoded , you should use decoded one

tfoldi, 2025-03-01:
use the decoded point cloud instead of the raw compressed voxel grids. that's how it is done in the image shown in README

closed 2025-03-01 by tfoldi