abizovnuralem/go2_ros2_sdk #112: Three bugs solution:

weiluduan2, 2025-01-02:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

A clear and concise description of what the bug is.

### Steps to reproduce

Please try to provide a minimal example of how to reproduce the bug. Error messages and stack traces are also helpful.

### System Info

Describe the characteristics of your environment:

- Firmware version of the dog
- Connection type (webrtc or cyclonedds)
- OS: [e.g. Ubuntu 20.04]
- Wifi version

### Additional context

Add any other context about the problem here.

### Checklist

- [ ] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Ros2 itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Criteria 1
- [ ] Criteria 2

weiluduan2, 2025-01-02:
I write four solutions for bugs of these project, most of them are because of Nav2 parameters setting and Launch code.

 Here is: https://github.com/weiluduan2/Debug-record/blob/main/Unofficial%20go2%20SDK/readme.md

These bugs' key points are about:  Global map drift, Navigation performance bad, robot can't rotate when doing navigation, Wrong launch file in robot.launch which cause amcl and map server not working, slam closure when use pre-built map.

And I provide the correct code and  Nav2 parameters file to help solve them. After test, my unitree go2 pro can correctly do navigation and mapping, and there is no any break or fail when it working.

abizovnuralem, 2025-01-17:
Hello @weiluduan2 , thanks for your great work! Can you create a pr for this, please? I will be happy to merge it

DedSecer, 2025-01-21:
Hello, I also trying nav2 on go2, I found the robot will always turn into recovery behavior as terminal out like that:
```
[planner_server-12] [ERROR] [1737460376.231322823] [transformPoseInTargetFrame]: Extrapolation Error looking up target frame: Lookup would require extrapolation into the past.  Requested time 1737460354.365724 but the earliest data is at time 1737460366.134330, when looking up transform from frame [odom] to frame [map]
[planner_server-12] 
[planner_server-12] [WARN] [1737460376.231352102] [planner_server]: Could not transform the start or goal pose in the costmap frame
[planner_server-12] [WARN] [1737460376.231376352] [planner_server]: [compute_path_to_pose] [ActionServer] Aborting handle.
[planner_server-12] [INFO] [1737460376.247054312] [global_costmap.global_costmap]: Received request to clear entirely the global_costmap
[planner_server-12] [ERROR] [1737460376.811317973] [transformPoseInTargetFrame]: Extrapolation Error looking up target frame: Lookup would require extrapolation into the past.  Requested time 1737460354.365724 but the earliest data is at time 1737460366.134330, when looking up transform from frame [odom] to frame [map]
[planner_server-12] 
[planner_server-12] [WARN] [1737460376.811390971] [planner_server]: Could not transform the start or goal pose in the costmap frame
[planner_server-12] [WARN] [1737460376.811450712] [planner_server]: [compute_path_to_pose] [ActionServer] Aborting handle.
[controller_server-10] [INFO] [1737460376.827195312] [local_costmap.local_costmap]: Received request to clear entirely the local_costmap
[planner_server-12] [INFO] [1737460376.827482217] [global_costmap.global_costmap]: Received request to clear entirely the global_costmap
[planner_server-12] [ERROR] [1737460377.141445833] [transformPoseInTargetFrame]: Extrapolation Error looking up target frame: Lookup would require extrapolation into the past.  Requested time 1737460354.365724 but the earliest data is at time 1737460368.161538, when looking up transform from frame [odom] to frame [map]
[planner_server-12] 
[planner_server-12] [WARN] [1737460377.141648228] [planner_server]: Could not transform the start or goal pose in the costmap frame
[planner_server-12] [WARN] [1737460377.141847811] [planner_server]: [compute_path_to_pose] [ActionServer] Aborting handle.
[planner_server-12] [INFO] [1737460377.157099607] [global_costmap.global_costmap]: Received request to clear entirely the global_costmap
[planner_server-12] [ERROR] [1737460377.480890699] [transformPoseInTargetFrame]: Extrapolation Error looking up target frame: Lookup would require extrapolation into the past.  Requested time 1737460354.365724 but the earliest data is at time 1737460368.161538, when looking up transform from frame [odom] to frame [map]
[planner_server-12] 
[planner_server-12] [WARN] [1737460377.480913142] [planner_server]: Could not transform the start or goal pose in the costmap frame
[planner_server-12] [WARN] [1737460377.480935572] [planner_server]: [compute_path_to_pose] [ActionServer] Aborting handle.
[behavior_server-13] [INFO] [1737460377.497453298] [behavior_server]: Running spin
[behavior_server-13] [INFO] [1737460377.497533602] [behavior_server]: Turning 1.57 for spin behavior.
[go2_driver_node-3] [INFO] [1737460377.580117371] [go2_driver_node]: Move
[go2_driver_node-3] [INFO] [1737460377.681666150] [go2_driver_node]: Move
[go2_driver_node-3] [INFO] [1737460377.782669764] [go2_driver_node]: Move
[go2_driver_node-3] [INFO] [1737460377.969538955] [go2_driver_node]: Move
[go2_driver_node-3] [INFO] [1737460378.097328991] [go2_driver_node]: Move
[go2_driver_node-3] [INFO] [1737460378.198695335] [go2_driver_node]: Move
[go2_driver_node-3] [INFO] [1737460378.299705065] [go2_driver_node]: Move
[go2_driver_node-3] [INFO] [1737460378.519069797] [go2_driver_node]: Move
[go2_driver_node-3] [INFO] [1737460378.620124623] [go2_driver_node]: Move
[go2_driver_node-3] [INFO] [1737460378.720910359] [go2_driver_node]: Move
[go2_driver_node-3] [INFO] [1737460378.821978669] [go2_driver_node]: Move
[go2_driver_node-3] [INFO] [1737460378.928622259] [go2_driver_node]: Move
[go2_driver_node-3] [INFO] [1737460379.029439620] [go2_driver_node]: Move
[go2_driver_node-3] [INFO] [1737460379.131509372] [go2_driver_node]: Move
[go2_driver_node-3] [INFO] [1737460379.232295183] [go2_driver_node]: Move
[go2_driver_node-3] [INFO] [1737460379.416125828] [go2_driver_node]: Move
[go2_driver_node-3] [INFO] [1737460379.517591738] [go2_driver_node]: Move
[go2_driver_node-3] [INFO] [1737460379.620328410] [go2_driver_node]: Move
[behavior_server-13] [INFO] [1737460379.697680769] [behavior_server]: spin completed successfully

```
It will happen on the code from this repo or the code changed by @weiluduan2 .

DedSecer, 2025-01-22:
> Hello, I also trying nav2 on go2, I found the robot will always turn into recovery behavior as terminal out like that:
>
> ```
> [planner_server-12] [ERROR] [1737460376.231322823] [...]

sorry, this happen when I use "global_frame: map" under global_costmap section in nav2_params.yaml.

weiluduan2, 2025-01-23:
> sorry, this happen when I use "global_frame: map" under global_costmap section in nav2_params.yaml.

Sorry for late reply. I think you will meet another problem if you set global frame as odom or other things. 
Actually, it is true to use map as global frame, because the update of global costmap will be calculated to "map" which will ensure global costmap can match the map. If you use odom as global frame, the error will be accumulated as time goes by, although the robotic dog can do navigation at first,  it will fail for a long time navigation.

The real problem is lidar. 
Because lidar won't publish data with correct time stamp,  AMCL will stop update the transform from global costmap to map(global frame). So planner can't plan the path. 
So why, why lidar is wrong. 
Because qos is wrong. The original project use default qos which is "reliable"（https://docs.ros.org/en/rolling/Concepts/Intermediate/About-Quality-of-Service-Settings.html）
That means if some data missed, your ros will try to ask robotic dog to transmit it again, which will cause the remaining lidar data missed. And that's why AMCL can't provide transform to planner, because lidar transmission is retrying.

So, you need to change qos  of lidar transmission to "best effort". 
        lidar_qos = QoSProfile(
            reliability=QoSReliabilityPolicy.BEST_EFFORT,
            history=QoSHistoryPolicy.KEEP_LAST,
            depth=1
        )

 self.go2_lidar_pub.append(self.create_publisher(
                PointCloud2, 'point_cloud2', lidar_qos))

You can find the code in my repositories.(https://github.com/weiluduan2/Debug-record/blob/main/Unofficial%20go2%20SDK/go2_driver_node.py) . Actually , I think it is better to change all of my three file to your project. There are still other problems.

DedSecer, 2025-01-23:
Thanks for your reply. But I tried to use 3 files in your repo but it the program raised error. It seems you have define a new function `gen_topic_command` in go2_robot_sdk/scripts/go2_func.py, but I couldn't find the new function.

DedSecer, 2025-01-23:
Now I just changed the global frame in "amcl","bt_navigator","global_costmap", and the qos of lidar. It will raise the error above as I send a navigation goal.

weiluduan2, 2025-01-23:
> Now I just changed the global frame in "amcl","bt_navigator","global_costmap", and the qos of lidar as your code. It will raise the error above as I send a navigation goal.

It doesn't matter, you can just comment it or delete the code using gen_topic_command, it is a test function for me, it won't affect the navigation.

DedSecer, 2025-01-23:
Unfortunately I still cannot run the code successfully. It seems you have created a map and published it by map server. When I use previous config of map server in `nav2_params.yaml` and send a nav goal, it outs continuously:
```
[amcl-18] [WARN] [1737634262.016429662] [amcl]: Waiting for map....
[planner_server-10] [WARN] [1737634262.436850418] [nav2_costmap_2d]: Robot is out of bounds of the costmap!
[controller_server-8] [WARN] [1737634262.570710660] [local_costmap.local_costmap]: Can't update static costmap layer, no map received
[planner_server-10] [WARN] [1737634262.936707274] [nav2_costmap_2d]: Robot is out of bounds of the costmap!
[rviz2-16] [INFO] [1737634263.181892503] [rviz_navigation_dialog_action_client]: NavigateToPose will be called using the BT Navigator's default behavior tree.
[bt_navigator-12] [INFO] [1737634263.182448544] [bt_navigator]: Begin navigating from current location (1.52, -0.48) to (-0.16, 0.57)
[planner_server-10] [WARN] [1737634263.436700575] [nav2_costmap_2d]: Robot is out of bounds of the costmap!
[controller_server-8] [WARN] [1737634263.570652287] [local_costmap.local_costmap]: Can't update static costmap layer, no map received
[planner_server-10] [WARN] [1737634263.936694855] [nav2_costmap_2d]: Robot is out of bounds of the costmap!
[planner_server-10] [WARN] [1737634264.437003487] [nav2_costmap_2d]: Robot is out of bounds of the costmap!
[amcl-18] [WARN] [1737634264.492282269] [amcl]: Waiting for map....
[controller_server-8] [WARN] [1737634264.570710017] [local_costmap.local_costmap]: Can't update static costmap layer, no map received
[planner_server-10] [WARN] [1737634264.936755342] [nav2_costmap_2d]: Robot is out of bounds of the costmap!
[planner_server-10] [WARN] [1737634265.436692694] [nav2_costmap_2d]: Robot is out of bounds of the costmap!
[controller_server-8] [WARN] [1737634265.570693054] [local_costmap.local_costmap]: Can't update static costmap layer, no map received
```

DedSecer, 2025-01-25:
> sorry, this happen when I use "global_frame: map" under global_costmap section in nav2_params.yaml.

I finally found there's a delay between `map` and `odom` transform which is published by slam_toolbox, but the publish period has been set to 0.02 in slam_toolbox config. 
I increase the `transform_timout` and it seems to be work. But the localization not good, the map become messy as I walking the robot around.

tfoldi, 2025-02-18:
> The real problem is lidar. Because lidar won't publish data with correct time stamp, AMCL will stop update the transform from global costmap to map(global frame). So planner can't plan the path. So [...]

This has been fixed, both images and point clouds are now with `BEST_EFFORT` with option to redefine them at startup time. hope that helps

closed 2025-05-08 by abizovnuralem