abizovnuralem/go2_ros2_sdk #115: How to develop the project in unitree go2 with its Accessory NVIDIA Jetson Orin(ubuntu20 foxy)

7yzx, 2025-01-17:
I have an EDU go2, but this repo is for ubuntu22 humble. I not sure how to use this project on the EDU go2. I think there are two options:

1. Continue using the Accessory jetson orin to run Foxy code, then deploy Ubuntu 22.04 Humble on another computer. Is there any issue with communication between Foxy and Humble?
 
2. Remove the jetson orin and replace it with a new computer running Ubuntu 22. (I know that Unitree Office does not provide an Ubuntu 22 image.) 

Could you please offer some help? Thank you very much!

abizovnuralem, 2025-01-17:
Hello, you can update your Nvidia jetson to the latest jetpack that has Ubuntu 22 and supports humble

abizovnuralem, 2025-01-17:
1. Yes, an original Nvidia image , but you need to ask Unitree support for help 2. It has the same functionality + nav2 support and some small improvements

7yzx, 2025-01-17:
Thanks a lot!!!

closed 2025-01-17 by 7yzx

7yzx, 2025-01-18:
Sorry to bother you again. Unitree's official response is that there is no Ubuntu 22 image. I also tried using the official NVIDIA image, but it was unsuccessful. Here’s a [related](https://forums.developer.nvidia.com/t/nvidia-jetson-orin-nano-8-gb-ssd-image/306949/12) link. I would like to know, when you were developing this, did you use an external computer or the original onboard computer

DedSecer, 2025-01-18:
I have also tried that, unfortunately there is no screen output from type-c now.

7yzx, 2025-01-19:
> I have also tried that, unfortunately there is no screen output from type-c now.

My problem of using Nvidia image is that the ssd cannot be recognized. Then I flash the office image of unitree, there is no screen output from type-c after the startup interface. Have you tried to revert to the original version? Or do you know how to recover?

DedSecer, 2025-01-19:
> My problem of using Nvidia image is that the ssd cannot be recognized. Then I flash the office image of unitree, there is no screen output from type-c after the startup interface. Have you tried to [...]

I fresh the official image download from netdisk, the jetson work correctly now.

abizovnuralem, 2025-01-19:
> Sorry to bother you again. Unitree's official response is that there is no Ubuntu 22 image. I also tried using the official NVIDIA image, but it was unsuccessful. Here’s a [...]

I use external Nvidia agx board and 3d printed holder for my go2 pro

kExU9853, 2025-06-10:
Hi 7yzx,
Thank you for the question and all reply! That helps a lot for me. Just wanna double check for this question, did you flash the SSD with offical image from netdisk to use the Ubuntu22.04?