abizovnuralem/go2_ros2_sdk #124: Robot control failed

edugm94, 2025-02-14:
Hello, 

Firstly, thanks for this nice project!

I have installed it in my machine, and there was a smooth installation of all pacjkages.I am encountering the following issue. 

Initially I have run `ros2 launch go2_robot_sdk robot.launch.py`. I checked the list of topics. I tried to do a simple test by using the topic `/cmd_vel`. 

In one terminal, I run: `ros2 topic echo /cmd_vel`; and in a second terminal I run: `ros2 topic pub /cmd_vel geometry_msgs/msg/Twist '{linear: {x: 0.3, y: 0.0, z: 0.0}, angular: {x: 0.0, y: 0.0, z: 0.1}}'`. 

I can in the first terminal the messages sent in the second terminal. However, the robot does not move. 

In all terminals, I first run the following commands: 

`source install/setup.bash
export ROBOT_IP="robot_ip"
export CONN_TYPE="webrtc"
`

I ensured that I am connected to the same network the robot is connected. I checked through the App. 

Is there something I am doing wrong? 
Thanks for any support!

edugm94, 2025-02-14:
Hi,

I checked rviz and realised the RobotModel looks like this: 

[image]

I am running this project in a machine with: 
- Ubuntu 22
-  ROS2 Humble 

Any support is really appreciated!

tfoldi, 2025-02-14:
share the logs please. make sure your robot is on `STA-L` network and you set ROBOT_IP correctly

tfoldi, 2025-02-17:
thank you. can you share the logs from the `ros2 launch` as well?

edugm94, 2025-02-17:
Okay. Here it is: 

`ros2 launch go2_robot_sdk robot.launch.py      `                                                                                                                                                               
```
[INFO] [launch]: All log files can be found below /home/eduardo/.ros/log/2025-02-17-09-46-53-836872-Edu-17533
[INFO] [launch]: Default logging verbosity is set to INFO
IP list: ['']
[INFO] [robot_state_publisher-1]: process started with pid [17534]
[INFO] [pointcloud_to_laserscan_node-2]: process started with pid [17536]
[INFO] [go2_driver_node-3]: process started with pid [17538]
[INFO] [lidar_to_pointcloud-4]: process started with pid [17540]
[INFO] [rviz2-5]: process started with pid [17542]
[INFO] [joy_node-6]: process started with pid [17544]
[INFO] [teleop_node-7]: process started with pid [17546]
[INFO] [twist_mux-8]: process started with pid [17548]
[INFO] [foxglove_bridge-9]: process started with pid [17550]
[INFO] [async_slam_toolbox_node-10]: process started with pid [17552]
[INFO] [controller_server-11]: process started with pid [17554]
[INFO] [smoother_server-12]: process started with pid [17556]
[INFO] [planner_server-13]: process started with pid [17558]
[INFO] [behavior_server-14]: process started with pid [17560]
[INFO] [bt_navigator-15]: process started with pid [17562]
[INFO] [waypoint_follower-16]: process started with pid [17564]
[INFO] [velocity_smoother-17]: process started with pid [17566]
[INFO] [lifecycle_manager-18]: process started with pid [17568]
[twist_mux-8] [INFO] [1739782014.323173582] [twist_mux]: Topic handler 'topics.joy' subscribed to topic 'cmd_vel_joy': timeout = 0.500000s , priority = 10.
[twist_mux-8] [INFO] [1739782014.329890900] [twist_mux]: Topic handler 'topics.navigation' subscribed to topic 'cmd_vel': timeout = 0.500000s , priority = 5.
[velocity_smoother-17] [INFO] [1739782014.331220974] [velocity_smoother]: 
[velocity_smoother-17] 	velocity_smoother lifecycle node launched. 
[velocity_smoother-17] 	Waiting on external lifecycle transitions to activate
[velocity_smoother-17] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[behavior_server-14] [INFO] [1739782014.338486970] [behavior_server]: 
[behavior_server-14] 	behavior_server lifecycle node launched. 
[behavior_server-14] 	Waiting on external lifecycle transitions to activate
[behavior_server-14] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[planner_server-13] [INFO] [1739782014.339223162] [planner_server]: 
[planner_server-13] 	planner_server lifecycle node launched. 
[planner_server-13] 	Waiting on external lifecycle transitions to activate
[planner_server-13] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[async_slam_toolbox_node-10] [INFO] [1739782014.339954397] [slam_toolbox]: Node using stack size 40000000
[waypoint_follower-16] [INFO] [1739782014.343064590] [waypoint_follower]: 
[waypoint_follower-16] 	waypoint_follower lifecycle node launched. 
[waypoint_follower-16] 	Waiting on external lifecycle transitions to activate
[waypoint_follower-16] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[waypoint_follower-16] [INFO] [1739782014.345708162] [waypoint_follower]: Creating
[smoother_server-12] [INFO] [1739782014.347084290] [smoother_server]: 
[smoother_server-12] 	smoother_server lifecycle node launched. 
[smoother_server-12] 	Waiting on external lifecycle transitions to activate
[smoother_server-12] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[lifecycle_manager-18] [INFO] [1739782014.349223920] [lifecycle_manager_navigation]: Creating
[teleop_node-7] [INFO] [1739782014.351585238] [TeleopTwistJoy]: Linear axis x on 1 at scale 0.200000.
[teleop_node-7] [INFO] [1739782014.351662874] [TeleopTwistJoy]: Linear axis y on 3 at scale 0.200000.
[teleop_node-7] [INFO] [1739782014.351676854] [TeleopTwistJoy]: Angular axis yaw on 6 at scale 0.200000.
[smoother_server-12] [INFO] [1739782014.355475864] [smoother_server]: Creating smoother server
[planner_server-13] [INFO] [1739782014.356349292] [planner_server]: Creating
[controller_server-11] [INFO] [1739782014.357634391] [controller_server]: 
[controller_server-11] 	controller_server lifecycle node launched. 
[controller_server-11] 	Waiting on external lifecycle transitions to activate
[controller_server-11] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[bt_navigator-15] [INFO] [1739782014.363625949] [bt_navigator]: 
[bt_navigator-15] 	bt_navigator lifecycle node launched. 
[bt_navigator-15] 	Waiting on external lifecycle transitions to activate
[bt_navigator-15] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[bt_navigator-15] [INFO] [1739782014.363814803] [bt_navigator]: Creating
[foxglove_bridge-9] [INFO] [1739782014.363851653] [foxglove_bridge_component_manager]: Load Library: /opt/ros/humble/lib/libfoxglove_bridge_component.so
[controller_server-11] [INFO] [1739782014.365270728] [controller_server]: Creating controller server
[lifecycle_manager-18] [INFO] [1739782014.368259293] [lifecycle_manager_navigation]: Creating and initializing lifecycle service clients
[foxglove_bridge-9] [INFO] [1739782014.381054587] [foxglove_bridge_component_manager]: Found class: rclcpp_components::NodeFactoryTemplate<foxglove_bridge::FoxgloveBridge>
[... cut ...]
[planner_server-13] [WARN] [1739782035.742925871] [global_costmap.global_costmap]: Can't update static costmap layer, no map received
[controller_server-11] [WARN] [1739782045.309155304] [local_costmap.local_costmap]: Can't update static costmap layer, no map received
[planner_server-13] [WARN] [1739782045.749427231] [global_costmap.global_costmap]: Can't update static costmap layer, no map received
```

I just realised there are some errors in the log. When I installed and compiled the package, I did not see any error. Could you please give me any insights? Thanks!

tfoldi, 2025-02-17:
from the logs it seems you did not specify the ROBOT_IP:

```
[go2_driver_node-3] ERROR:root:An error occurred: Invalid URL 'http://:9991/con_notify': No host supplied
```

As per [usage](https://github.com/abizovnuralem/go2_ros2_sdk#usage), please specify the robot_ip as environment variable (or launch parameters):

```
source install/setup.bash
export ROBOT_IP="robot_ip" #for muliple robots, just split by ,
export CONN_TYPE="webrtc"
ros2 launch go2_robot_sdk robot.launch.py
```

edugm94, 2025-02-17:
Hi,

Yes. I did the mistake of launching the file first and then setting up the IP address. Now, I could launch the file without problems. 
Thanks for your support!

closed 2025-02-17 by tfoldi