abizovnuralem/go2_ros2_sdk #135: Running ros2 launch go2_robot_sdk robot.launch.py ​​reports [ERROR] [go2_driver_node-3]: process has died and [ERROR] [lidar_to_pointcloud-4]: process has died

zhouleiqiang, 2025-03-05:
Hello, my robot model is GO2 pro. When I configured these contents according to your tutorial, the following error message appeared:

[ERROR] [go2_driver_node-3]: process has died [pid 693039, exit code 255, cmd '/home/msi/ros2_ws/install/go2_robot_sdk/lib/go2_robot_sdk/go2_driver_node --ros-args --params-file /tmp/launch_params_odq7jwj4'].

[ERROR] [lidar_to_pointcloud-4]: process has died [pid 693041, exit code 255, cmd '/home/msi/ros2_ws/install/go2_robot_sdk/lib/go2_robot_sdk/lidar_to_pointcloud --ros-args --params-file /tmp/launch_params_oxoi32ow'].

The specific log files are as follows:
1741143140.0147340 [INFO] [launch]: All log files can be found below /home/msi/.ros/log/2025-03-05-10-52-20-013958-msi-Katana-15-B13VFK-12669
1741143140.0148537 [INFO] [launch]: Default logging verbosity is set to INFO
1741143140.2829173 [INFO] [robot_state_publisher-1]: Aprocess started with pid [12670]
1741143140.2831330 [INFO] [pointcloud_to_laserscan_node-2]: process started with pid [12672]
1741143140.2832301 [INFO] [go2_driver_node-3]: process started with pid [12674]
1741143140.2833083 [INFO] [lidar_to_pointcloud-4]: process started with pid [12676]
1741143140.2833710 [INFO] [rviz2-5]: process started with pid [12678]
1741143140.2834272 [INFO] [joy_node-6]: process started with pid [12680]
1741143140.2834809 [INFO] [teleop_node-7]: process started with pid [12682]A1741143140.2835386 [INFO] [twist_mux-8]: process started with pid [12684]
1741143140.2835889 [INFO] [foxglove_bridge-9]: process started with pid [12686]
1741143140.2836399 [INFO] [async_slam_toolbox_node-10]: process started with pid [12688]
1741143140.2836936 [INFO] [controller_server-11]: process started with pid [12690]
1741143140.2837427 [INFO] [smoother_server-12]: process started with pid [12692]
1741143140.2837949 [INFO] [planner_server-13]: process started with pid [12694]
1741143140.2838476 [INFO] [behavior_server-14]: process started with pid [12696]
1741143140.2838962 [INFO] [bt_navigator-15]: process started with pid [12698]
1741143140.2839353 [INFO] [waypoint_follower-16]: process started with pid [12701]
1741143140.2839723 [INFO] [velocity_smoother-17]: process started with pid [12703]
1741143140.2840061 [INFO] [lifecycle_manager-18]: process started with pid [12705]
1741143140.2990854 [rviz2-5] Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
1741143140.3499150 [teleop_node-7] [INFO] [1741143140.349627471] [TeleopTwistJoy]: Linear axis x on 1 at scale 0.200000.
1741143140.3501554 [teleop_node-7] [INFO] [1741143140.349668445] [TeleopTwistJoy]: Linear axis y on 3 at scale 0.200000.
1741143140.3502715 [teleop_node-7] [INFO] [1741143140.349672084] [TeleopTwistJoy]: Angular axis yaw on 6 at scale 0.200000.
1741143140.3873901 [lidar_to_pointcloud-4] aioice submodule is not initalized. please init submodules recursively
1741143140.3950961 [ERROR] [lidar_to_pointcloud-4]: process has died [pid 12676, exit code 255, cmd '/home/msi/ros_ws/install/go2_robot_sdk/lib/go2_robot_sdk/lidar_to_pointcloud --ros-args --params-file /tmp/launch_params_0lkhu70r'].
1741143140.4198000 [go2_driver_node-3] aioice submodule is not initalized. please init submodules recursively
1741143140.4323587 [ERROR] [go2_driver_node-3]: process has died [pid 12674, exit code 255, cmd '/home/msi/ros_ws/install/go2_robot_sdk/lib/go2_robot_sdk/go2_driver_node --ros-args --params-file /tmp/launch_params_jaqm73nn'].
1741143140.4703898 [foxglove_bridge-9] [INFO] [1741143140.466298687] [foxglove_bridge_component_manager]: Load Library: /opt/ros/humble/lib/libfoxglove_bridge_component.so
1741143140.5078638 [foxglove_bridge-9] [INFO] [1741143140.504689261] [foxglove_bridge_component_manager]: Found class: rclcpp_components::NodeFactoryTemplate<foxglove_bridge::FoxgloveBridge>
1741143140.5088670 [foxglove_bridge-9] [INFO] [1741143140.504716717] [foxglove_bridge_component_manager]: Instantiate class: rclcpp_components::NodeFactoryTemplate<foxglove_bridge::FoxgloveBridge>
1741143140.5354855 [foxglove_bridge-9] [INFO] [1741143140.534156036] [foxglove_bridge]: Starting foxglove_bridge (humble, 0.8.3@) with WebSocket++/0.8.2
1741143140.5374827 [foxglove_bridge-9] [INFO] [1741143140.534471205] [foxglove_bridge]: [WS] Server running without TLS
1741143140.5461144 [foxglove_bridge-9] [INFO] [1741143140.545138469] [foxglove_bridge]: [WS] asio listen error: asio.system:98 (Address already in use)
1741143140.5484276 [foxglove_bridge-9] terminate called after throwing an instance of 'std::runtime_error'
1741143140.5502625 [foxglove_bridge-9]   what():  Failed to listen on port 8765: Address already in use
1741143140.8654509 [rviz2-5] [INFO] [1741143140.865099108] [rviz2]: Stereo is NOT SUPPORTED
1741143140.8659518 [rviz2-5] [INFO] [1741143140.865202859] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
1741143140.8811655 [rviz2-5] [INFO] [1741143140.880799415] [rviz2]: Stereo is NOT SUPPORTED
1741143141.0158501 [rviz2-5] [WARN] [1741143141.015407735] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
1741143141.0476582 [rviz2-5] [WARN] [1741143141.047280750] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
1741143141.6903143 [rviz2-5] [INFO] [1741143141.689863044] [rviz2]: Stereo is NOT SUPPORTED
1741143141.6906879 [ERROR] [foxglove_bridge-9]: process has died [pid 12686, exit code -6, cmd '/opt/ros/humble/lib/foxglove_bridge/foxglove_bridge --ros-args --params-file /tmp/launch_params_620fxplz --params-file /tmp/launch_params_psjhakpt --params-file /tmp/launch_params_02zsf055 --params-file /tmp/launch_params_z6fqeb7w --params-file /tmp/launch_params_j1f_92xn --params-file /tmp/launch_params_34k9cdor --params-file /tmp/launch_params_5rfcmz6b --params-file /tmp/launch_params_0pp2_y5g --params-file /tmp/launch_params_potbdkw_ --params-file /tmp/launch_params_6_ka42ef --params-file /tmp/launch_params_2xu5gnbj --params-file /tmp/launch_params_u_2cb4vy --params-file /tmp/launch_params_2jldbaqn --params-file /tmp/launch_params_7jjrf242 --params-file /tmp/launch_params_6z3glxpe --params-file /tmp/launch_params_0dvid8mn --params-file /tmp/launch_params_zsjg64ip'].
1741143173.6345999 [WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
1741143173.6516941 [teleop_node-7] [INFO] [1741143173.632834641] [rclcpp]: signal_handler(signum=2)
1741143173.6563172 [rviz2-5] [INFO] [1741143173.632881630] [rclcpp]: signal_handler(signum=2)
1741143173.6593969 [joy_node-6] [INFO] [1741143173.632951221] [rclcpp]: signal_handler(signum=2)
1741143173.7735868 [INFO] [teleop_node-7]: process has finished cleanly [pid 12682]
1741143173.7797627 [INFO] [async_slam_toolbox_node-10]: process has finished cleanly [pid 12688]
1741143173.7831910 [INFO] [smoother_server-12]: process has finished cleanly [pid 12692]
1741143173.7876565 [INFO] [waypoint_follower-16]: process has finished cleanly [pid 12701]
1741143173.7879198 [INFO] [velocity_smoother-17]: process has finished cleanly [pid 12703]
1741143173.7885056 [INFO] [robot_state_publisher-1]: process has finished cleanly [pid 12670]
1741143173.7911441 [INFO] [joy_node-6]: process has finished cleanly [pid 12680]
1741143173.7915325 [INFO] [pointcloud_to_laserscan_node-2]: process has finished cleanly [pid 12672]
1741143173.7918761 [INFO] [twist_mux-8]: process has finished cleanly [pid 12684]
1741143173.7928793 [INFO] [bt_navigator-15]: process has finished cleanly [pid 12698]
1741143173.7943070 [INFO] [behavior_server-14]: process has finished cleanly [pid 12696]
1741143173.7954268 [INFO] [planner_server-13]: process has finished cleanly [pid 12694]
1741143173.7994604 [INFO] [lifecycle_manager-18]: process has finished cleanly [pid 12705]
1741143173.8196735 [INFO] [controller_server-11]: process has finished cleanly [pid 12690]
1741143174.8619294 [ERROR] [rviz2-5]: process has died [pid 12678, exit code -11, cmd '/opt/ros/humble/lib/rviz2/rviz2 -d/home/msi/ros_ws/install/go2_robot_sdk/share/go2_robot_sdk/config/single_robot_conf.rviz --ros-args -r __node:=rviz2 -r __ns:=/'].

How should I solve it? I look forward to your reply.

tfoldi, 2025-03-05:
it is in the logs: 

1741143140.4198000 [go2_driver_node-3] aioice submodule is not initalized. please init submodules recursively

can you make sure you init and fetched all submodules?

zhouleiqiang, 2025-03-05:
> it is in the logs:
> 
> 1741143140.4198000 [go2_driver_node-3] aioice submodule is not initalized. please init submodules recursively
> 
> can you make sure you init and fetched all submodules?

This error is no longer reported. Is this what the error is now? It seems that communication is impossible. I noticed that our project only supports firmware up to v1.1.1, but our go2 pro firmware is 1.1.2. Currently, an error is reported. ERROR:root:An error occurred: 502 Server Error: Bad Gateway for url: http://192.168.137.254:9991/con_notify [go2_driver_node-3] [ERROR] [1741158168.912308488] [go2_driver_node]: Failed to connect to robot 0 - exiting: Failed to receive initial public key response with new method. [go2_driver_node-3] [ERROR] [1741158168.912938624] [go2_driver_node]: Error in robot 0: Failed to connect to robot 0: Failed to receive initial public key response with new method. [go2_driver_node-3] [ERROR] [1741158168.913408984] [go2_driver_node]: Error in completed task: Failed to connect to robot 0: Failed to receive initial public key response with new method. How do I solve this? Looking forward to your reply

zhouleiqiang, 2025-03-05:
> are you on STA-L connection and http://192.168.137.254:9991 is connectable? Did you get this IP from the mobile app or somewhere else? Can you confirm that there is no http_proxy defined?
> 
> It must work with 1.1.2, so eventually it should work for you as well.

> 您是否处于 STA-L 连接状态并且http://192.168.137.254:9991可以连接？您是从移动应用程序还是其他地方获取此 IP？您能确认没有定义 http_proxy 吗？
> 
> 它必须与 1.1.2 一起工作，所以最终它也应该适合您。

I followed the instructions in the readme to get the IP address

【Don't forget to set up your Go2 robot in Wifi-mode and obtain the IP. You can use the mobile app to get it. Go to Device -> Data -> Automatic Machine Inspection and look for STA Network: wlan0.”】

Because my device is GO2 PRO, I can't directly enter the device's desktop

My current connection method is: based on Device -> Data -> Automatic to obtain the IP. My laptop and GO2 PRO are connected to the same WIFI. I don't quite understand how to define http_proxy. At present, my device status is that there is no data in all topics, and the rviz interface is as follows. I don't know how to solve it.

I don't know if there are other places that need to be set in addition to setting the IP?

Looking forward to your reply, thank you again

[image]

zhouleiqiang, 2025-03-05:
> are you on STA-L connection and http://192.168.137.254:9991 is connectable? Did you get this IP from the mobile app or somewhere else? Can you confirm that there is no http_proxy defined?
> 
> It must work with 1.1.2, so eventually it should work for you as well.

> are you on STA-L connection and http://192.168.137.254:9991 is connectable? Did you get this IP from the mobile app or somewhere else? Can you confirm that there is no http_proxy defined?
> 
> It must work with 1.1.2, so eventually it should work for you as well.

I can ping 192.168.137.254

I don't understand how to ensure that http://192.168.137.254:9991 is connectable. What does the suffix 9991 represent? There is no description of 9991 in the readme. Do I need to set it up additionally? Thanks again for your reply

tfoldi, 2025-03-05:
let me ask it differently: how did you get that IP? is it from the mobile application, from your router? like how do you obtained your robot's IP address

zhouleiqiang, 2025-03-05:
> let me ask it differently: how did you get that IP? is it from the mobile application, from your router? like how do you obtained your robot's IP address

from the mobile application

zhouleiqiang, 2025-03-06:
> 让我换个问法：你是怎么得到那个 IP 的？是从移动应用程序还是从路由器？比如你是如何获得机器人的 IP 地址的

Hello, I have now eliminated the error and run it successfully. How should I control the GO2 pro to move forward 5 meters, such a basic function? Do you have a related routine? Looking forward to your reply

tfoldi, 2025-03-06:
glad it works.

you can use default ROS 2 control mechanism via `/cmd_vel` like:

```
ros2 topic pub /cmd_vel geometry_msgs/msg/Twist "{linear: {x: 1.0, y: 0.0, z: 0.0}, angular: {x: 0.0, y: 0.0, z: 0.0}}" --once
```

You can use joystick, `teleop_twist_keyboard` or anything else you want. To move 5 meters ahead you need a twist message sent for 5 seconds with 1 m/s x linear velocity.

closed 2025-03-06 by tfoldi

HR-zju, 2025-03-20:
> > 让我换个问法：你是怎么得到那个 IP 的？是从移动应用程序还是从路由器？比如你是如何获得机器人的 IP 地址的
> 
> Hello, I have now eliminated the error and run it successfully. How should I control the GO2 pro to move forward 5 meters, such a basic function? Do you have a related routine? Looking forward to your reply

hello, how did you solve this problem? @zhouleiqiang

zhouleiqiang, 2025-03-21:
> > > 让我换个问法：你是怎么获得那个IP的？是从移动应用程序还是从路由器？比如你是如何获得机器人的IP地址的
> > 
> > 
> > 你好，现在错误已经排除，运行成功。那么基本的功能，如何控制GO2 pro前进5米呢？有相关例程吗？期待你的回复
> 
> 你好，你是如何解决这个问题的？[@zhouleiqiang](https://github.com/zhouleiqiang)

Actually, I had an environment and dependency problem. I reinstalled ROS2 and the problem was solved.