abizovnuralem/go2_ros2_sdk #139: AssertionError: The 'position' field must be a set or sequence and each value of type 'float' and each double

5uvin, 2025-03-06:
Hello, 
Can i get some help with this error i'm facing 
```
[go2_driver_node-3]   File "/mnt/robot_ws/install/go2_robot_sdk/lib/python3.10/site-packages/go2_robot_sdk/go2_driver_node.py", line 262, in timer_callback
[go2_driver_node-3]     self.publish_joint_state_webrtc()
[go2_driver_node-3]   File "/mnt/robot_ws/install/go2_robot_sdk/lib/python3.10/site-packages/go2_robot_sdk/go2_driver_node.py", line 600, in publish_joint_state_webrtc
[go2_driver_node-3]     joint_state.position = [
[go2_driver_node-3]   File "/opt/ros/humble/local/lib/python3.10/dist-packages/sensor_msgs/msg/_joint_state.py", line 203, in position
[go2_driver_node-3]     assert \
[go2_driver_node-3] AssertionError: The 'position' field must be a set or sequence and each value of type 'float' and each double in [-179769313486231570814527423731704356798070567525844996598917476803157260780028538760589558632766878171540458953514382464234321326889464182768467546703537516986049910576551282076245490090389328944075868508455133942304583236903222948165808559332123348274797826204144723168738177180919299881250404026184124858368.000000, 179769313486231570814527423731704356798070567525844996598917476803157260780028538760589558632766878171540458953514382464234321326889464182768467546703537516986049910576551282076245490090389328944075868508455133942304583236903222948165808559332123348274797826204144723168738177180919299881250404026184124858368.000000]
```
Im using the master branch, last commit 5e296bf0abf9c282498b5f94af4f44e78c405c74
Dog Frimware: v1.1.1
Connection Type: WebRTC
OS: Ubuntu 22.04 (Docker)

tfoldi, 2025-03-07:
hmm I belive I fixed this. let me check today and get back to you

tfoldi, 2025-03-07:
can you give me a bit more context? can you share the full log from the execution? also, can you also share the output of  `ros2 topic` list and `ros2 node list`? I suspect some other nodes are running like `joint_state_publisher_gui`

I will add checks to ensure no NaN or Inf numbers will published, but would be nice to know what the root cause is. 

thanks!

5uvin, 2025-03-07:
Full Log
```
[INFO] [robot_state_publisher-1]: process started with pid [257]
[INFO] [pointcloud_to_laserscan_node-2]: process started with pid [259]
[INFO] [go2_driver_node-3]: process started with pid [261]
[INFO] [joy_node-4]: process started with pid [263]
[INFO] [teleop_node-5]: process started with pid [265]
[INFO] [twist_mux-6]: process started with pid [267]
[twist_mux-6] [INFO] [1741345871.416769060] [twist_mux]: Topic handler 'topics.joy' subscribed to topic 'cmd_vel_joy': timeout = 0.500000s , priority = 10.
[twist_mux-6] [INFO] [1741345871.417632563] [twist_mux]: Topic handler 'topics.navigation' subscribed to topic 'cmd_vel': timeout = 0.500000s , priority = 5.
[robot_state_publisher-1] [INFO] [1741345871.420912441] [robot_state_publisher]: got segment FL_calf
[robot_state_publisher-1] [INFO] [1741345871.420980715] [robot_state_publisher]: got segment FL_calflower
[robot_state_publisher-1] [INFO] [1741345871.420987047] [robot_state_publisher]: got segment FL_calflower1
[robot_state_publisher-1] [INFO] [1741345871.420990141] [robot_state_publisher]: got segment FL_foot
[robot_state_publisher-1] [INFO] [1741345871.420992609] [robot_state_publisher]: got segment FL_hip
[robot_state_publisher-1] [INFO] [1741345871.420994934] [robot_state_publisher]: got segment FL_thigh
[robot_state_publisher-1] [INFO] [1741345871.420997582] [robot_state_publisher]: got segment FR_calf
[robot_state_publisher-1] [INFO] [1741345871.421001343] [robot_state_publisher]: got segment FR_calflower
[robot_state_publisher-1] [INFO] [1741345871.421005265] [robot_state_publisher]: got segment FR_calflower1
[robot_state_publisher-1] [INFO] [1741345871.421009054] [robot_state_publisher]: got segment FR_foot
[robot_state_publisher-1] [INFO] [1741345871.421012737] [robot_state_publisher]: got segment FR_hip
[robot_state_publisher-1] [INFO] [1741345871.421016389] [robot_state_publisher]: got segment FR_thigh
[robot_state_publisher-1] [INFO] [1741345871.421019912] [robot_state_publisher]: got segment Head_lower
[robot_state_publisher-1] [INFO] [1741345871.421023821] [robot_state_publisher]: got segment Head_upper
[robot_state_publisher-1] [INFO] [1741345871.421027399] [robot_state_publisher]: got segment RL_calf
[robot_state_publisher-1] [INFO] [1741345871.421031238] [robot_state_publisher]: got segment RL_calflower
[robot_state_publisher-1] [INFO] [1741345871.421035245] [robot_state_publisher]: got segment RL_calflower1
[robot_state_publisher-1] [INFO] [1741345871.421039007] [robot_state_publisher]: got segment RL_foot
[robot_state_publisher-1] [INFO] [1741345871.421043374] [robot_state_publisher]: got segment RL_hip
[robot_state_publisher-1] [INFO] [1741345871.421047233] [robot_state_publisher]: got segment RL_thigh
[robot_state_publisher-1] [INFO] [1741345871.421051198] [robot_state_publisher]: got segment RR_calf
[robot_state_publisher-1] [INFO] [1741345871.421055159] [robot_state_publisher]: got segment RR_calflower
[robot_state_publisher-1] [INFO] [1741345871.421059141] [robot_state_publisher]: got segment RR_calflower1
[robot_state_publisher-1] [INFO] [1741345871.421063005] [robot_state_publisher]: got segment RR_foot
[robot_state_publisher-1] [INFO] [1741345871.421067116] [robot_state_publisher]: got segment RR_hip
[robot_state_publisher-1] [INFO] [1741345871.421071205] [robot_state_publisher]: got segment RR_thigh
[robot_state_publisher-1] [INFO] [1741345871.421075367] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-1] [INFO] [1741345871.421079618] [robot_state_publisher]: got segment base_link
[robot_state_publisher-1] [INFO] [1741345871.421083636] [robot_state_publisher]: got segment front_camera
[robot_state_publisher-1] [INFO] [1741345871.421087647] [robot_state_publisher]: got segment imu
[robot_state_publisher-1] [INFO] [1741345871.421091950] [robot_state_publisher]: got segment map
[robot_state_publisher-1] [INFO] [1741345871.421095992] [robot_state_publisher]: got segment odom
[robot_state_publisher-1] [INFO] [1741345871.421099865] [robot_state_publisher]: got segment radar
[teleop_node-5] [INFO] [1741345871.422171225] [TeleopTwistJoy]: Linear axis x on 1 at scale 0.200000.
[teleop_node-5] [INFO] [1741345871.422224322] [TeleopTwistJoy]: Linear axis y on 3 at scale 0.200000.
[teleop_node-5] [INFO] [1741345871.422232645] [TeleopTwistJoy]: Angular axis yaw on 6 at scale 0.200000.
[go2_driver_node-3] /opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/node.py:441: UserWarning: when declaring parameter named 'robot_ip', declaring a parameter only providing its name is deprecated. You have to either:
[go2_driver_node-3] 	- Pass a name and a default value different to "PARAMETER NOT SET" (and optionally a descriptor).
[go2_driver_node-3] 	- Pass a name and a parameter type.
[go2_driver_node-3] 	- Pass a name and a descriptor with `dynamic_typing=True
[go2_driver_node-3]   warnings.warn(
[go2_driver_node-3] [INFO] [1741345871.884836265] [go2_driver_node]: Received ip list: ['192.168.1.104']
[go2_driver_node-3] [INFO] [1741345871.885070724] [go2_driver_node]: Connection type is webrtc
[go2_driver_node-3] [INFO] [1741345871.885230859] [go2_driver_node]: Connection mode is single
[go2_driver_node-3] [INFO] [1741345871.885386020] [go2_driver_node]: Enable video is True
[go2_driver_node-3] [INFO] [1741345871.885549248] [go2_driver_node]: Decode lidar is True
[go2_driver_node-3] [INFO] [1741345871.885707861] [go2_driver_node]: Publish raw voxel is False
[... cut ...]
[go2_driver_node-3] INFO:scripts.webrtc_driver:Data channel is open
[go2_driver_node-3] INFO:scripts.webrtc_driver:-> Sending message {"type": "validation", "topic": "", "data": "FqTMy677xDkKm5XkX3nNIQ=="}
[... 123 lines cut ...]
/twist_mux
```
Topic list
```
/camera/camera_info
/camera/image_raw
/cmd_vel
/cmd_vel_joy
/cmd_vel_out
/diagnostics
/go2_states
/imu
/joint_states
/joy
/joy/set_feedback
/odom
/parameter_events
/point_cloud2
/robot_description
/rosout
/scan
/tf
/tf_static
/webrtc_req
```

5uvin, 2025-03-07:
Made the changes you recommended to `go2_math.py`. but i'm getting the same results. 
Made the following changes in `go2_math.py`
```
    else:
        J = P * (L + np.arcsin(HIP_LENGTH / O))

    if not all(math.isfinite(val) for val in [J, A, S]):
        return 0, 0, 0

    return J, A, S
```
```
[INFO] [robot_state_publisher-1]: process started with pid [31652]
[INFO] [pointcloud_to_laserscan_node-2]: process started with pid [31654]
[INFO] [go2_driver_node-3]: process started with pid [31656]
[INFO] [joy_node-4]: process started with pid [31658]
[INFO] [teleop_node-5]: process started with pid [31660]
[INFO] [twist_mux-6]: process started with pid [31662]
[twist_mux-6] [INFO] [1741357253.406346059] [twist_mux]: Topic handler 'topics.joy' subscribed to topic 'cmd_vel_joy': timeout = 0.500000s , priority = 10.
[twist_mux-6] [INFO] [1741357253.407018114] [twist_mux]: Topic handler 'topics.navigation' subscribed to topic 'cmd_vel': timeout = 0.500000s , priority = 5.
[robot_state_publisher-1] [INFO] [1741357253.408726069] [robot_state_publisher]: got segment FL_calf
[robot_state_publisher-1] [INFO] [1741357253.408775714] [robot_state_publisher]: got segment FL_calflower
[robot_state_publisher-1] [INFO] [1741357253.408780247] [robot_state_publisher]: got segment FL_calflower1
[robot_state_publisher-1] [INFO] [1741357253.408782918] [robot_state_publisher]: got segment FL_foot
[robot_state_publisher-1] [INFO] [1741357253.408785348] [robot_state_publisher]: got segment FL_hip
[robot_state_publisher-1] [INFO] [1741357253.408787807] [robot_state_publisher]: got segment FL_thigh
[robot_state_publisher-1] [INFO] [1741357253.408790027] [robot_state_publisher]: got segment FR_calf
[robot_state_publisher-1] [INFO] [1741357253.408792317] [robot_state_publisher]: got segment FR_calflower
[robot_state_publisher-1] [INFO] [1741357253.408794626] [robot_state_publisher]: got segment FR_calflower1
[robot_state_publisher-1] [INFO] [1741357253.408796875] [robot_state_publisher]: got segment FR_foot
[robot_state_publisher-1] [INFO] [1741357253.408799089] [robot_state_publisher]: got segment FR_hip
[robot_state_publisher-1] [INFO] [1741357253.408801347] [robot_state_publisher]: got segment FR_thigh
[robot_state_publisher-1] [INFO] [1741357253.408803597] [robot_state_publisher]: got segment Head_lower
[robot_state_publisher-1] [INFO] [1741357253.408805882] [robot_state_publisher]: got segment Head_upper
[robot_state_publisher-1] [INFO] [1741357253.408808148] [robot_state_publisher]: got segment RL_calf
[robot_state_publisher-1] [INFO] [1741357253.408810405] [robot_state_publisher]: got segment RL_calflower
[robot_state_publisher-1] [INFO] [1741357253.408812769] [robot_state_publisher]: got segment RL_calflower1
[robot_state_publisher-1] [INFO] [1741357253.408815011] [robot_state_publisher]: got segment RL_foot
[robot_state_publisher-1] [INFO] [1741357253.408817301] [robot_state_publisher]: got segment RL_hip
[robot_state_publisher-1] [INFO] [1741357253.408819624] [robot_state_publisher]: got segment RL_thigh
[robot_state_publisher-1] [INFO] [1741357253.408821871] [robot_state_publisher]: got segment RR_calf
[robot_state_publisher-1] [INFO] [1741357253.408824113] [robot_state_publisher]: got segment RR_calflower
[robot_state_publisher-1] [INFO] [1741357253.408826389] [robot_state_publisher]: got segment RR_calflower1
[robot_state_publisher-1] [INFO] [1741357253.408828649] [robot_state_publisher]: got segment RR_foot
[robot_state_publisher-1] [INFO] [1741357253.408831071] [robot_state_publisher]: got segment RR_hip
[robot_state_publisher-1] [INFO] [1741357253.408833288] [robot_state_publisher]: got segment RR_thigh
[robot_state_publisher-1] [INFO] [1741357253.408835514] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-1] [INFO] [1741357253.408837879] [robot_state_publisher]: got segment base_link
[robot_state_publisher-1] [INFO] [1741357253.408840154] [robot_state_publisher]: got segment front_camera
[robot_state_publisher-1] [INFO] [1741357253.408842423] [robot_state_publisher]: got segment imu
[robot_state_publisher-1] [INFO] [1741357253.408844832] [robot_state_publisher]: got segment map
[robot_state_publisher-1] [INFO] [1741357253.408847266] [robot_state_publisher]: got segment odom
[robot_state_publisher-1] [INFO] [1741357253.408849500] [robot_state_publisher]: got segment radar
[teleop_node-5] [INFO] [1741357253.414601692] [TeleopTwistJoy]: Linear axis x on 1 at scale 0.200000.
[teleop_node-5] [INFO] [1741357253.414656617] [TeleopTwistJoy]: Linear axis y on 3 at scale 0.200000.
[teleop_node-5] [INFO] [1741357253.414664279] [TeleopTwistJoy]: Angular axis yaw on 6 at scale 0.200000.
[go2_driver_node-3] /opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/node.py:441: UserWarning: when declaring parameter named 'robot_ip', declaring a parameter only providing its name is deprecated. You have to either:
[go2_driver_node-3] 	- Pass a name and a default value different to "PARAMETER NOT SET" (and optionally a descriptor).
[go2_driver_node-3] 	- Pass a name and a parameter type.
[go2_driver_node-3] 	- Pass a name and a descriptor with `dynamic_typing=True
[go2_driver_node-3]   warnings.warn(
[go2_driver_node-3] [INFO] [1741357253.862583405] [go2_driver_node]: Received ip list: ['192.168.1.104']
[go2_driver_node-3] [INFO] [1741357253.862831254] [go2_driver_node]: Connection type is webrtc
[go2_driver_node-3] [INFO] [1741357253.862990706] [go2_driver_node]: Connection mode is single
[... cut ...] 68467546703537516986049910576551282076245490090389328944075868508455133942304583236903222948165808559332123348274797826204144723168738177180919299881250404026184124858368.000000, 179769313486231570814527423731704356798070567525844996598917476803157260780028538760589558632766878171540458953514382464234321326889464182768467546703537516986049910576551282076245490090389328944075868508455133942304583236903222948165808559332123348274797826204144723168738177180919299881250404026184124858368.000000]
```

5uvin, 2025-03-07:
Is there a way to switch the dog from AI mode without the app, because the app does not let me switch to other modes unless i upgrade the dogs frimware.
Anyways thank you for looking into this. 

[image]

5uvin, 2025-03-07:
Good news, I was able to get it to work with your recommendations. Used legion's sdk to switch out of AI mode and made changes to `go2_math.py`. Now i can connect to the robot as expected. 
If its not too much trouble please do add a service to switch mode, regardless Thank you so much for you help and you can close this issue.

tfoldi, 2025-03-07:
thanks and will do!

closed 2025-03-07 by tfoldi

andy-zhuo-02, 2025-06-08:
Recently Unitree Official Update their Fireware and merge AI mode, classic mode and so on. I encountered with the same problem. Any advice? 

Here are logs:

> andy_zhuo@andy-desktop:~/go2_ros2_ws$ ros2 launch go2_robot_sdk robot.launch.py
[INFO] [launch]: All log files can be found below /home/andy_zhuo/.ros/log/2025-06-08-12-22-35-693913-andy-desktop-49883
[INFO] [launch]: Default logging verbosity is set to INFO
IP list: ['192.168.3.101']
[INFO] [robot_state_publisher-1]: process started with pid [49910]
[INFO] [pointcloud_to_laserscan_node-2]: process started with pid [49912]
[INFO] [go2_driver_node-3]: process started with pid [49914]
[INFO] [lidar_to_pointcloud-4]: process started with pid [49916]
[INFO] [rviz2-5]: process started with pid [49918]
[INFO] [joy_node-6]: process started with pid [49920]
[INFO] [teleop_node-7]: process started with pid [49922]
[INFO] [twist_mux-8]: process started with pid [49924]
[INFO] [foxglove_bridge-9]: process started with pid [49926]
[INFO] [async_slam_toolbox_node-10]: process started with pid [49950]
[INFO] [controller_server-11]: process started with pid [49953]
[INFO] [smoother_server-12]: process started with pid [49966]
[INFO] [planner_server-13]: process started with pid [49972]
[INFO] [behavior_server-14]: process started with pid [49974]
[INFO] [bt_navigator-15]: process started with pid [49982]
[INFO] [waypoint_follower-16]: process started with pid [49984]
[INFO] [velocity_smoother-17]: process started with pid [49986]
[INFO] [lifecycle_manager-18]: process started with pid [49989]
[rviz2-5] QStandardPaths: wrong permissions on runtime directory /run/user/1000/, 0755 instead of 0700
[teleop_node-7] [INFO] [1749356556.527427890] [TeleopTwistJoy]: Linear axis x on 1 at scale 0.200000.
[teleop_node-7] [INFO] [1749356556.527526248] [TeleopTwistJoy]: Linear axis y on 3 at scale 0.200000.
[teleop_node-7] [INFO] [1749356556.527538623] [TeleopTwistJoy]: Angular axis yaw on 6 at scale 0.200000.
[twist_mux-8] [INFO] [1749356556.532201708] [twist_mux]: Topic handler 'topics.joy' subscribed to topic 'cmd_vel_joy': timeout = 0.500000s , priority = 10.
[twist_mux-8] [INFO] [1749356556.551949740] [twist_mux]: Topic handler 'topics.navigation' subscribed to topic 'cmd_vel': timeout = 0.500000s , priority = 5.
[robot_state_publisher-1] [INFO] [1749356556.596312662] [robot_state_publisher]: got segment FL_calf
[robot_state_publisher-1] [INFO] [1749356556.596591054] [robot_state_publisher]: got segment FL_calflower
[robot_state_publisher-1] [INFO] [1749356556.596623779] [robot_state_publisher]: got segment FL_calflower1
[robot_state_publisher-1] [INFO] [1749356556.596634687] [robot_state_publisher]: got segment FL_foot
[robot_state_publisher-1] [INFO] [1749356556.596643029] [robot_state_publisher]: got segment FL_hip
[robot_state_publisher-1] [INFO] [1749356556.596650821] [robot_state_publisher]: got segment FL_thigh
[robot_state_publisher-1] [INFO] [1749356556.596658704] [robot_state_publisher]: got segment FR_calf
[robot_state_publisher-1] [INFO] [1749356556.596667229] [robot_state_publisher]: got segment FR_calflower
[robot_state_publisher-1] [INFO] [1749356556.596675387] [robot_state_publisher]: got segment FR_calflower1
[robot_state_publisher-1] [INFO] [1749356556.596683362] [robot_state_publisher]: got segment FR_foot
[robot_state_publisher-1] [INFO] [1749356556.596691521] [robot_state_publisher]: got segment FR_hip
[robot_state_publisher-1] [INFO] [1749356556.596699496] [robot_state_publisher]: got segment FR_thigh
[robot_state_publisher-1] [INFO] [1749356556.596707746] [robot_state_publisher]: got segment Head_lower
[robot_state_publisher-1] [INFO] [1749356556.596715537] [robot_state_publisher]: got segment Head_upper
[robot_state_publisher-1] [INFO] [1749356556.596724246] [robot_state_publisher]: got segment RL_calf
[robot_state_publisher-1] [INFO] [1749356556.596733046] [robot_state_publisher]: got segment RL_calflower
[robot_state_publisher-1] [INFO] [1749356556.596741479] [robot_state_publisher]: got segment RL_calflower1
[robot_state_publisher-1] [INFO] [1749356556.596749821] [robot_state_publisher]: got segment RL_foot
[robot_state_publisher-1] [INFO] [1749356556.596757521] [robot_state_publisher]: got segment RL_hip
[robot_state_publisher-1] [INFO] [1749356556.596765221] [robot_state_publisher]: got segment RL_thigh
[robot_state_publisher-1] [INFO] [1749356556.596773196] [robot_state_publisher]: got segment RR_calf
[robot_state_publisher-1] [INFO] [1749356556.596796662] [robot_state_publisher]: got segment RR_calflower
[robot_state_publisher-1] [INFO] [1749356556.596806562] [robot_state_publisher]: got segment RR_calflower1
[robot_state_publisher-1] [INFO] [1749356556.596814813] [robot_state_publisher]: got segment RR_foot
[robot_state_publisher-1] [INFO] [1749356556.596822971] [robot_state_publisher]: got segment RR_hip
[robot_state_publisher-1] [INFO] [1749356556.596831221] [robot_state_publisher]: got segment RR_thigh
[robot_state_publisher-1] [INFO] [1749356556.596839379] [robot_state_publisher]: got segment base_footprint
[robot_state_publisher-1] [INFO] [1749356556.596847721] [robot_state_publisher]: got segment base_link
[robot_state_publisher-1] [INFO] [1749356556.596855513] [robot_state_publisher]: got segment front_camera
[robot_state_publisher-1] [INFO] [1749356556.596863488] [robot_state_publisher]: got segment imu
[robot_state_publisher-1] [INFO] [1749356556.596871463] [robot_state_publisher]: got segment map
[... cut ...]
[go2_driver_node-3] /home/andy_zhuo/go2_ros2_ws/install/go2_robot_sdk/lib/python3.10/site-packages/scripts/go2_math.py:147: RuntimeWarning: invalid value encountered in arcsin
[go2_driver_node-3]   J = P * (L + np.arcsin(HIP_LENGTH / O))
[go2_driver_node-3] [h264 @ 0x7f469c000e40] non-existing PPS 0 referenced
[go2_driver_node-3] [h264 @ 0x7f469c000e40] decode_slice_header error
[go2_driver_node-3] [h264 @ 0x7f469c000e40] no frame!
[go2_driver_node-3] WARNING:aiortc.codecs.h264:H264Decoder() failed to decode, skipping package: [Errno 1094995529] Invalid data found when processing input

DedSecer, 2025-06-15:
@tfoldi 
I encounter the same promblem like 
```
AssertionError: The 'position' field must be a set or sequence and each value of type 'float' and each double
```
and I have the warning continuously:
```
[go2_driver_node-3] WARNING:aiortc.codecs.h264:H264Decoder() failed to decode, skipping package: [Errno 1094995529] Invalid data found when processing input
[go2_driver_node-3] [h264 @ 0x7f2024000e40] non-existing PPS 0 referenced
[go2_driver_node-3] [h264 @ 0x7f2024000e40] decode_slice_header error
[go2_driver_node-3] [h264 @ 0x7f2024000e40] no frame!
```
even if I modify the `go2_math.py` as you said and switch sport mode  to normal or classic mode by mobile app.
I think it may be I updated to the lastest firmware?
Does this ros2 sdk work well in 1.1.7 firmware?