abizovnuralem/go2_ros2_sdk #140: Are there any interfaces, topics or services provided to control the robot's leg joints?

zhouleiqiang, 2025-03-07:
Hello, this is a great work. I have controlled the basic movement of the robot dog based on cmd_vel, but I have not found any topic or service that can control the robot's foot joints. Because I want to be able to control the robot dog to walk on stairs instead of just on flat ground, does this SDK support it? Looking forward to your reply.

tfoldi, 2025-03-07:
you can access mostly these functions: https://support.unitree.com/home/en/developer/sports_services

zhouleiqiang, 2025-03-10:
> you can access mostly these functions: https://support.unitree.com/home/en/developer/sports_services

I noticed that this document requires the name of the network card. My robot dog version is GO2 pro, and I cannot get the network card of the robot dog. Does this official document support GO2 pro? Or does it only support GO2 EDU?

abizovnuralem, 2025-03-10:
officially, GO2 pro and GO2 Air are not supported

closed 2025-05-08 by abizovnuralem