abizovnuralem/go2_ros2_sdk #144: Robot Dog Not Responding to /cmd_vel Commands After ROS SDK Installation in Docker

HR-zju, 2025-03-21:
Hello, thank you for your great work! I followed your instructions and successfully installed the ROS SDK in Docker. As shown in the screenshot below, I am able to receive topics normally. However, when I send data to /cmd_vel, the robot dog does not respond.

The bottom-right corner of the screenshot shows the error messages when I launched the program. The logs do not mention any network or "driver node" issues. Could you please help me identify what might be causing this problem?

[image]

spomichter, 2025-03-21:
You're sending to "cmd_vel_out"

spomichter, 2025-03-21:
Those two local costmap errors are not issues. They are warnings and do not crash ROS. So must be some other issue.

HR-zju, 2025-03-21:
> You're sending to "cmd_vel_out"?

ros2 topic pub /cmd_vel geometry_msgs/msg/Twist "{linear: {x: 1.0, y: 0.0, z: 0.0}, angular: {x: 0.0, y: 0.0, z: 0.0}}" --once

HR-zju, 2025-03-21:
@

> The second local costmap error is not an issue. It's a warning and does not crash ROS. So it's likely going to be the first error.

I fixed the first error, but I got the warning:
```bash
[WARN] [1742593809.003139496] [foxglove_bridge]: Failed to add channel for topic "/api/sport_lease/request" (unitree_api/msg/Request): package 'unitree_api' not found, searching: [/ros2_ws/install/go2_robot_sdk, /ros2_ws/install/unitree_go, /ros2_ws/install/go2_interfaces, /ros2_ws/install/coco_detector, /opt/ros/humble]
```
What's wrong>?

abizovnuralem, 2025-08-02:
I’ve refactored the project. If you encounter any problems, please feel free to reopen the issue.

closed 2025-08-02 by abizovnuralem