abizovnuralem/go2_ros2_sdk #145: Can't make it work

dvelho, 2025-03-22:
[image]
Stuck on this.
 I can see the image from the robot on rviz. 
followed the tutorial but no luck. 
any help appreciated. 

I can connect via the other library that connects to webrtc with no issues. just ros is killling me :(

tfoldi, 2025-03-23:
please share the log files and highlight any ERRORs in them, thx

dvelho, 2025-03-24:
I cant see any error on my logs 

`1742826535.6252620 [INFO] [launch]: Default logging verbosity is set to INFO
1742826536.1002021 [INFO] [robot_state_publisher-1]: process started with pid [967911]
1742826536.1004641 [INFO] [pointcloud_to_laserscan_node-2]: process started with pid [967913]
1742826536.1006117 [INFO] [go2_driver_node-3]: process started with pid [967915]
1742826536.1007357 [INFO] [lidar_to_pointcloud-4]: process started with pid [967917]
1742826536.1029739 [INFO] [rviz2-5]: process started with pid [967919]
1742826536.1031601 [INFO] [joy_node-6]: process started with pid [967921]
1742826536.1033008 [INFO] [teleop_node-7]: process started with pid [967923]
1742826536.1034353 [INFO] [twist_mux-8]: process started with pid [967925]
1742826536.1035669 [INFO] [foxglove_bridge-9]: process started with pid [967927]
1742826536.1036999 [INFO] [async_slam_toolbox_node-10]: process started with pid [967929]
1742826536.1038380 [INFO] [controller_server-11]: process started with pid [967931]
1742826536.1039748 [INFO] [smoother_server-12]: process started with pid [967933]
1742826536.1041090 [INFO] [planner_server-13]: process started with pid [967938]
1742826536.1042922 [INFO] [behavior_server-14]: process started with pid [967953]
1742826536.1044798 [INFO] [bt_navigator-15]: process started with pid [967971]
1742826536.1046610 [INFO] [waypoint_follower-16]: process started with pid [967974]
1742826536.1048334 [INFO] [velocity_smoother-17]: process started with pid [967988]
1742826536.1050091 [INFO] [lifecycle_manager-18]: process started with pid [967997]
1742826536.1091576 [rviz2-5] Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
1742826536.2570581 [teleop_node-7] [INFO] [1742826536.256664890] [TeleopTwistJoy]: Linear axis x on 1 at scale 0.200000.
1742826536.2572474 [teleop_node-7] [INFO] [1742826536.256731092] [TeleopTwistJoy]: Linear axis y on 3 at scale 0.200000.
1742826536.2573426 [teleop_node-7] [INFO] [1742826536.256742027] [TeleopTwistJoy]: Angular axis yaw on 6 at scale 0.200000.
1742826536.5058904 [foxglove_bridge-9] [INFO] [1742826536.494369896] [foxglove_bridge_component_manager]: Load Library: /opt/ros/humble/lib/libfoxglove_bridge_component.so
1742826536.6131110 [foxglove_bridge-9] [INFO] [1742826536.607771005] [foxglove_bridge_component_manager]: Found class: rclcpp_components::NodeFactoryTemplate<foxglove_bridge::FoxgloveBridge>
1742826536.6239505 [foxglove_bridge-9] [INFO] [1742826536.611587608] [foxglove_bridge_component_manager]: Instantiate class: rclcpp_components::NodeFactoryTemplate<foxglove_bridge::FoxgloveBridge>
1742826536.7362158 [foxglove_bridge-9] [INFO] [1742826536.730494292] [foxglove_bridge]: Starting foxglove_bridge (humble, 0.8.3@) with WebSocket++/0.8.2
1742826536.7537277 [foxglove_bridge-9] [INFO] [1742826536.737630530] [foxglove_bridge]: [WS] Server running without TLS
1742826536.7546787 [foxglove_bridge-9] [INFO] [1742826536.737945512] [foxglove_bridge]: [WS] WebSocket server listening at ws://0.0.0.0:8765
1742826536.7572227 [foxglove_bridge-9] [INFO] [1742826536.740199714] [foxglove_bridge]: [WS] WebSocket server run loop started
1742826537.3723426 [rviz2-5] [INFO] [1742826537.371520319] [rviz2]: Stereo is NOT SUPPORTED
1742826537.3731656 [rviz2-5] [INFO] [1742826537.371742263] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
1742826537.4017465 [rviz2-5] [INFO] [1742826537.401403997] [rviz2]: Stereo is NOT SUPPORTED
1742826537.7785804 [rviz2-5] [WARN] [1742826537.778212834] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
1742826537.8161600 [rviz2-5] [WARN] [1742826537.815760187] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
1742826538.5912471 [go2_driver_node-3] [INFO] [1742826538.590859982] [go2_driver_node]: Received ip list: ['192.168.2.126']
1742826538.5918405 [go2_driver_node-3] [INFO] [1742826538.591468410] [go2_driver_node]: Connection type is webrtc
1742826538.5923390 [go2_driver_node-3] [INFO] [1742826538.592027096] [go2_driver_node]: Connection mode is single
1742826538.5928447 [go2_driver_node-3] [INFO] [1742826538.592614803] [go2_driver_node]: Enable video is True
1742826538.5935864 [go2_driver_node-3] [INFO] [1742826538.593197853] [go2_driver_node]: Decode lidar is True
1742826538.5940719 [go2_driver_node-3] [INFO] [1742826538.593740616] [go2_driver_node]: Publish raw voxel is False
1742826538.6247211 [go2_driver_node-3] INFO:scripts.go2_camerainfo:Loading camera info from file: /home/mindera/ros2_ws/install/go2_robot_sdk/share/go2_robot_sdk/calibration/front_camera.yaml
1742826538.6390753 [go2_driver_node-3] INFO:root:Trying to send SDP using a NEW method...
1742826538.7315791 [rviz2-5] [INFO] [1742826538.731179527] [rviz2]: Stereo is NOT SUPPORTED
1742826538.9493670 [go2_driver_node-3] INFO:scripts.webrtc_driver:Connection state is connecting
[... cut ...]
1742826539.5097091 [lidar_to_pointcloud-4] 	- Pass a name and a default value different to "PARAMETER NOT SET" (and optionally a descriptor).
1742826539.5107431 [lidar_to_pointcloud-4] 	- Pass a name and a parameter type.
1742826539.5108869 [lidar_to_pointcloud-4] 	- Pass a name and a descriptor with `dynamic_typing=True
1742826539.5113518 [lidar_to_pointcloud-4]   warnings.warn(
1742826539.6657264 [lidar_to_pointcloud-4] [INFO] [1742826539.665475280] [lidar_to_pointcloud]: Map will be saved
1742826539.6662159 [lidar_to_pointcloud-4] [INFO] [1742826539.665854508] [lidar_to_pointcloud]: Map name is 3d_map.ply
`

LivvynHell, 2025-03-27:
Hi, did anyone manage to figure out why such errors are occuring? We've been running into similar problems. Thanks!

tfoldi, 2025-03-27:
couple of issues based on the log:

```[rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic```

before you run the robot or webrtc launch file, can you make sure that no other nodes are running in your system? (`ros2 node list` should be empty)

if you check `go2_driver_node` was not able to fully open the connection. this could be due to multiple reasons. Can you ping 192.168.2.126? Also, make sure that the mobile app is not running as only one application can connect via webrtc.

In short, it looks like for some reason multiple connections are open to your robot - which should not be the case. 

Also please make sure your robot is in normal mode (not in AI)

dvelho, 2025-03-27:
Thank you @tfoldi  Hi, yes I can ping the robot. I think I've tried without AI mode but will double check when I get time to try again. Will post if I can find a solution or what i've done wrong :)

jackhou66, 2025-03-27:
> Thank you [@tfoldi](https://github.com/tfoldi) Hi, yes I can ping the robot. I think I've tried without AI mode but will double check when I get time to try again. Will post if I can find a solution or what i've done wrong :)

have you get answers  , the same issue I met today

LivvynHell, 2025-04-02:
We've figured out it might be something connecting to the multiple connections or the werbtc or cyclonedds CONN_TYPE. 

We turned off the other router that was previously used and started with using our private one that had no other connections. The map was visualised just fine in foxglove studio, but rviz didn't generate the map from /robot0/point_cloud2 topic.

After two crashes that were probably caused by another PC being connected by ssh to the expansion dock the robot again didn't generate any map and all the limbs had no transform form to odom status in rviz.

Currently, we reinstalled the official version to test on the edu features and write our own topics since it's our uni project to try and figure out the robot for further educational purposes.

So far, the robot doesn't have the transform forms while it works on the official repo's software either - as visible in rviz. I will be sure to write more if we figure anything out... 😢

abizovnuralem, 2025-05-08:
Hello @LivvynHell,  it is interesting... I think you need to contact official Unitree support with your problem, hope they will help you to fix that!

closed 2025-05-08 by abizovnuralem