abizovnuralem/go2_ros2_sdk #146: Coco_detector_node can't receive images

Tony-t-coder, 2025-03-24:
### Description

When I launch coco_detector_node only, there is no output and warning or error messages. Then I launch robot.launch.py, coco_detector_node still can't receive images, warning:
```
"[WARN] [1742734969.449676632] [coco_detector_node]: New publisher discovered on topic '/camera/image_raw', offering incompatible QoS. No messages will be received from it. Last incompatible policy: RELIABILITY".
``` 
It is clearly that QoS policy leads to the issue.

### Locate the problem
There are 3 files called coco_detector_node.py in the workspace, when I stop coco_detector_node, it reports:
```
Traceback (most recent call last):
  File "/home/tyj/ros2_ws/install/coco_detector/lib/coco_detector/coco_detector_node", line 33, in <module>
    sys.exit(load_entry_point('coco-detector==0.0.0', 'console_scripts', 'coco_detector_node')())
  File "/home/tyj/ros2_ws/install/coco_detector/lib/coco_detector/coco_detector_node", line 25, in importlib_load_entry_point
    return next(matches).load()
  File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 171, in load
    module = import_module(match.group('module'))
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/tyj/ros2_ws/install/coco_detector/lib/python3.10/site-packages/coco_detector/coco_detector_node.py", line 126, in <module>
    rclpy.spin(coco_detector_node)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/__init__.py", line 226, in spin
    executor.spin_once()
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 739, in spin_once
    self._spin_once_impl(timeout_sec)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 728, in _spin_once_impl
    handler, entity, node = self.wait_for_ready_callbacks(timeout_sec=timeout_sec)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 711, in wait_for_ready_callbacks
    return next(self._cb_iter)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 608, in _wait_for_ready_callbacks
    wait_set.wait(timeout_nsec)
KeyboardInterrupt
[ros2run]: Interrupt
``` 
I noticed   **File "/home/tyj/ros2_ws/install/coco_detector/lib/python3.10/site-packages/coco_detector/coco_detector_node.py"** was working and could be the source of issue.

### Solve the problem
In class CocoDetectorNode, `__init__` function, the code about subscription definition is:
```python
self.subscription = self.create_subscription(
        Image,
        "/camera/image_raw",
        self.listener_callback,
        10)
```  
I believe it is not explicitly defining QoS policy that results in coco_detector_node receiving no image from "/camera/image_raw". I don't know whether it is my own problem or a common problem, but the code can be modified to fix it. 
First, check the QoS Profile of the publisher /camera/image_raw when robot.launch.py is launched:
```shell
ros2 topic info /camera/image_raw --verbose
```  
My output is shown below:
```
Node name: rviz2
Node namespace: /
Topic type: sensor_msgs/msg/Image
Endpoint type: SUBSCRIPTION
GID: 01.0f.22.9c.a7.16.66.4b.00.00.00.00.00.00.2c.04.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: BEST_EFFORT
  History (Depth): UNKNOWN
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite
``` 
According to "**Reliability: BEST_EFFORT**", modify the code as below

```python
qos_profile = QoSProfile(
        reliability=QoSReliabilityPolicy.BEST_EFFORT,  
        depth=10)
self.subscription = self.create_subscription(
        Image,
        "/camera/image_raw",
        self.listener_callback,
        qos_profile=qos_profile)
``` 

### System Info
- Connection type:webrtc
- Ubuntu 22.04
- ROS2: Humble

tfoldi, 2025-03-27:
thank you for reporting the issue. Would you mind sharing a PR with the fix? thanks!

abizovnuralem, 2025-08-02:
I’ve refactored the project. If you encounter any problems, please feel free to reopen the issue.