abizovnuralem/go2_ros2_sdk #147: Can I ask how to use unbuntu22.04 in unitree go2 edu

choikyoung, 2025-03-25:
Unitree go2 edu jetson orin nx 16gb is using ubuntu 20.04, ros2 foxy but I can't use this sdk becasue ubuntu version is not correct.
so I want to know how to change ubuntu20.04 to ubuntu 22.04 .
THANKS.

abizovnuralem, 2025-05-08:
I think it would be better to ask Unitree support first, since I know they have modifications on top of the NVIDIA Jetson.

closed 2025-05-08 by abizovnuralem

131460606, 2025-06-11:
do you know how to change ubuntu20.04 to ubuntu 22.04 and do you solve the above problem?

aptxlee, 2025-07-22:
Unitree Extension Dock does not support 20.04. However, it can be used with a separate Jetson or PC. If you want to use it, you can also use it by connecting to the host PC using cross communication between ROS2.