abizovnuralem/go2_ros2_sdk #149: can open the rviz2 but no data show and update

jackhou66, 2025-03-27:
I can get publish topic from robot ,when I show the rviz2
1.
/utlidar/mapping_cmd
/utlidar/range_info
/utlidar/range_map
/utlidar/robot_odom
/utlidar/robot_pose
/utlidar/switch
/utlidar/voxel_map
/utlidar/voxel_map_compressed
/uwbstate
/uwbswitch
/videohub/inner
/webrtcreq
/webrtcres
/wirelesscontroller
/wirelesscontroller_unprocessed
/xfk_webrtcreq
/xfk_webrtcres

2.issue   rviz2 has not update the data , no map , no camera data ,no lidar data 

[foxglove_bridge-9] [WARN] [1743092186.021554909] [foxglove_bridge]: Failed to add channel for topic "/arm_Feedback" (unitree_arm/msg/ArmString): package 'unitree_arm' not found, searching: [/home//Desktop/Nv2_tradit/ros2_ws/install/go2_robot_sdk, /home//Desktop/Nv2_tradit/ros2_ws/install/unitree_go, /home//Desktop/Nv2_tradit/ros2_ws/install/go2_interfaces, /home//Desktop/Nv2_tradit/ros2_ws/install/coco_detector, /home//unitree_ros2/cyclonedds_ws/install/unitree_hg, /home/jack/unitree_ros2/cyclonedds_ws/install/unitree_go, /home//unitree_ros2/cyclonedds_ws/install/unitree_api, /home//unitree_ros2/cyclonedds_ws/install/rmw_cyclonedds_cpp, /opt/ros/humble]
[controller_server-11] [WARN] [1743092186.348764058] [local_costmap.local_costmap]: Can't update static costmap layer, no map received
[planner_server-13] [WARN] [1743092187.866577586] [global_costmap.global_costmap]: Can't update static costmap layer, no map received
[controller_server-11] [WARN] [1743092196.348828345] [local_costmap.local_costmap]: Can't update static costmap layer, no map received
[planner_server-13] [WARN] [1743092197.878507105] [global_costmap.global_costmap]: Can't update static costmap layer, no map received
[controller_server-11] [WARN] [1743092206.348864587] [local_costmap.local_costmap]: Can't update static costmap layer, no map received
[planner_server-13] [WARN] [1743092207.864740374] [global_costmap.global_costmap]: Can't update static costmap layer, no map received
[controller_server-11] [WARN] [1743092216.348907058] [local_costmap.local_costmap]: Can't update static costmap layer, no map received
[planner_server-13] [WARN] [1743092217.861696454] [global_costmap.global_costmap]: Can't update static costmap layer, no map received
[controller_server-11] [WARN] [1743092226.348835912] [local_costmap.local_costmap]: Can't update static costmap layer, no map received
[planner_server-13] [WARN] [1743092227.863426966] [global_costmap.global_costmap]: Can't update static costmap layer, no map received

jackhou66, 2025-03-27:
and  the situation >
[local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "base_link" passed to canTransform argument source_frame - frame does not exist

youweicao, 2025-03-31:
Note that the IP is not 192.168.123.18, this is not robot ip, but a external computer ip (jetson's ip)

yitingHH-bit, 2025-04-10:
Oh , i will try again > Note that the IP is not 192.168.123.18, this is not robot ip, but a external computer ip (jetson's ip)

closed 2025-05-08 by abizovnuralem