abizovnuralem/go2_ros2_sdk #152: Question: What would your ros2 topic list and ros2 node list look like

countpro21, 2025-04-04:
Hi, i am trying to connect my pc to the go2 unitree EDU via wifi, i would like to know how would both ros2 topic and nodes list would look like after a complete connection. For example, my robot ip is 192.168.3.100 and the pc is 192.168.3.200, i have ssh in the robot ip (192.168.3.120) into my pc. I ensure that both ROS_DOMAIN_ID to be set to 0 and using rmw-cyclonedds_cpp to commnunicate, however i cannot see my robot topic list on my pc nor do i see the ros2 node list .

p.s. for some reason my robot doesn't seem to have any nodes running, that is not normal right?

Unitree GO2 EDU 
Ubuntu 20.04 , Foxy

PC
Ubuntu 22.04, humble

anyone who's project works could help out? Thanks!

abizovnuralem, 2025-05-08:
Hello @countpro21 , What method of connection are you trying?

countpro21, 2025-05-09:
Hi @abizovnuralem, i am currently using cyclonedds for the method of connection, i am unable to connect to the wifi via router for the robot, hence i am unable to use webrtc. I am working on a school project which requires Slam and nav2, does this feature work on cyclonedds or only in webrtc.

Problem with Wifi:
from my understanding of the GO2 EDU that i currently am working on, the robot is an mainland china unit as when i try to connect the robot to the app, the international verson does not allow for me to connect but the mainland app work, however, when promt to connect, i can only connect via ap mode and not router mode as the robot does not want to connect to the router. Does a wifi adapter work the same way as connecting to the app?

yurj1, 2025-05-15:
I also encountered a similar problem, I connected the go2 with an Ethernet cable

[image]

[image]

[image]

[image]

[image]

countpro21, 2025-05-16:
Hey @yurj1 , could you see your robot in rviz 2 being vizualised properly? From my understanding, if using cyclonedds, i did not need to enter ip address to vizualised the robot as it is connected via lan.

yurj1, 2025-05-16:
@countpro21 Hey Thank you for your reply. I have solved this problem by using the webrc method

closed 2025-08-02 by abizovnuralem