abizovnuralem/go2_ros2_sdk #158: Can't see lidar's points and mapping

p82garpr, 2025-05-22:
Hi,

It's really weird, i succesfully connect with the unitree go2 edu via webrtc, as you can see in the next screenshoot, it's possible to see the camera, but lidar is not showing any points.

[image]

But one thing that i do not understand is that, yesterday, i was trying the same thing, i didn't get it but i had to leave the room and then when i came suddenly it was working, indeed i saved a map too, but today it doesn't want to work.

I'm connecting as i said via webrtc with the IP that my router gave to the robot, i get this ip form unitree app.
 
I also checked the console, and i saw theses messages, idk if it could be related:
`[controller_server-11] [WARN] [1747907910.125239333] [local_costmap.local_costmap]: Can't update static costmap layer, no map received
[planner_server-13] [WARN] [1747907911.717838446] [global_costmap.global_costmap]: Can't update static costmap layer, no map received
[controller_server-11] [WARN] [1747907920.125269035] [local_costmap.local_costmap]: Can't update static costmap layer, no map received
[planner_server-13] [WARN] [1747907921.727976600] [global_costmap.global_costmap]: Can't update static costmap layer, no map received
[controller_server-11] [WARN] [1747907930.125057725] [local_costmap.local_costmap]: Can't update static costmap layer, no map received
[planner_server-13] [WARN] [1747907931.756137487] [global_costmap.global_costmap]: Can't update static costmap layer, no map received
[controller_server-11] [WARN] [1747907940.125073672] [local_costmap.local_costmap]: Can't update static costmap layer, no map received
[planner_server-13] [WARN] [1747907941.725433330] [global_costmap.global_costmap]: Can't update static costmap layer, no map received
[controller_server-11] [WARN] [1747907950.125033724] [local_costmap.local_costmap]: Can't update static costmap layer, no map received
[planner_server-13] [WARN] [1747907951.762299039] [global_costmap.global_costmap]: Can't update static costmap layer, no map received
[controller_server-11] [WARN] [1747907960.125068404] [local_costmap.local_costmap]: Can't update static costmap layer, no map received
[planner_server-13] [WARN] [1747907961.742283496] [global_costmap.global_costmap]: Can't update static costmap layer, no map received
[go2_driver_node-3] [h264 @ 0x76e50c000e40] No start code is found.
[go2_driver_node-3] [h264 @ 0x76e50c000e40] Error splitting the input into NAL units.
[go2_driver_node-3] WARNING:aiortc.codecs.h264:H264Decoder() failed to decode, skipping package: [Errno 1094995529] Invalid data found when processing input
[go2_driver_node-3] [h264 @ 0x76e50c000e40] No start code is found.
[go2_driver_node-3] [h264 @ 0x76e50c000e40] Error splitting the input into NAL units.
[go2_driver_node-3] WARNING:aiortc.codecs.h264:H264Decoder() failed to decode, skipping package: [Errno 1094995529] Invalid data found when processing input
[controller_server-11] [WARN] [1747907970.125090291] [local_costmap.local_costmap]: Can't update static costmap layer, no map received
[planner_server-13] [WARN] [1747907971.751432347] [global_costmap.global_costmap]: Can't update static costmap layer, no map received
`

Thanks for your help guys.

p82garpr, 2025-05-22:
[image]

maciejmajek, 2025-06-08:
In my case similar thing happened after updating the firmware of go2. Looks like the protocol changed? Im unable to retrieve anything other than the camera image. Nav2 stack also doesn't work but this is most likely caused by the lack of point cloud or/and movement protocol changes. 
Is there anything I could help with @abizovnuralem? Any pointers on how to approach this problem?

p82garpr, 2025-06-09:
> In my case similar thing happened after updating the firmware of go2. Looks like the protocol changed? Im unable to retrieve anything other than the camera image. Nav2 stack also doesn't work but [...]

That's what I thought, since everything conveniently stopped working right after the latest firmware update. I shouldn't have updated it, but who would have known that it would stop working because of it? I find it unacceptable that it's not possible to downgrade to an older version or that there's no changelog from Unitree detailing the changes suffered. I’ll say it again — if anyone has any approach to fix this... it's urgent for me.

abizovnuralem, 2025-06-09:
Hello @maciejmajek  @p82garpr ! I don't update my dog firmware, I am using the 1.0.x version, we had the same problems with 1.1.1, when the webrtc method didn't work, and we reupdated it. 
Currently, this SDK supports:
1.1.1 - 1.1.7 (latest available)
1.0.19 - 1.0.25

btw, I don't have a physical robot with me right now to debug it, but I think someone from the Roboverse can fix that soon, and I will update my repo to support a new version

zhaocong913, 2025-06-09:
i also meet the same problem，ubuntu22.04humble version

[image]

p82garpr, 2025-06-09:
> Hello [@maciejmajek](https://github.com/maciejmajek) [@p82garpr](https://github.com/p82garpr) ! I don't update my dog firmware, I am using the 1.0.x version, we had the same problems with 1.1.1, [...]

Thanks for your message, I hope it's fixed ASAP, if you have any clues let me know and we'll try to make a first approach

maciejmajek, 2025-06-10:
@p82garpr have you tried this? https://discord.com/channels/1205243330137690195/1209628854747136060/1381008315248803942
https://github.com/juhasch/go2_ros2_sdk/blob/fix/ros2_jazzy_mac/go2_robot_sdk/scripts/webrtc_driver.py

p82garpr, 2025-06-10:
Nop, i didn't, but as i read, seems like it never worked for the discord guy until he realised that was using incorrect aioice version, the fact here is that once worked, i updated firmware and then stopped working anymore. Anyway, i will try it later, maybe tomorrow.
And btw, yes, im on 1.1.7 as I can see on unitree app.

NicolasBlumetto, 2025-06-17:
Hi @p82garpr any progress on this issue? I am new to the SDK and I am facing the same problem. I am also running the latest version of the firmware.

louisarbor, 2025-06-18:
See Roboverse discord channel: https://discordapp.com/channels/1205243330137690195/1209628854747136060 

There is a patch:  https://github.com/juhasch/go2_ros2_sdk/blob/fix/ros2_jazzy_mac/go2_robot_sdk/go2_robot_sdk/go2_driver_node.py

kExU9853, 2025-06-19:
Hi there , I met the exact same issue now. Even worse, I cannot see the camera image, I found my hardware is v1.0 and software  version is 1.1.7 according to the unitree mobile app. Any ideas so far

Ammarazab, 2025-07-04:
Now after debugging I reach the status of foot value is 0
Like this 

[image]

But in Go2 App the data is shown correctly

[image]

Ammarazab, 2025-07-10:
I solve this issue its related to the sport status doesn't publish the foot position value so I need to take it from low state from motor_state and include it in the code
you can see my code in this repo

https://github.com/Ammarazab/go2_ros2_sdk

@p82garpr  give me your feedback I did it in go2 Pro