abizovnuralem/go2_ros2_sdk #169: Related Issue: No Lidar and Map from Docker — similar to #158

kExU9853, 2025-06-19:
Hi @abizovnuralem and contributors,

First of all, thank you for your great work on this SDK — it's an exciting project and I really appreciate the effort you're putting into it.

I'm encountering an issue very similar to [#158](https://github.com/abizovnuralem/go2_ros2_sdk/issues/158):
When launching the system inside Docker on Ubuntu 20.04, I cannot receive any Lidar data or map, and camera images are not displaying correctly in RViz.

My setup:

Hardware version: v1.0
Software version: v1.1.7 (as shown in the Unitree Mobile App)
I believe this is related to Docker compatibility, not a firmware version mismatch — similar to what was described in #158. I tried both connection type but neither of these two works

In that issue, a patched version of go2_driver_node.py was mentioned (by @juhasch):
🔗 https://github.com/juhasch/go2_ros2_sdk/blob/fix/ros2_jazzy_mac/go2_robot_sdk/go2_robot_sdk/go2_driver_node.py

However:

I’m not sure how to properly apply that patch within the Docker environment
The linked Roboverse Discord channel is not publicly accessible:
https://discordapp.com/channels/1205243330137690195/1209628854747136060
🙏 Questions:
Could someone clarify how to use this patched version of go2_driver_node.py within a Docker-based ROS 2 workspace?
Is there a recommended fix or patch for Docker + ROS 2 Humble + firmware v1.1.7 users?
Thanks again for your time and help — I really appreciate all the work being done on this project! 😊

phybrain, 2025-06-27:
same issue

abizovnuralem, 2025-08-02:
I’ve refactored the project. If you encounter any problems, please feel free to reopen the issue.

closed 2025-08-02 by abizovnuralem