abizovnuralem/go2_ros2_sdk #191: Has the issue with radar point cloud not displaying been fixed and merged into the main branch?

nanbloom001, 2025-08-09:
请问雷达点云不显示的问题的修复合并到主分支了吗，我使用了最新主分支的代码，但是进入rviz后还是没有积累的雷达点云的显示，但是/utlidar/cloud_deskewed是可以正常显示的，图像以及机器狗模型都可以正常显示，我的硬件版本是2.0.0,软件版本是1.1.7。我注意到这个issue在1个星期前被关闭，但是似乎使用了最新的代码仍然存在问题，下面附上我的图片和日志输出，可以帮我查看一下吗，谢谢了

Has the fix for the radar point cloud display issue been merged into the master branch? I'm using the latest master branch code, but after entering rviz, the accumulated radar point cloud still doesn't display. However, /utlidar/cloud_deskewed displays normally, as do the images and the robot dog model. My hardware version is 2.0.0, and my software version is 1.1.7. I noticed this issue was closed a week ago, but it seems the issue persists even with the latest code. Attached below are my images and log output. Could you please take a look? Thank you very much

[image]

[go2_logs.txt](https://github.com/user-attachments/files/21697777/go2_logs.txt)

bpesun, 2025-08-10:
You should let the dog stand up to get the lidar date from WebRTC.  Another thing to note is to ensure that your NumPy version is1.26.4.  When everything is running correctly, you can check whether the pose of Go2 in RViz2 is synchronized with the real pose, or if there is a delay. How long is the delay approximately? If possible, could you record a video（or link of your video） that shows the motion of both RViz and the real robot dog simultaneously? In my testing, I found that the delay is about 3 seconds, which is quite frustrating. Hope to chat with you later.

nanbloom001, 2025-08-10:
> You should let the dog stand up to get the lidar date from WebRTC. Another thing to note is to ensure that your NumPy version is1.26.4. When everything is running correctly, you can check whether the pose of Go2 in RViz2 is synchronized with the real pose, or if there is a delay. How long is the delay approximately? If possible, could you record a video（or link of your video） that shows the motion of both RViz and the real robot dog simultaneously? In my testing, I found that the delay is about 3 seconds, which is quite frustrating. Hope to chat with you later.你应该让狗狗站起来，以便从 WebRTC 获取激光雷达数据。另外需要注意的是，确保你的 NumPy 版本是 1.26.4。一切正常运行后，你可以检查 RViz2 中 Go2 的姿态是否与真实姿态同步，或者是否存在延迟。延迟大约多长时间？如果可以的话，能否录制一段视频（或视频链接），同时展示 RViz 和真实机器狗的运动？在我的测试中，我发现延迟大约有 3 秒，这很令人沮丧。希望以后能和你交流。

I can't seem to upload the video on github, I uploaded him to Baidu Netdisk, you can try to download it

通过网盘分享的文件：log.mp4
链接: https://pan.baidu.com/s/1YKVRbZmJt0ib1o4CBHnqfg?pwd=2ybr 提取码: 2ybr 复制这段内容后打开百度网盘手机App，操作更方便哦 
--来自百度网盘超级会员v5的分享

nanbloom001, 2025-08-10:
> So what is your conclusion? 1）How long is the delay between the motion of the real robot dog and the robot dog in RViz?那么你的结论是什么？1）真实机器狗的运动和 RViz 中的机器狗的运动之间的延迟是多少？ 2）Can Nav2 achieve navigation functions smoothly?2）Nav2 能顺利实现导航功能吗？

The delay is about 3 seconds, and the limbs are separated from the body model. The navigation cannot reach the destination smoothly, and the go2 will keep spinning.

abizovnuralem, 2025-08-13:
The delay between the robot joints’ synchronization is due to the new method we implemented to address an issue with the 1.1.7 firmware. I identified the cause — the WebRTC topic that contains the joint state information has an update rate of about 1 Hz. Will be happy for the pr to fix it; the previous method, which was designed by me, doesn't have this issue.

kouyanying, 2025-09-15:
> > > > You should let the dog stand up to get the lidar date from WebRTC. Another thing to note is to ensure that your NumPy version is1.26.4. When everything is running correctly, you can check whether the pose of Go2 in RViz2 is synchronized with the real pose, or if there is a delay. How long is the delay approximately? If possible, could you record a video（or link of your video） that shows the motion of both RViz and the real robot dog simultaneously? In my testing, I found that the delay is about 3 seconds, which is quite frustrating. Hope to chat with you later.你应该让狗狗站起来，以便从 WebRTC 获取激光雷达数据。另外需要注意的是，确保你的 NumPy 版本是 1.26.4。一切正常运行后，你可以检查 RViz2 中 Go2 的姿态是否与真实姿态同步，或者是否存在延迟。延迟大约多长时间？如果可以的话，能否录制一段视频（或视频链接），同时展示 RViz 和真实机器狗的运动？在我的测试中，我发现延迟大约有 3 秒，这很令人沮丧。希望以后能和你交流。你应该让狗狗站起来，以便从 WebRTC 获取激光雷达数据。另外需要注意的是，确保你的 NumPy 版本是 1.26.4。一切正常运行后，你可以检查 RViz2 中 Go2 的姿态是否与真实姿态同步，或者是否存在延迟。延迟大约多长时间？如果可以的话，能否录制一段视频（或视频链接），同时展示 RViz 和真实机器狗的运动？在我的测试中，我发现延迟大约为 3 秒，这相当令人沮丧。希望稍后能和你聊天。你应该让狗狗站起来，以便从 WebRTC 获取激光雷达数据。另外需要注意的是，确保你的 NumPy 版本是 1.26.4。一切正常运行后，你检查 RViz2 中 Go2 的姿势是否可以与真实姿势同步，或者是否存在延迟。延迟多长时间？如果的话，可以延长录制一段视频（或视频链接），同时显示 RViz 和真正的机器狗的运动？在我的测试中，我发现延迟大约有 3 秒，这很令人沮丧。希望以后能和你交流。
> > > 
> > > 
> > > I can't seem to upload the video on github, I uploaded it to Baidu Netdisk, you can try to download it我好像无法上传视频到 github，我把它上传到百度网盘了，你可以尝试下载
> > > 通过网盘分享的文件：log.mp4 链接: https://pan.baidu.com/s/1YKVRbZmJt0ib1o4CBHnqfg?pwd=2ybr 提取码: 2ybr 复制这段内容后打开百度网盘手机App，操作更方便哦 --来自百度网盘超级会员v5的分享
> > 
> > 
> > 您好，我也遇到了这个问题，请问您解决了吗？Hello, I’m having the same issue. May I ask if you’ve resolved it?您好，我也遇到了这个问题，请问您解决了吗？您好，我也遇到了同样的问题。请问你解决了吗？
> 
> 这个问题可能需要等待作者的更新，我是使用了另一个项目来实现建图导航的功能

我已经卡在这里好久了，另一个项目是开源的吗，方便分享一下吗