abizovnuralem/go2_ros2_sdk #198: CycloneDDS - no data

joujo1212, 2025-09-05:
Hi,
I'm struggling with CycloneDDS type of connection. I follow readme, but seems like no data are coming into go2_ros2_sdk topics. 

I'm sure that cycloneDDS is working on my jetson as I can see data via "cyclonedds ls", also coming data thru ros2 topics as I installed unitree_ros2 library. My setup:

1. Jetson is on the same subnet as robot's computer
2. Cyclonedds configured via xml to see data from correct ethernet
3. Running these commands:
```
export ROBOT_IP="192.168.123.161"
export CONN_TYPE="cyclonedds"
ros2 launch go2_robot_sdk robot.launch.py
```
4. I don't see any valid errors when launch robo.launch.py, just no data in topics like /point_cloud2. When I'm using webrtc, data are coming nicely. Maybe the CycloneDDS is not implemented in this lib (even though the check list saying yes)?

abizovnuralem, 2025-09-18:
Hello! What version of Unitree Go2 do you have?

samanthagoodhue, 2025-09-30:
Hello @abizovnuralem I am having the same issue with the Go2 EDU. If you have any insight

yyymmmlll1234, 2025-12-09:
I  have the same issue with the Go2 EDU,hardware v2.0 and fireware v1.1.11. How can I to support it?
export ROBOT_IP="192.168.123.18"
export CONN_TYPE="cyclonedds"
ros2 launch go2_robot_sdk robot.launch.py
It is very nice when use webrtc to run Go2 EDU.